ilpincy/argos3 #12: no member named 'SaveAsImage' in 'argos::CFloorEntity'

dbl001, 2015-08-01:
Hi,

I'm running on OS X 10.10.4.

I got this error building ARGoS:

/Users/davidlaxer/argos3/src/plugins/simulator/visualizations/qt-opengl/qtopengl_widget.cpp:726:39: error: 
      no member named 'SaveAsImage' in 'argos::CFloorEntity'
            m_cSpace.GetFloorEntity().SaveAsImage("/tmp/argos_floor.png");
            ~~~~~~~~~~~~~~~~~~~~~~~~~ ^
3 warnings and 2 errors generated.
make[2]: *** [plugins/simulator/visualizations/qt-opengl/CMakeFiles/argos3plugin_simulator_qtopengl.dir/qtopengl_widget.cpp.o] Error 1
make[1]: *** [plugins/simulator/visualizations/qt-opengl/CMakeFiles/argos3plugin_simulator_qtopengl.dir/all] Error 2
make: *** [all] Error 2

Could this be the problem?
-- FreeImage not found, image source and OpenGL drawing for the floor entity won't be available

David-Laxers-MacBook-Pro:build_simulator davidlaxer$ gcc --version
gcc (MacPorts gcc48 4.8.5_0) 4.8.5
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

David-Laxers-MacBook-Pro:build_simulator davidlaxer$ 

I first installed FreeImage with MacPorts.  But 'cmake ../src' couldn't find it.

David-Laxers-MacBook-Pro:src davidlaxer$ sudo port install FreeImage
Password:
--->  Fetching archive for freeimage
--->  Attempting to fetch freeimage-3.16.0_1.darwin_14.x86_64.tbz2 from http://packages.macports.org/freeimage
--->  Attempting to fetch freeimage-3.16.0_1.darwin_14.x86_64.tbz2.rmd160 from http://packages.macports.org/freeimage
--->  Installing freeimage @3.16.0_1
--->  Activating freeimage @3.16.0_1
--->  Cleaning freeimage
--->  Updating database of binaries
--->  Scanning binaries for linking errors
--->  No broken files found.                      

David-Laxers-MacBook-Pro:build_simulator davidlaxer$ cmake ../src
-- GCC/G++ version >= 4.2.1
-- GSL using gsl-config /opt/local/bin/gsl-config
-- Using GSL from /opt/local
-- FreeImage not found, image source and OpenGL drawing for the floor entity won't be available
-- Could NOT find ASCIIDoc (missing:  ASCIIDOC_EXECUTABLE) 
-- Could NOT find Lua52 (missing:  LUA_LIBRARIES LUA_INCLUDE_DIR) 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/davidlaxer/argos3/build_simulator

-- FreeImage not found, image source and OpenGL drawing for the floor entity won't be available

Next, I installed and build FreeImage from github, that didn't work either.

David-Laxers-MacBook-Pro:~ davidlaxer$ git clone https://github.com/Kanma/FreeImage.git
Cloning into 'FreeImage'...
remote: Counting objects: 650, done.
remote: Total 650 (delta 0), reused 0 (delta 0), pack-reused 650
Receiving objects: 100% (650/650), 2.40 MiB | 1.53 MiB/s, done.
Resolving deltas: 100% (199/199), done.
Checking connectivity... done.
David-Laxers-MacBook-Pro:~ davidlaxer$ git submodule init
fatal: Not a git repository (or any of the parent directories): .git
David-Laxers-MacBook-Pro:~ davidlaxer$ cd FreeImage/
David-Laxers-MacBook-Pro:FreeImage davidlaxer$ git submodule init
Submodule 'dependencies/XMake' (git://github.com/Kanma/XMake.git) registered for path 'dependencies/XMake'
Submodule 'dependencies/zlib' (git://github.com/Kanma/zlib.git) registered for path 'dependencies/zlib'
David-Laxers-MacBook-Pro:FreeImage davidlaxer$ git submodule update
Cloning into 'dependencies/XMake'...
[... 545 lines cut ...]
      [-Wdeprecated-declarations]
      gluPickMatrix(un_x,
      ^
/System/Library/Frameworks/OpenGL.framework/Headers/glu.h:279:13: note: 
      'gluPickMatrix' has been explicitly marked deprecated here
extern void gluPickMatrix (GLdouble x, GLdouble y, GLdouble delX, GLdoub...
            ^
/Users/davidlaxer/argos3/src/plugins/simulator/visualizations/qt-opengl/qtopengl_widget.cpp:286:7: warning: 
      'gluPerspective' is deprecated: first deprecated in OS X 10.9 - "Use
      GLKMatrix4MakePerspective" [-Wdeprecated-declarations]
      gluPerspective(m_cCamera.GetActiveSettings().YFieldOfView.GetValue(),
      ^
/System/Library/Frameworks/OpenGL.framework/Headers/glu.h:278:13: note: 
      'gluPerspective' has been explicitly marked deprecated here
extern void gluPerspective (GLdouble fovy, GLdouble aspect, GLdouble zNe...
            ^
/Users/davidlaxer/argos3/src/plugins/simulator/visualizations/qt-opengl/qtopengl_widget.cpp:726:39: error: 
      no member named 'SaveAsImage' in 'argos::CFloorEntity'
            m_cSpace.GetFloorEntity().SaveAsImage("/tmp/argos_floor.png");
            ~~~~~~~~~~~~~~~~~~~~~~~~~ ^
3 warnings and 2 errors generated.
make[2]: *** [plugins/simulator/visualizations/qt-opengl/CMakeFiles/argos3plugin_simulator_qtopengl.dir/qtopengl_widget.cpp.o] Error 1
make[1]: *** [plugins/simulator/visualizations/qt-opengl/CMakeFiles/argos3plugin_simulator_qtopengl.dir/all] Error 2
make: *** [all] Error 2
David-Laxers-MacBook-Pro:build_simulator davidlaxer$

ilpincy, 2015-08-01:
Hi David,

There is indeed a little compilation issue in ARGoS when FreeImage is not
found.

Whenever you install a new library and want CMake to retry and find it, you
need to delete the CMakeCache.txt file. If the library is installed in a
non-standard place, it is enough to use tools such as  cmake-gui or ccmake
to set the value of the relevant variables manually. Just refer to the
manual of these tools to see how it's done. Alternatively, you can run
cmake as follows:

$ cmake -DFREEIMAGE_PLUS_INCLUDE_PATH=/path/to/freeimage/headers
-DFREEIMAGE_PLUS_LIBRARY=/path/to/freeimage/library.dylib

I have committed a fix that should allow ARGoS to compile when FreeImage is
not installed, but the QTOpenGL visualization is compiled. It's definitely
a bad configuration, though, because it corresponds to the situation in
which you set a custom texture for the floor which is visible by the ground
sensor, but can't be drawn by OpenGL. I would avoid it if possible.

Cheers,
Carlo

closed 2015-08-01 by ilpincy

dbl001, 2015-08-01:
Progress!

CMake Error at cmake_install.cmake:43 (file):
  file INSTALL cannot find
  "/Users/davidlaxer/argos3/build_simulator/argos3.1.gz".

David-Laxers-MacBook-Pro:build_simulator davidlaxer$ sudo make install
[ 21%] Built target argos3core_simulator
[ 21%] Built target argos3
[ 22%] Built target argos3plugin_simulator_media
[ 29%] Built target argos3plugin_simulator_entities
[ 44%] Built target argos3plugin_simulator_qtopengl
[ 61%] Built target argos3plugin_simulator_dynamics2d
[ 63%] Built target argos3plugin_simulator_pointmass3d
[ 77%] Built target argos3plugin_simulator_genericrobot
[ 87%] Built target argos3plugin_simulator_footbot
[ 91%] Built target argos3plugin_simulator_eyebot
[ 92%] Built target argos3plugin_simulator_epuck
[ 93%] Built target argos3plugin_simulator_spiri
[ 94%] Built target argos3_prof
[ 94%] Built target test-client
[ 95%] Built target test-cylinder
[ 95%] Built target test-grid
[ 96%] Built target test-list
[ 96%] Built target test-range
[ 97%] Built target test-server
[ 97%] Built target test_eyebot_controller
[ 98%] Built target test_footbot_controller
[100%] Built target test_loop_functions
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/bin/argos3
-- Installing: /usr/local/share/argos3/uninstall_argos3.sh
-- Installing: /usr/local/include/argos3/core/config.h
CMake Error at cmake_install.cmake:43 (file):
  file INSTALL cannot find
  "/Users/davidlaxer/argos3/build_simulator/argos3.1.gz".

make: *** [install] Error 1
David-Laxers-MacBook-Pro:build_simulator davidlaxer$ find .. -name argos3.1.gz -ls
David-Laxers-MacBook-Pro:build_simulator davidlaxer$

ilpincy, 2015-08-01:
You need to create the documentation before trying to install. In practice,
before you can type 'make install', you need to type 'make doc'. Check
here:
https://github.com/ilpincy/argos3#installing-argos-from-the-compiled-binaries

Also, since you're compiling ARGoS manually, make sure you pick the release
compilation configuration. To do this, you need to type:

$ cmake -DCMAKE_BUILD_TYPE=Release ../src

If necessary, you also need to specify the -D flags corresponding to
FreeImage as discussed before. Check here:
https://github.com/ilpincy/argos3#advanced-compilation-configuration.

Cheers,
Carlo

dbl001, 2015-08-01:
Ok. Progress …but,

--   package 'argos3_simulator' not found

1. I rebuild ARGoS: 
$ cmake -DCMAKE_BUILD_TYPE=Release ../src
$ make
$ make doc
$ sudo make install

No errors!

2. argos3-examples
 I removed the argos3-examples build directory, then

$ cmake -DCMAKE_BUILD_TYPE=Release ..
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ rm -rf build
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ mkdir build
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ cd build
David-Laxers-MacBook-Pro:build davidlaxer$ cmake -DCMAKE_BUILD_TYPE=Release ..
-- The C compiler identification is AppleClang 6.1.0.6020053
-- The CXX compiler identification is AppleClang 6.1.0.6020053
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.28") 
-- checking for module 'argos3_simulator'
--   package 'argos3_simulator' not found
CMake Error at /opt/local/share/cmake-3.2/Modules/FindPkgConfig.cmake:340 (message):
  A required package was not found
Call Stack (most recent call first):
  /opt/local/share/cmake-3.2/Modules/FindPkgConfig.cmake:502 (_pkg_check_modules_internal)
  CMakeLists.txt:12 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/Users/davidlaxer/argos3-examples/build/CMakeFiles/CMakeOutput.log".
David-Laxers-MacBook-Pro:build davidlaxer$

ilpincy, 2015-08-01:
This is due to MacPorts. It installs stuff in /opt, and apparently it
messes up the search path of pkg-config.

Where did you install ARGoS? If you left the default location, it should be
/usr/local. Let's call it $PREFIX. You can suggest additional search paths
for pkg-config by doing the following:

$ export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:$PREFIX/lib/pkgconfig

Make sure that the file argos3_simulator.pc is indeed in the
folder $PREFIX/lib/pkgconfig.

And then retry cmake:

$ cmake -DCMAKE_BUILD_TYPE=Release ..

Cheers,
Carlo

dbl001, 2015-08-01:
> Where did you install ARGoS?
/usr/local

$ find /usr/local -name argos3_simulator.pc -ls
122693128        8 -rw-r--r--    1 root             staff                 319 Jul 31 19:47 /usr/local/lib/pkgconfig/argos3_simulator.pc

(Sorry for the long output!)

David-Laxers-MacBook-Pro:build_simulator davidlaxer$ sudo make install
Password:
[ 21%] Built target argos3core_simulator
[ 21%] Built target argos3
[ 22%] Built target argos3plugin_simulator_media
[ 29%] Built target argos3plugin_simulator_entities
[ 44%] Built target argos3plugin_simulator_qtopengl
[ 61%] Built target argos3plugin_simulator_dynamics2d
[ 63%] Built target argos3plugin_simulator_pointmass3d
[ 77%] Built target argos3plugin_simulator_genericrobot
[ 87%] Built target argos3plugin_simulator_footbot
[ 91%] Built target argos3plugin_simulator_eyebot
[ 92%] Built target argos3plugin_simulator_epuck
[ 93%] Built target argos3plugin_simulator_spiri
[ 94%] Built target argos3_prof
[ 94%] Built target test-client
[ 95%] Built target test-cylinder
[ 95%] Built target test-grid
[ 96%] Built target test-list
[ 96%] Built target test-range
[ 97%] Built target test-server
[ 97%] Built target test_eyebot_controller
[ 98%] Built target test_footbot_controller
[100%] Built target test_loop_functions
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/bin/argos3
-- Up-to-date: /usr/local/share/argos3/uninstall_argos3.sh
-- Up-to-date: /usr/local/include/argos3/core/config.h
-- Installing: /usr/local/share/man/man1/argos3.1.gz
-- Up-to-date: /usr/local/doc/argos3/api
-- Installing: /usr/local/doc/argos3/api/a00001.html
-- Installing: /usr/local/doc/argos3/api/a00001.js
-- Installing: /usr/local/doc/argos3/api/a00002.html
-- Installing: /usr/local/doc/argos3/api/a00002.js
-- Installing: /usr/local/doc/argos3/api/a00003.html
-- Installing: /usr/local/doc/argos3/api/a00003.js
-- Installing: /usr/local/doc/argos3/api/a00004.html
-- Installing: /usr/local/doc/argos3/api/a00004.js
-- Installing: /usr/local/doc/argos3/api/a00005.html
-- Installing: /usr/local/doc/argos3/api/a00005.js
-- Installing: /usr/local/doc/argos3/api/a00006.html
-- Installing: /usr/local/doc/argos3/api/a00006.js
-- Installing: /usr/local/doc/argos3/api/a00007.html
-- Installing: /usr/local/doc/argos3/api/a00007.js
-- Installing: /usr/local/doc/argos3/api/a00008.html
-- Installing: /usr/local/doc/argos3/api/a00008.js
-- Installing: /usr/local/doc/argos3/api/a00009.html
-- Installing: /usr/local/doc/argos3/api/a00009.js
-- Installing: /usr/local/doc/argos3/api/a00010.html
-- Installing: /usr/local/doc/argos3/api/a00010.js
-- Installing: /usr/local/doc/argos3/api/a00011.html
-- Installing: /usr/local/doc/argos3/api/a00011.js
-- Installing: /usr/local/doc/argos3/api/a00012.html
-- Installing: /usr/local/doc/argos3/api/a00012.js
-- Installing: /usr/local/doc/argos3/api/a00013.html
-- Installing: /usr/local/doc/argos3/api/a00013.js
-- Installing: /usr/local/doc/argos3/api/a00014.html
-- Installing: /usr/local/doc/argos3/api/a00014.js
-- Installing: /usr/local/doc/argos3/api/a00015.html
-- Installing: /usr/local/doc/argos3/api/a00015.js
-- Installing: /usr/local/doc/argos3/api/a00016.html
[... 1174 lines cut ...]
-- Installing: /usr/local/doc/argos3/api/a00611.html
-- Installing: /usr/local/doc/argos3/api/a00611.js
-- Installing: /usr/local/doc/argos3/api/a00612.html
-- Installing: /usr/local/doc/argos3/api/a00612.js
-- Installing: /usr/local/doc/argos3/api/a00613.html
-- Installing: /usr/local/doc/argos3/api/a00613.js
-- Installing: /usr/local/doc/argos3/api/a00614.html
-- Installing: /usr/local/doc/argos3/api/a00614.js
-- Installing: /usr/local/doc/argos3/api/a00615.html
-- Installing: /usr/local/doc/argos3/api/a00615.js
-- Installing: /usr/local/doc/argos3/api/a00616.html
-- Installing: /usr/local/doc/argos3/api/a00616.js
-- Installing: /usr/local/doc/argos3/api/a00617.html
-- Installing: /usr/local/doc/argos3/api/a00617.js
-- Installing: /usr/local/doc/argos3/api/a00618.html
-- Installing: /usr/local/doc/argos3/api/a00618.js
-- Installing: /usr/local/doc/argos3/api/a00619.html
-- Installing: /usr/local/doc/argos3/api/a00619.js
-- Installing: /usr/local/doc/argos3/api/a00620.html
-- Installing: /usr/local/doc/argos3/api/a00620.js
-- Installing: /usr/local/doc/argos3/api/a00621.html
-- Installing: /usr/local/doc/argos3/api/a00621.js
-- Installing: /usr/local/doc/argos3/api/a00622.html
-- Installing: /usr/local/doc/argos3/api/a00622.js
-- Installing: /usr/local/doc/ar

ilpincy, 2015-08-01:
OK, then you just need to type these two commands:

$ export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/usr/local/lib/pkgconfig
$ cmake -DCMAKE_BUILD_TYPE=Release ..

And it should work. Also, consider switching to HomeBrew. Life is usually easier than with MacPorts due to the fact that HomeBrew installs stuff in standard paths (/usr/local) rather than non-standard ones (/opt).

dbl001, 2015-08-01:
>> Also, consider switching to HomeBrew.
I have considered switching.  However, I recently read that the HomeBrew coverage is still a small subset Macports.

http://stackoverflow.com/questions/21374366/what-is-the-difference-usage-of-homebrew-macports-or-other-package-installation <http://stackoverflow.com/questions/21374366/what-is-the-difference-usage-of-homebrew-macports-or-other-package-installation>

Macports has many more packages (~18.6 K) than there are Homebrew formulae (~3.1K), owing to its maturity. Homebrew is slowly catching up though.

dbl001, 2015-08-01:
More progress … but,

$ argos3 -c experiments/diffusion_1.argos
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.dylib"
[FATAL] Can't load library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib" even after trying to add extensions for shared library (dylib) and module library (so)

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ find /usr/local -name libargos3plugin_simulator_epuck.dylib -ls
122700698      408 -rwxr-xr-x    1 root             wheel              207316 Aug  1 09:24 /usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib

ilpincy, 2015-08-01:
That's a weird one. I have never seen such a problem, and I have hundreds
of Mac users. Is that all the output you get?

dbl001, 2015-08-01:
Yes.  Here’s another example:

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ argos3 -c experiments/diffusion_10.argos
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.dylib"
[FATAL] Can't load library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib" even after trying to add extensions for shared library (dylib) and module library (so)
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ sudo find /usr/local -name libargos3plugin_simulator_epuck.dylib -ls
Password:
122700698      408 -rwxr-xr-x    1 root             wheel              207316 Aug  1 09:24 /usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ 

Is there a log file generated or a way for me to enable tracing?

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ sudo !!
sudo dtruss argos3 -c experiments/diffusion_10.argos
SYSCALL(args) 		 = return
thread_selfid(0x0, 0x0, 0x0)		 = 4861695 0
csops(0x0, 0x0, 0x7FFF5ADF2CD8)		 = 0 0
issetugid(0x0, 0x0, 0x7FFF5ADF2CD8)		 = 0 0
shared_region_check_np(0x7FFF5ADF0C18, 0x0, 0x7FFF5ADF2CD8)		 = 0 0
stat64("/usr/lib/dtrace/libdtrace_dyld.dylib\0", 0x7FFF5ADF1DA8, 0x7FFF5ADF2CD8 = 0 0
open("/usr/lib/dtrace/libdtrace_dyld.dylib\0", 0x0, 0x0)		 = 3 0
pread(0x3, "\312\376\272\276\0", 0x1000, 0x0)		 = 4096 0
pread(0x3, "\317\372\355\376\a\0", 0x1000, 0x6000)		 = 4096 0
fcntl(0x3, 0x3D, 0x7FFF5ADF0110)		 = 0 0
mmap(0x104EAB000, 0x2000, 0x5, 0x12, 0x3, 0x6000)		 = 0x104EAB000 0
mmap(0x104EAD000, 0x1000, 0x3, 0x12, 0x3, 0x8000)		 = 0x104EAD000 0
mmap(0x104EAE000, 0x1FC0, 0x1, 0x12, 0x3, 0x9000)		 = 0x104EAE000 0
close(0x3)		 = 0 0
stat64("/usr/lib/dtrace/libdtrace_dyld.dylib\0", 0x7FFF5ADF2728, 0x1)		 = 0 0
stat64("/usr/lib/libncurses.5.4.dylib\0", 0x7FFF5ADF1BC8, 0x1)		 = 0 0
stat64("/usr/lib/libSystem.B.dylib\0", 0x7FFF5ADF1BC8, 0x1)		 = 0 0
stat64("/usr/lib/system/libcache.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libcommonCrypto.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libcompiler_rt.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libcopyfile.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libcorecrypto.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libdispatch.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libdyld.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libkeymgr.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/liblaunch.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libmacho.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libquarantine.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libremovefile.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_asl.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_blocks.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_c.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_configuration.dylib\0", 0x7FFF5ADF16D8, 0x1)	 = 0 0
stat64("/usr/lib/system/libsystem_coreservices.dylib\0", 0x7FFF5ADF16D8, 0x1)	 = 0 0
stat64("/usr/lib/system/libsystem_coretls.dylib\0", 0x7FFF5ADF16D8, 0x1)	 = 0 0
stat64("/usr/lib/system/libsystem_dnssd.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_info.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_kernel.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_m.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_malloc.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_network.dylib\0", 0x7FFF5ADF16D8, 0x1)	 = 0 0
stat64("/usr/lib/system/libsystem_networkextension.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_notify.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_platform.dylib\0", 0x7FFF5ADF16D8, 0x1)	 = 0 0
stat64("/usr/lib/system/libsystem_pthread.dylib\0", 0x7FFF5ADF16D8, 0x1)	 = 0 0
stat64("/usr/lib/system/libsystem_sandbox.dylib\0", 0x7FFF5ADF16D8, 0x1)	 = 0 0
stat64("/usr/lib/system/libsystem_secinit.dylib\0", 0x7FFF5ADF16D8, 0x1)	 = 0 0
stat64("/usr/lib/system/libsystem_stats.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libsystem_trace.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libunc.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libunwind.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/system/libxpc.dylib\0", 0x7FFF5ADF16D8, 0x1)		 = 0 0
stat64("/usr/lib/libobjc.A.dylib\0", 0x7FFF5ADF09B8, 0x1)		 = 0 0
stat64("/usr/lib/libauto.dylib\0", 0x7FFF5ADF09B8, 0x1)		 = 0 0
[... 108 lines cut ...]
sigprocmask(0x1, 0x0, 0x7FFF5ADF33E0)		 = 0x0 0
sigaltstack(0x0, 0x7FFF5ADF33D0, 0x7FFF5ADF33E0)		 = 0 0
sigprocmask(0x1, 0x0, 0x7FFF5ADF2AE0)		 = 0x0 0
sigaltstack(0x0, 0x7FFF5ADF2AD0, 0x7FFF5ADF2AE0)		 = 0 0
sigprocmask(0x3, 0x7FFF5ADF2AE0, 0x0)		 = 0x0 0
sigreturn(0x0, 0x80000000, 0x0)		 = 0 0
sigprocmask(0x1, 0x0, 0x7FFF5ADF33E0)		 = 0x0 0
sigaltstack(0x0, 0x7FFF5ADF33D0, 0x7FFF5ADF33E0)		 = 0 0
sigprocmask(0x1, 0x0, 0x7FFF5ADF33E0)		 = 0x0 0
sigaltstack(0x0, 0x7FFF5ADF33D0, 0x7FFF5ADF33E0)		 = 0 0
sigprocmask(0x1, 0x0, 0x7FFF5ADF33E0)		 = 0x0 0
[... cut ...]
sigprocmask(0x1, 0x0, 0x7FFF5ADF33E0)		 = 0x0 0
sigaltstack(0x0, 0x7FFF5ADF33D0, 0x7FFF5ADF33E0)		 = 0 0
sigprocmask(0x1, 0x0, 0x7FFF5ADF33E0)		 = 0x0 0
sigaltstack(0x0, 0x7FFF5ADF33D0, 0x7FFF5ADF33E0)		 = 0 0
sigprocmask(0x1, 0x0, 0x7FFF5ADF33E0)		 = 0x0 0
sigaltstack(0x0, 0x7FFF5ADF33D0, 0x7FFF5ADF33E0)		 = 0 0
sigaction(0x2, 0x7FFF5ADF2E98, 0x7FFF5ADF2EC0)		 = 0 0
sigaction(0x3, 0x7FFF5ADF2E98, 0x7FFF5ADF2EC0)		 = 0 0
sigaction(0x14, 0x7FFF5ADF2E98, 0x7FFF5ADF2EC0)		 = 0 0

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$

ilpincy, 2015-08-01:
It's a linking error - not a problem of ARGoS. It is almost surely related
to the non-standard paths of MacPorts.

Try to inspect the output of this command. If it shows a library not found,
you have the cause of the problem.

$ ls -1 /usr/local/lib/argos3/*.dylib | xargs -n 1 otool -L

dbl001, 2015-08-01:
ls -1 /usr/local/lib/argos3/*.dylib | xargs -n 1 otool -L
/usr/local/lib/argos3/libargos3core_simulator.dylib:
	@rpath/libargos3core_simulator.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
	/opt/local/lib/libgsl.0.dylib (compatibility version 18.0.0, current version 18.0.0)
	/opt/local/lib/libgslcblas.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/local/lib/libfreeimage.3.dylib (compatibility version 3.0.0, current version 3.16.0)
	/opt/local/lib/libfreeimageplus.3.dylib (compatibility version 3.0.0, current version 3.16.0)
	/opt/local/lib/liblua.dylib (compatibility version 5.2.0, current version 5.2.3)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.0.0)
/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.dylib:
	@rpath/libargos3plugin_simulator_dynamics2d.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3core_simulator.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
	/opt/local/lib/libgsl.0.dylib (compatibility version 18.0.0, current version 18.0.0)
	/opt/local/lib/libgslcblas.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/local/lib/libfreeimage.3.dylib (compatibility version 3.0.0, current version 3.16.0)
	/opt/local/lib/libfreeimageplus.3.dylib (compatibility version 3.0.0, current version 3.16.0)
	/opt/local/lib/liblua.dylib (compatibility version 5.2.0, current version 5.2.3)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.0.0)
/usr/local/lib/argos3/libargos3plugin_simulator_entities.dylib:
	@rpath/libargos3plugin_simulator_entities.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_media.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3core_simulator.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
	/opt/local/lib/libgsl.0.dylib (compatibility version 18.0.0, current version 18.0.0)
	/opt/local/lib/libgslcblas.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/local/lib/libfreeimage.3.dylib (compatibility version 3.0.0, current version 3.16.0)
	/opt/local/lib/libfreeimageplus.3.dylib (compatibility version 3.0.0, current version 3.16.0)
	/opt/local/lib/liblua.dylib (compatibility version 5.2.0, current version 5.2.3)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.0.0)
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib:
	@rpath/libargos3plugin_simulator_epuck.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_genericrobot.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_dynamics2d.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_qtopengl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_entities.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_media.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3core_simulator.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
	/opt/local/lib/libgsl.0.dylib (compatibility version 18.0.0, current version 18.0.0)
	/opt/local/lib/libgslcblas.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/local/lib/libfreeimage.3.dylib (compatibility version 3.0.0, current version 3.16.0)
	/opt/local/lib/libfreeimageplus.3.dylib (compatibility version 3.0.0, current version 3.16.0)
	/opt/local/lib/liblua.dylib (compatibility version 5.2.0, current version 5.2.3)
	@rpath/./libQtOpenGL.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtGui.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtCore.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	/System/Library/Frameworks/GLUT.framework/Versions/A/GLUT (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 21.0.0)
	/System/Library/Frameworks/AGL.framework/Versions/A/AGL (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.0.0)
/usr/local/lib/argos3/libargos3plugin_simulator_eyebot.dylib:
	@rpath/libargos3plugin_simulator_eyebot.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_genericrobot.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_pointmass3d.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_qtopengl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_entities.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3plugin_simulator_media.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libargos3core_simulator.dylib (compatibility version 0.0.0, current version 0.0.0)
[... cut ...]
	/System/Library/Frameworks/GLUT.framework/Versions/A/GLUT (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 21.0.0)
	/System/Library/Frameworks/AGL.framework/Versions/A/AGL (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.0.0)
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$

ilpincy, 2015-08-01:
I'm adding a little more output to ARGoS to make the error message more
informative. Hold on.

ilpincy, 2015-08-01:
OK, I have pushed the new version. Do the following:

$ cd /path/to/argos3/build
$ git pull
$ make
$ sudo make install

Then try again:

$ cd /path/to/argos3-examples
$ argos3 -c experiments/diffusion_1.argos

This should print more informative messages on why the linking failed.

dbl001, 2015-08-01:
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ argos3 -c experiments/diffusion_1.argos
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.dylib"
[FATAL] Can't load library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib" even after trying to add extensions for shared library (dylib) and module library (so): 
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib: dlopen(/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib, 1): Library not loaded: @rpath/./libQtOpenGL.4.dylib
  Referenced from: /usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib
  Reason: image not found
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.dylib: dlopen(/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.dylib, 1): image not found
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.so: OK

dbl001, 2015-08-01:
> davidl@softintel.com <mailto:davidl@softintel.com>>

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ ls -l /usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib
-rwxr-xr-x  1 root  wheel  501964 Aug  1 12:23 /usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ file !$
file /usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib: Mach-O 64-bit dynamically linked shared library x86_64
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$

ilpincy, 2015-08-01:
So, the error seems to be:

/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib:
dlopen(/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib, 1):
Library not loaded: @rpath/./libQtOpenGL.4.dylib
Referenced from: /usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib
Reason: image not found

In other words, it can't find libQtOpenGL.4.dylib. Where is this installed?
You obviously have it, because otherwise the compilation configuration
would have skipped using it (it's checked). A possible solution is setting
the variable DYLD_LIBRARY_PATH to point to where that library is located.

$ export DYLD_LIBRARY_PATH=/path/to/qt/libs

Make sure you launch ARGoS in the same Bash shell as the one you executed
the above command.

dbl001, 2015-08-01:
Hmmm … It’s in Anaconda … but things are getting weirder (down the ‘rabbit hole’):

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ sudo find /usr/local -name libQtOpenGL.4.dylib -ls
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ sudo find /opt/local -name libQtOpenGL.4.dylib -ls
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ find ~ -name libQtOpenGL.4.dylib -ls
115484669        8 lrwxr-xr-x    1 davidlaxer       staff                  23 Jun  1 07:02 /Users/davidlaxer/anaconda/lib/libQtOpenGL.4.dylib -> libQtOpenGL.4.8.6.dylib
107376094        8 lrwxr-xr-x    1 davidlaxer       staff                  23 Mar  4 16:25 /Users/davidlaxer/anaconda/pkgs/qt-4.8.6-0/lib/libQtOpenGL.4.dylib -> libQtOpenGL.4.8.6.dylib
115473940        8 lrwxr-xr-x    1 davidlaxer       staff                  23 Jun  1 07:02 /Users/davidlaxer/anaconda/pkgs/qt-4.8.6-3/lib/libQtOpenGL.4.dylib -> libQtOpenGL.4.8.6.dylib

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ export DYLD_LIBRARY_PATH=/Users/davidlaxer/anaconda/lib/
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ argos3 -c experiments/diffusion_1.argos
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.dylib"
[FATAL] Can't load library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib" even after trying to add extensions for shared library (dylib) and module library (so): 
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib: dlopen(/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib, 1): Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.dylib: dlopen(/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.dylib, 1): image not found
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.so: OK
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ 

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ sudo find /opt/local -name libsqlite3.dylib -ls
Password:
119817574        8 lrwxr-xr-x    1 root             admin                  18 Jun 23 08:43 /opt/local/lib/libsqlite3.dylib -> libsqlite3.0.dylib
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ export DYLD_LIBRARY_PATH=/Users/davidlaxer/anaconda/lib/:/opt/local/lib/:
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ env | grep DYLD
DYLD_LIBRARY_PATH=/Users/davidlaxer/anaconda/lib/:/opt/local/lib/:
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ argos3 -c experiments/diffusion_1.argos
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.dylib"
[FATAL] Can't load library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib" even after trying to add extensions for shared library (dylib) and module library (so): 
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib: dlopen(/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib, 1): Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.dylib: dlopen(/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.dylib, 1): image not found
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.so: OK

ilpincy, 2015-08-01:
Try this command:

$ ls -1 /usr/local/lib/argos3/*.dylib | xargs -n 1 otool -L | grep -i qt

It tells where Qt is expected to be found after the compilation process.
Try to set the environment variable DYLD_LIBRARY_PATH to the directory
mentioned in that output.

dbl001, 2015-08-01:
What does it mean that  libQtOpenGL.4.dylib was only found in Anaconda?  Then the Anaconda library had linkages to other packages which couldn’t be resolved.
How is this dependency checked?  In ‘cmake’?
Do I need to build QTOpenGL manually and install it?

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ find ~ -name libQtOpenGL.4.dylib -ls
115484669        8 lrwxr-xr-x    1 davidlaxer       staff                  23 Jun  1 07:02 /Users/davidlaxer/anaconda/lib/libQtOpenGL.4.dylib -> libQtOpenGL.4.8.6.dylib
107376094        8 lrwxr-xr-x    1 davidlaxer       staff                  23 Mar  4 16:25 /Users/davidlaxer/anaconda/pkgs/qt-4.8.6-0/lib/libQtOpenGL.4.dylib -> libQtOpenGL.4.8.6.dylib
115473940        8 lrwxr-xr-x    1 davidlaxer       staff                  23 Jun  1 07:02 /Users/davidlaxer/anaconda/pkgs/qt-4.8.6-3/lib/libQtOpenGL.4.dylib -> libQtOpenGL.4.8.6.dylib

David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ argos3 -c experiments/diffusion_1.argos
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.dylib"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.dylib"
[FATAL] Can't load library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib" even after trying to add extensions for shared library (dylib) and module library (so): 
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib: dlopen(/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib, 1): Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.dylib: dlopen(/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.dylib, 1): image not found
/usr/local/lib/argos3/libargos3plugin_simulator_epuck.dylib.so: OK
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ s -1 /usr/local/lib/argos3/*.dylib | xargs -n 1 otool -L | grep -i qt
-bash: s: command not found
David-Laxers-MacBook-Pro:argos3-examples davidlaxer$ l!!
ls -1 /usr/local/lib/argos3/*.dylib | xargs -n 1 otool -L | grep -i qt
	@rpath/libargos3plugin_simulator_qtopengl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/./libQtOpenGL.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtGui.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtCore.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/libargos3plugin_simulator_qtopengl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/./libQtOpenGL.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtGui.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtCore.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/libargos3plugin_simulator_qtopengl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/./libQtOpenGL.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtGui.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtCore.4.dylib (compatibility version 4.8.0, current version 4.8.6)
/usr/local/lib/argos3/libargos3plugin_simulator_qtopengl.dylib:
	@rpath/libargos3plugin_simulator_qtopengl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/./libQtOpenGL.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtGui.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtCore.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/libargos3plugin_simulator_qtopengl.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/./libQtOpenGL.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtGui.4.dylib (compatibility version 4.8.0, current version 4.8.6)
	@rpath/./libQtCore.4.dylib (compatibility version 4.8.0, current version 4.8.6)

ilpincy, 2015-08-01:
The problem is indeed how Qt is installed. How did you install it? I
would definitely avoid installing it by hand. MacPorts has a native
way to do it.

ARGoS checks dependencies in CMake. The file CMakeCache.txt in the
build/ directory tells you where QT was found. You can open it, it's a
text file.

dbl001, 2015-08-01:
There is some QT4 dependency with Anaconda that I don’t understand (and probably just made worse) ...

David-Laxers-MacBook-Pro:build_simulator davidlaxer$ sudo port install qt4-mac

--->  Computing dependencies for qt4-mac
--->  Fetching archive for qt4-mac
--->  Attempting to fetch qt4-mac-4.8.7_0.darwin_14.x86_64.tbz2 from http://packages.macports.org/qt4-mac
--->  Attempting to fetch qt4-mac-4.8.7_0.darwin_14.x86_64.tbz2.rmd160 from http://packages.macports.org/qt4-mac
--->  Installing qt4-mac @4.8.7_0
--->  Activating qt4-mac @4.8.7_0
NOTE: Qt database plugins for mysql55, postgresql91, and sqlite2 are NOT installed by this port; they are installed by qt4-mac-*-plugin instead.
--->  Cleaning qt4-mac
--->  Updating database of binaries
--->  Scanning binaries for linking errors               
--->  No broken files found.                             

David-Laxers-MacBook-Pro:build_simulator davidlaxer$ cmake -DCMAKE_BUILD_TYPE=Release ../src
-- GCC/G++ version >= 4.2.1
-- GSL using gsl-config /opt/local/bin/gsl-config
-- Using GSL from /opt/local
dyld: Symbol not found: _CGLGetCurrentContext
  Referenced from: /System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo
  Expected in: /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
 in /System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/davidlaxer/argos3/build_simulator
David-Laxers-MacBook-Pro:build_simulator davidlaxer$ rm CMakeCache.txt 
David-Laxers-MacBook-Pro:build_simulator davidlaxer$ cmake -DCMAKE_BUILD_TYPE=Release ../src
-- GCC/G++ version >= 4.2.1
-- Found DLFCN: /usr/lib/libdl.dylib  
-- Found Pthreads: /usr/lib/libpthread.dylib  
-- GSL using gsl-config /opt/local/bin/gsl-config
-- Using GSL from /opt/local
[... 1385 lines cut ...]
> >> > > > >> > > > > > sigaction(0x3, 0x7FFF5ADF33E8, 0x7FFF5ADF3418) = 0 0
> >> > > > >> > > > > > sysctl(0x7FFF5ADF3418, 0x2, 0x7FFF5ADF3310) = 0 0
> >> > > > >> > > > > > stat64("/Users/davidlaxer/argos3-examples\0",
> >> > 0x7FFF5ADF32D8,
> >> > > > >> > > > > > 0x7FFF5ADF3310) = 0 0
> >> > > > >> > > > > > stat64(".\0", 0x7FFF5ADF3248, 0x7FFF5ADF3310) = 0 0
> >> > > > >> > > > > > getppid(0x104E80527, 0x7FFF5ADF3248, 0x7FFF5ADF3310) =
> >> > 86816 0
> >> > > > >> > > > > > getpgrp(0x104E80527, 0x7FFF5ADF3248, 0x7FFF5ADF3310) =
> >> > 86813 0
> >> > > > >> > > > > > sigaction(0x14, 0x7FFF5ADF33D8, 0x7FFF5ADF3400) = 0 0
> >> > > > >> > > > > > getrlimit(0x1007, 0x7FFF5ADF33D0, 0x7FFF5ADF3400) = 0 0
> >> > > > >> > > > > > sigprocmask(0x1, 0x0, 0x7FFF5ADF3430) = 0x0 0
> >> > > > >> > > > > > sigaltstack(0x0, 0x7FFF5ADF3420, 0x7FFF5ADF3430) = 0 0
> >> > > > >> > > > > > open("/usr/local/bin/argos3\0", 0x0, 0x0) = 3 0
> >> > > > >> > > > > > ioctl(0x3, 0x4004667A, 0x7FFF5ADF33DC) = -1 Err#25
> >> > > > >> > > > > > ioctl(0x3, 0x40487413, 0x7FFF5ADF33E0) = -1 Err#25
> >> > > > >> > > > > > lseek(0x3, 0x0, 0x1) = 0 0
> >> > > > >> > > > > > read(0x3, "#!/bin/bash\n\n# Add path to ARGoS3
> >> > > > >> > > > > > libraries
> >> > to
> >> > > > the
> >> > > > >> > > > > > DYLD_LIBRARY_PATH env variabl@\004\b\0", 0x50) = 80 0
> >> > > > >> > > > > > lseek(0x3, 0x0, 0x0) = 0 0
> >> > > > >> > > > > > getdtablesize(0x3, 0

ilpincy, 2015-08-01:
I would stay away from Anaconda. It's a different system that ships
its own dependencies.
Just install QT4 through Macports and recompile ARGoS.

On Sat, Aug 1, 2015 at 6:12 PM, dbl001 <notifications@github.com> wrote:
> There is some QT4 dependency with Anaconda that I don’t understand (and
> probably just made worse) ...
>
> David-Laxers-MacBook-Pro:build_simulator davidlaxer$ sudo port install
> qt4-mac
>
> ---> Computing [...]

dbl001, 2015-08-01:
I tried that … which led to these errors:

David-Laxers-MacBook-Pro:build_simulator davidlaxer$ sudo port install qt4-mac
Password:
--->  Computing dependencies for qt4-mac
--->  Cleaning qt4-mac
--->  Scanning binaries for linking errors

David-Laxers-MacBook-Pro:argos3 davidlaxer$ rm -rf build_simulator/
David-Laxers-MacBook-Pro:argos3 davidlaxer$ mkdir build_simulator
David-Laxers-MacBook-Pro:argos3 davidlaxer$ cd !$
cd build_simulator
David-Laxers-MacBook-Pro:build_simulator davidlaxer$ cmake -DCMAKE_BUILD_TYPE=Release ../src
-- The C compiler identification is AppleClang 6.1.0.6020053
-- The CXX compiler identification is AppleClang 6.1.0.6020053
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- GCC/G++ version >= 4.2.1
-- Found DLFCN: /usr/lib/libdl.dylib  
-- Found Pthreads: /usr/lib/libpthread.dylib  
-- GSL using gsl-config /opt/local/bin/gsl-config
-- Using GSL from /opt/local
-- Found GooglePerfTools: /usr/local/lib/libtcmalloc.dylib  
dyld: Symbol not found: _CGLGetCurrentContext
  Referenced from: /System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo
  Expected in: /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
 in /System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo
-- Found Doxygen: /opt/local/bin/doxygen  
-- Found ASCIIDoc: /opt/local/bin/asciidoc  
-- Found Lua52: /opt/local/lib/liblua.dylib  
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
dyld: Symbol not found: _sqlite3_intarray_bind
  Referenced from: /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
  Expected in: /Users/davidlaxer/anaconda/lib//libsqlite3.dylib
 in /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData
-- Found unsuitable Qt version "" from NOTFOUND
-- Qt4 not found, skipping compilation of QT-OpenGL visualization.
[... 1379 lines cut ...]
> >> >> > > > >> > > > 0 0
> >> >> > > > >> > > > > > stat64("/usr/lib/system/libsystem_blocks.dylib\0",
> >> >> > > > 0x7FFF5ADF16D8,
> >> >> > > > >> > > > 0x1) =
> >> >> > > > >> > > > > > 0 0
> >> >> > > > >> > > > > > stat64("/usr/lib/system/libsystem_c.dylib\0",
> >> >> > 0x7FFF5ADF16D8,
> >> >> > > > 0x1)
> >> >> > > > >> > = 0
> >> >> > > > >> > > > 0
> >> >> > > > >> > > > > >
> >> >> > > > >> > > > > >
> >> >> > > > >> > > > > > stat64("/usr/lib/system/libsystem_configuration.dylib\0",
> >> >> > > > >> > > > 0x7FFF5ADF16D8,
> >> >> > > > >> > > > > > 0x1) = 0 0
> >> >> > > > >> > > > > >
> >> >> > > > >> > > > > >
> >> >> > > > >> > > > > > stat64("/usr/lib/system/libsystem_coreservices.dylib\0",
> >> >> > > > >> > > > 0x7FFF5ADF16D8,
> >> >> > > > >> > > > > > 0x1) = 0 0
> >> >> > > > >> > > > > > stat64("/usr/lib/system/libsystem_coretls.dylib\0",
> >> >> > > > 0x7FFF5ADF16D8,
> >> >> > > > >> > > > 0x1) =
> >> >> > > > >> > > > > > 0 0
> >> >> > > > >> > > > > > stat64("/usr/lib/system/libsy

ilpincy, 2015-08-02:
It looks like Anaconda is interfering with compilation. Can you
temporarily uninstall it?

On Sat, Aug 1, 2015 at 7:47 PM, dbl001 <notifications@github.com> wrote:
> I tried that … which led to these errors:
>
> David-Laxers-MacBook-Pro:build_simulator davidlaxer$ sudo port install
> qt4-mac
> Password:
> ---> Computing dependencies for qt4-mac
> ---> Cleaning qt4-mac
> ---> [...]

dbl001, 2015-08-02:
Now Anaconda is broken:
E.g. - ipython qtconsole

I sent off a question to Anaconda support @Continuum.  
Stay tuned.

David-Laxers-MacBook-Pro:build_simulator davidlaxer$ sudo port upgrade outdated
Nothing to upgrade.
David-Laxers-MacBook-Pro:build_simulator davidlaxer$ conda update conda
Fetching package metadata: ....
Solving package specifications: .
Package plan for installation in environment /Users/davidlaxer/anaconda:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    conda-3.15.1               |           py27_0         171 KB

The following packages will be UPDATED:

    conda: 3.14.1-py27_0 --> 3.15.1-py27_0

Proceed ([y]/n)? y

Fetching packages ...
conda-3.15.1-p 100% |################################| Time: 0:00:00 389.64 kB/s
Extracting packages ...
[      COMPLETE      ]|###################################################| 100%
Unlinking packages ...
[      COMPLETE      ]|###################################################| 100%
Linking packages ...
[      COMPLETE      ]|###################################################| 100%
David-Laxers-MacBook-Pro:build_simulator davidlaxer$ conda update anaconda
Fetching package metadata: ....
# All requested packages already installed.
# packages in environment at /Users/davidlaxer/anaconda:
#
anaconda                  2.3.0                np19py27_0  
David-Laxers-MacBook-Pro:build_simulator davidlaxer$ ipython qtconsole
Traceback (most recent call last):
  File "/Users/davidlaxer/anaconda/bin/ipython", line 6, in <module>
    sys.exit(start_ipython())
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/__init__.py", line 120, in start_ipython
    return launch_new_instance(argv=argv, **kwargs)
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/config/application.py", line 573, in launch_instance
    app.initialize(argv)
  File "<string>", line 2, in initialize
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/config/application.py", line 75, in catch_config_error
    return method(app, *args, **kwargs)
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/terminal/ipapp.py", line 321, in initialize
    super(TerminalIPythonApp, self).initialize(argv)
  File "<string>", line 2, in initialize
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/config/application.py", line 75, in catch_config_error
    return method(app, *args, **kwargs)
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/core/application.py", line 369, in initialize
    self.parse_command_line(argv)
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/terminal/ipapp.py", line 316, in parse_command_line
    return super(TerminalIPythonApp, self).parse_command_line(argv)
  File "<string>", line 2, in parse_command_line
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/config/application.py", line 75, in catch_config_error
    return method(app, *args, **kwargs)
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/config/application.py", line 471, in parse_command_line
    return self.initialize_subcommand(subc, subargv)
  File "<string>", line 2, in initialize_subcommand
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/config/application.py", line 75, in catch_config_error
    return method(app, *args, **kwargs)
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/config/application.py", line 402, in initialize_subcommand
    subapp = import_item(subapp)
  File "/Users/davidlaxer/anaconda/lib/python2.7/site-packages/IPython/utils/importstring.py", line 42, in import_item
[... 1337 lines cut ...]
> >> >> >> > > > version
> >> >> >> > > > >> > 0.0.0,
> >> >> >> > > > >> > > > current version 0.0.0)
> >> >> >> > > > >> > > > @rpath/libargos3plugin_simulator_genericrobot.dylib
> >> >> >> > (compatibility
> >> >> >> > > > >> > version
> >> >> >> > > > >> > > > 0.0.0, current version 0.0.0)
> >> >> >> > > > >> > > > @rpath/libargos3plugin_simulator_pointmass3d.dylib
> >> >> >> > (compatibility
> >> >> >> > > > >> > version
> >> >> >> > > > >> > > > 0.0.0, current version 0.0.0)
> >> >> >> > > > >> > > > @rpath/libargos3plugin_simulator_qtopengl.dylib
> >> >> >> > > > >> > > > (compatibility
> >> >> >> > > > version
> >> >> >> > > > >> > > > 0.0.0, current version 0.0.0)
> >> >> >> > > > >> > > > @rpath/libargos3plugin_simulator_entities.dylib
> >> >> >> > > > >> > > > (compatibility
> >> >> >> > > > version
> >> >> >> > > > >> > > > 0.0.0, current version 0.0.0)
> >> >> >> > > > >> > > > @rpath/libargos3plugin_simulator_media.dylib
> >> >> >> > > > >> > > > (compatibility
> >> >> >> > > > version
> >> >> >> > > > >> > 0.0.0,
> >> >> >> > > > >> > > > current version 0.0.0)
> >> >> >> > > > >> > > >

commit 14eb41da (2015-08-01): QTOpenGL visualization compiles correctly when FreeImage is not found.

Fixes #12