ilpincy/argos3 #13: Could NOT find Lua52 (missing:  LUA_LIBRARIES LUA_INCLUDE_DIR

dbl001, 2015-08-04:
Hi,

I'm running on Ubuntu 14.04.2 on an EC2 instance.

ubuntu@ip-10-0-1-107:~/argos3/build$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.2 LTS
Release:	14.04
Codename:	trusty
ubuntu@ip-10-0-1-107:~/argos3/build$ 

ubuntu@ip-10-0-1-107:~/argos3/build$ lua -v
Lua 5.2.3  Copyright (C) 1994-2013 Lua.org, PUC-Rio
ubuntu@ip-10-0-1-107:~/argos3/build$ which lua
/usr/bin/lua

ubuntu@ip-10-0-1-107:~/argos3/build$ whereis lua
lua: /usr/bin/lua5.2 /usr/bin/lua /usr/bin/lua5.1 /usr/bin/X11/lua5.2 /usr/bin/X11/lua /usr/bin/X11/lua5.1 /usr/share/man/man1/lua.1.gz

Any ideas?

ilpincy, 2015-08-04:
Delete lua5.1. It conflicts with Lua5.2.

ilpincy, 2015-08-04:
Also: if you're running Ubuntu, it is highly recommended to install ARGoS
through the package at http://argos-sim.info/core.php.

closed 2015-08-04 by ilpincy

dbl001, 2015-08-04:
Same problem:

-- Could NOT find Lua52 (missing:  LUA_LIBRARIES LUA_INCLUDE_DIR) 

ubuntu@ip-10-0-1-107:~/argos3/build$ sudo apt-get remove lua5.1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-3.13.0-49 linux-headers-3.13.0-49-generic
  linux-image-3.13.0-49-generic linux-image-extra-3.13.0-49-generic
Use 'apt-get autoremove' to remove them.
The following packages will be REMOVED:
  lua5.1
0 upgraded, 0 newly installed, 1 to remove and 277 not upgraded.
After this operation, 337 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 298796 files and directories currently installed.)
Removing lua5.1 (5.1.5-5ubuntu0.1) ...
Processing triggers for man-db (2.6.7.1-1ubuntu1) ...
ubuntu@ip-10-0-1-107:~/argos3/build$ lua -v
Lua 5.2.3  Copyright (C) 1994-2013 Lua.org, PUC-Rio
ubuntu@ip-10-0-1-107:~/argos3/build$ ls
argos3                 CMakeFiles               Doxyfile_standalone
argos3.conf            cmake_install.cmake      Makefile
argos3.desktop         core                     plugins
argos3_simulator.pc    CPackConfig.cmake        postinst
argos_post_install.sh  CPackSourceConfig.cmake  setup_env.sh
argos.SlackBuild       doinst.sh                slack-desc
CMakeCache.txt         Doxyfile_embedded        testing
ubuntu@ip-10-0-1-107:~/argos3/build$ rm CMakeCache.txt 
ubuntu@ip-10-0-1-107:~/argos3/build$ cmake -DCMAKE_BUILD_TYPE=Release ../src
-- GCC/G++ version >= 4.3
-- Found DLFCN: /usr/lib/x86_64-linux-gnu/libdl.so  
-- Found Pthreads: /usr/lib/x86_64-linux-gnu/libpthread.so  
-- GSL using gsl-config /usr/bin/gsl-config
-- Using GSL from /usr
-- Could NOT find GooglePerfTools (missing:  GOOGLEPERFTOOLS_LIBRARY GOOGLEPERFTOOLS_INCLUDE_DIR) 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6") 
-- Found ASCIIDoc: /usr/bin/asciidoc  
-- Could NOT find Lua52 (missing:  LUA_LIBRARIES LUA_INCLUDE_DIR) 
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /home/ubuntu/anaconda/bin/qmake (found version "4.8.6") 
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found GLUT: /usr/lib/x86_64-linux-gnu/libglut.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/argos3/build
ubuntu@ip-10-0-1-107:~/argos3/build$

dbl001, 2015-08-04:
Update:

I had to install lua5.2-dev.

ilpincy, 2015-08-04:
Why don't you use the precompiled package?

dbl001, 2015-08-04:
I did and had the same problem with lua52.
After installing lua52-dev with apt-get, but the pre-compiled package as well as the source worked.

dbl001, 2015-08-04:
btw - ARGoS runs with the Anaconda stack on Ubuntu 14.04.

-- Found Qt4: /home/ubuntu/anaconda/bin/qmake (found version "4.8.6")

dbl001, 2015-08-04:
> btw - ARGoS runs with the Anaconda stack on Ubuntu 14.04.2

I’m currently running Ubuntu 14.04.2 LTS on an EC2 M3.Large server.
Can ARGoS and/or Buzz utilize a relatively older Kepler GPU?

ilpincy, 2015-08-04:
Consider that Lua is optional for ARGoS, so that should not prevent you
from using it.
I'm puzzled by all these problems you're encountering. I have installed
ARGoS on hundreds of computers without issues. Probably if I had access to
the computer I'd be able to spot the issue. With the information you're
providing me, I don't know how to reproduce the problem. My virtual
machines work perfectly.

ilpincy, 2015-08-04:
Yes, I run it on old computers and it works reasonably.