ilpincy/argos3 #14: Black area and no GUI icons

pawel-jakubowski, 2015-11-29:
Hi,

I compiled simulator under Ubuntu 15.10 64-bit and I get black floor in simulation. What is more there is no icons for controls. There is any way to fix it (maybe I forgot to set some variable during compilation?).

It looks like this:
[image]

ilpincy, 2015-11-29:
You probably have the wrong paths in $HOME/.config/Iridia-ULB/ARGoS.conf (see also https://github.com/ilpincy/argos3#running-the-argos-simulator). Either fix the paths (/usr/local instead of /usr, I imagine), or simply erase the file and ARGoS will create a new one automatically.

pawel-jakubowski, 2015-11-29:
That was it, thanks!

closed 2015-11-29 by pawel-jakubowski