ilpincy/argos3 #15: Symbol "qtopengl_render" not found

pawel-jakubowski, 2016-01-05:
Help of argos3 says that I should use:
```
<visualization>
    <qtopengl_render>
      <user_functions library="/path/to/libmyuserfunctions.so"
                      label="my_user_functions" />
    </qtopengl_render>
 </visualization>
```

However when I try this I obtain:
```
[ 75%] Built target argos3plugin_simulator_flocking
[100%] Built target flocking
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_qtopengl.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "[0[FATAL] Failed to initialize the visualization. Parse error in the <visualization> subtree.
[FATAL] Symbol "qtopengl_render" not found
m/usr/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Loaded library "/home/pjakubow/workspaces/Swarm/Flocking/Debug/robots/flocking/libargos3plugin_simulator_flocking.so"
[INFO] Not using threads
[INFO] Using random seed = 911832
[INFO] Using simulation clock tick = 0.0166667
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./controllers/flocking/libflocking.so"
[INFO] The physics engine "dyn2d" will perform 10 iterations per tick (dt = 0.00166667 sec)
[100%] Built target start
```

When I change qtopengl_render to qt-opengl everything works fine. Probably it is only help page problem.

allsey87, 2016-01-05:
As you suggested, it is just an issue on the help page. The correct symbol is "qt-opengl" as per: src/plugins/simulator/visualizations/qt-opengl/qtopengl_render.cpp#L116

ilpincy, 2016-01-05:
Thanks for spotting the problem! :-)

commit 81dd50ae (2016-01-05): Fixes #15
closed 2016-01-05 by ilpincy