ilpincy/argos3 #17: Segmentation fault after "./argos3 -q all"

EMerckx, 2016-03-02:
Hello ilpincy, 

I have noticed an issue while executing the following commands.

    $ cd argos3
    $ cd build_simulator
    $ . setup_env.sh     # or 'source setup_env.sh'
    $ cd core
    $ ./argos3 -q all    # this shows all the plugins recognized by ARGoS

When executing the last command, all the actuators, sensors, entities, ... are shown. But after that, there is a segmentation fault and the core gets dumped.

    Segmentation fault (core dumped)

ilpincy, 2016-03-02:
Hi Ewout,

Do you have ARGoS installed system wide? If so, you can't execute ARGoS
from the sources.

If not, why are you executing ARGoS from the sources? Why not installing it
system wide?

Final question: on which operating system are you working?

Cheers,
Carlo

EMerckx, 2016-03-02:
Hello Carlo,

I had installed ARGoS system wide, but I've removed it. I managed to run experiments with ARGoS from the sources, with the only exception that the core gets dumped after execution.

For my Master's thesis, I need to implement some new entities. In order to do that, I need to use ARGoS from the sources.

I'm currently working on Elementary OS (which is based on Ubuntu 14.04 LTS).

Thanks,
Ewout Merckx

ilpincy, 2016-03-02:
On Wed, Mar 2, 2016 at 9:06 AM, Ewout Merckx <notifications@github.com>
wrote:

> I had installed ARGoS system wide, but I've removed it.
>
Are you sure? What's the output of the ARGoS log when you launch it?

> For my Master's thesis, I need to implement some new entities. In order to
> do that, I need to use ARGoS from the sources.
>
Not necessarily, see the development manual that I am writing:
http://www.argos-sim.info/dev_manual.php.
-- 
Carlo Pinciroli, Ph.D.
http://carlo.pinciroli.net/

EMerckx, 2016-03-02:
I got the following output:

    [INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.so"
    [INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.so"
    [INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.so"
    ...

Apparently, the lib files in the usr directory were still present. I removed them from my system and now I don't get this issue anymore.

Thank you very much for your help and I will certainly read through the developer manual.

Thanks again,
sincerely yours,
Ewout Merckx

closed 2016-03-02 by EMerckx

ilpincy, 2016-03-02:
Awesome! :-)