ilpincy/argos3 #22: Eye-bot proximity sensor and box bug

EMerckx, 2016-05-17:
Dear @ilpincy,

I have noticed a bug while simulating an experiment with an eye-bot. In this experiment, I have placed an eye-bot (containing a proximity sensor) in front of a box entity. 

[image]

As you can see in the picture, some rays of the proximity sensor that aren't intersecting with the entity, are marked in magenta. But the correct color of those rays should be cyan, which means: not intersecting with an entity.

I think this issue is caused by the [mathematical model of the box](
https://github.com/ilpincy/argos3/blob/master/src/core/utility/math/box.cpp). More precisely, I think there is a check missing in the intersection method: if the intersection point is behind the start point of the ray, no intersection occurs.

Does this seem like a possible solution for the problem to you?

Sincerely yours,
Ewout Merckx

ilpincy, 2016-05-17:
Hi Ewout,

Thanks for spotting this bug! Yes, it looks like the check is missing. Are
you able to come up with a patch that fixes this issue? If so, thanks! :-)
Otherwise I'll work on it during the week.

Cheers,
Carlo

EMerckx, 2016-05-17:
I've had a quick look and the following line seems to have done the trick. It tells us that there is no intersection when the distance is negative (if I understand the code correctly).

```c++
if(fTMin < 0.0f) return false;
```

This can be inserted into [this line](https://github.com/ilpincy/argos3/blob/master/src/core/utility/math/box.cpp#L41). If it seems like a good solution to you, I'll generate a pull request for it.

ilpincy, 2016-05-17:
I'm not sure without testing :-) Please try it, and if it works make a pull
request. Thank you very much! :-)

EMerckx, 2016-05-17:
The fix that I've mentioned above worked only when the eye-bot was in front of the box. I've implemented and tested another approach, which seems to work in all cases. You can view the code below.

```c++
/* Direction of the ray */
CVector3 cRayDir;
c_ray.GetDirection(cRayDir);
/* Vector between the ray origin and box center */
CVector3 cRayOrig2BoxCenter = cBoxCenter - cRayStart;
/* Scalar distance between the ray origin and point in box
 * No intersection if the ray intersects in the opposite direction */
if( cRayOrig2BoxCenter.DotProduct(cRayDir) < 0.0f ){
    return false;
}
```

However, I've noticed that there are still some other bugs in the intersection method of the box. It would be great if you could take a look at it, since I don't fully understand the code. In the picture below, you can see an example of a bug.

[image]

To make testing easier, I've attached the argos configuration file for the specific experiment to this comment. Here, the controller of the argos-circle experiment is used.
[eyebot_box.zip](https://github.com/ilpincy/argos3/files/269190/eyebot_box.zip)

ilpincy, 2017-04-09:
Fixed in commit e81c7536db58e166102d903f62f298b6ef789d91

closed 2017-04-09 by ilpincy