ilpincy/argos3 #25: Camera angled at 90 degrees

rubenvereecken, 2016-05-25:
Strangely, a custom-defined camera is angled at 90 degrees using the following specific setting:

`        <placement idx="1" position="0,-5, 1" look_at="0,0,0" lens_focal_length="20" />`

as can be seen on this image:

[image]

I can fix it by having a non-zero `x` position, like so:

`        <placement idx="1" position="0.00001,-5, 1" look_at="0,0,0" lens_focal_length="20" />`

ilpincy, 2016-05-25:
Thanks for the bug report! At the end of the week I'll work on this. I'll
add the possibility to specify the Up vector of the camera manually, for
situations in which the automatic calculation is ambiguous (like this one).

Cheers,
Carlo

rubenvereecken, 2016-05-25:
Great! Thanks for taking a look so swiftly.

commit 1e029304 (2016-05-27): Fixes #25.

A few amboguities in camera positioning are now resolved. For the
extremely weird cases, a warning is emitted and the up vector can be
specified explicitly.
closed 2016-05-27 by ilpincy