ilpincy/argos3 #31: pointmass3d physics engine causes foot-bot's proximity sensors to detect non-existent obstacles

alanmillard, 2017-01-19:
This is a bit of a contrived issue, but was raised by a student who was trying to use foot-bots and eye-bots simultaneously, so it may cause problems for other people.

Adding the `pointmass3d` physics engine to `diffusion_1.argos`, as is done in `eyebot_circle.argos`, allows the bug to be reproduced:

```
<physics_engines>
    <dynamics2d id="dyn2d" />
    <pointmass3d id="pm3d" iterations="10" />
</physics_engines>
```

Simply adding the `pointmass3d` physics engine to the ARGoS configuration file causes the foot-bot's proximity sensors to detect non-existent obstacles, as shown in the screenshot below:

[image]

The sensors appear to be triggered by existing walls, but from further away than should be possible. It also seems to depend on which side of the wall the robot is on - if the robot approaches from the other side, it will detect the wall correctly.

allsey87, 2017-01-20:
I can't solve this issue right now, but I can point to where the problem is and suggest some quick workarounds until Carlo has time to look at this.

Non-movable objects, such as the boxes you are using for walls are added to all physics engines. Furthermore, the proximity sensor will query all physics engines for an intersection with the proximity sensors ray. The issue is most likely in the maths of bool CBox::Intersects in core/utility/math/box.cpp. This is used by the pointmass3d engine to determine whether a ray intersects the box or not.

As for a temporary workaround, setting the box to movable and just giving it a large mass might work. This should, I think, force ARGoS to place it into the first physics engine available, which would be dynamics2d, assuming the data structures inside ARGoS are initialized in the same order as things are specified in the XML (usually the case).

Note, you may want to remove the overlap from the walls as the collision forces will now come into play since the walls (boxes) are movable.

ilpincy, 2017-01-20:
@allsey87 is most likely right, because this bug is probably related to issue #22.

ilpincy, 2017-04-09:
Finally fixed in commit e81c7536db58e166102d903f62f298b6ef789d91

closed 2017-04-09 by ilpincy