ilpincy/argos3 #33: Symbol "qt-opengl" Not Found -- macOS 10.12.1

bhylak, 2017-02-15:
Just trying to get started with running examples in argos. After installation and compilation, I get the following:

argos3 -c experiments/diffusion_1.argos

[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3core_simulator.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3plugin_simulator_dynamics2d.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3plugin_simulator_entities.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3plugin_simulator_epuck.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3plugin_simulator_eyebot.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3plugin_simulator_footbot.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3plugin_simulator_genericrobot.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3plugin_simulator_media.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3plugin_simulator_pointmass3d.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta43/lib/argos3/libargos3plugin_simulator_spiri.dylib"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./build/controllers/footbot_diffusion/libfootbot_diffusion.so"
[INFO] The physics engine "dyn2d" will perform 10 iterations per tick (dt = 0.01 sec)
**[FATAL] Failed to initialize the visualization. Parse error in the <visualization> subtree.
[FATAL] Symbol "qt-opengl" not found**

ilpincy, 2017-02-15:
It's a known issue, I'm working on it. It's because ARGoS requires Qt4, but
HomeBrew ditched it for Qt5. I need to finish porting the ARGoS
visualization to Qt5. I hope to be done soon with that.

bhylak, 2017-02-15:
Tried installing qt4 via macports to no avail. Any other workaround in the mean time?

ilpincy, 2017-02-15:
Look here:

https://github.com/cartr/homebrew-qt4

jdelgadome, 2017-02-19:
I have the same issue

ilpincy, 2017-02-25:
Beta44 solves this issue!

closed 2017-02-25 by ilpincy

prpandey26, 2017-07-07:
I am also facing the same issue : 
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./build/controllers/footbot_diffusion/libfootbot_diffusion.so"
[INFO] The physics engine "dyn2d" will perform 10 iterations per tick (dt = 0.01 sec)
**[FATAL] Failed to initialize the visualization. Parse error in the <visualization> subtree.
[FATAL] Symbol "qt-opengl" not found**

I am using ubuntu 14.04 system and the Qt version supported by my system is 5.7.1 (>5.5) :
qmake -v
QMake version 3.0
Using Qt version 5.7.1 in /opt/Qt/5.7/gcc_64/lib

But even then while running "cmake ../src" , below mentioned warning is getting generated:
Minimum required version for Qt (>= 5.5) not found. Skipping compilation of QT-OpenGL visualization.

Please guide how to resolve the issue.

cardinot, 2017-07-07:
Hi @prpandey26,

AFAIK, by default, Ubuntu 14.04 supports Qt 5,2.

So, I guess that you probably have some old qt5.2 stuff installed, which is not allowing ARGOS to find qt5.7.

Have you tried to redefine your QTDIR variable and add your '/opt/Qt/5.7/gcc_64/bin' to the PATH variable?
```
export PATH=/opt/Qt/5.7/gcc_64/bin:$PATH
export QTDIR=/opt/Qt/5.7/gcc_64
```

ilpincy, 2017-07-07:
Hello @prpandey26, this issue is about Mac, not Ubuntu - you should have opened a new issue about this. The solution to your problem is to set the variable `CMAKE_PREFIX_PATH` to point to the right `cmake` directory in your system. Look into `/opt/Qt/5.7/gcc_64` and search for that directory. Say the directory is `/opt/Qt/5.7/gcc_64/lib/cmake` (I'm just guessing), the commands to compile ARGoS correctly are:

    cd argos3
    rm -rf build
    mkdir build
    cd build
    export CMAKE_PREFIX_PATH=/opt/Qt/5.7/gcc_64/lib/cmake
    cmake -DCMAKE_BUILD_TYPE=Release ../src
    make -j8
    make doc
    sudo make install

prpandey26, 2017-07-11:
Hi @cardinot  @ilpincy 
Thank you for providing support on the issue.
After making the above mentioned changes, the **Minimum required version for Qt (>= 5.5) not found. Skipping compilation of QT-OpenGL visualization.** issue which was coming while running **cmake** got resolved but another error was coming while running **make -j8.**
So, now I am using a virtual machine with ubuntu 16.04 because in Ubuntu 16.04 Argos got compiled without any error and Argos-Examples are running successfully.

ilpincy, 2017-07-11:
If you're using ARGoS on Ubuntu 16.04, then you could have just used the binary packages at http://argos-sim.info/core.php, which are the suggested way to use ARGoS. Have you compiled the code with `cmake -DCMAKE_BUILD_TYPE=Release` ? If not, ARGoS will be (much) slower. If you have further issues, please open a new issue. Commenting here is definitely off-topic.