ilpincy/argos3 #41: Unimplemented functions

allsey87, 2017-07-20:
The following function prototypes are not implemented and never called.

https://github.com/ilpincy/argos3/blob/3.0.0-beta46/src/plugins/simulator/entities/led_equipped_entity.h#L56
https://github.com/ilpincy/argos3/blob/3.0.0-beta46/src/plugins/simulator/entities/led_equipped_entity.h#L57

ilpincy, 2017-07-20:
Their implementation is here:

https://github.com/ilpincy/argos3/blob/3.0.0-beta46/src/plugins/simulator/entities/led_equipped_entity.cpp#L108

and here:

https://github.com/ilpincy/argos3/blob/3.0.0-beta46/src/plugins/simulator/entities/led_equipped_entity.cpp#L118

allsey87, 2017-07-20:
Look again. Those methods are inside the SActuator struct and are not implemented.

ilpincy, 2017-07-20:
Oh, I see - you're right! They're a remnant of my implementation of `CLEDEquippedEntity::Enable()` and `CLEDEquippedEntity::Disable()`. The unimplemented methods can be safely removed.

closed 2017-11-13 by allsey87