ilpincy/argos3 #42: Problem running an example

Mbt925, 2017-08-08:
I installed ARGoS successfully, but when i try to run one of the ARGoS examples, I get the following error:
```
/Desktop/argos3-examples$ argos3 -c experiments/diffusion_1.argos
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_qtopengl.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./build/controllers/footbot_diffusion/libfootbot_diffusion.so"
[INFO] The physics engine "dyn2d" will perform 10 iterations per tick (dt = 0.01 sec)

[FATAL] Failed to initialize the space.
[FATAL] Failed to initialize entity "fb_0".
[FATAL] Failed to initialize controllable entity "controller_0".
[FATAL] Can't set controller for controllable entity "controller_0"
[FATAL] Symbol "footbot_diffusion_controller" not found
```

ilpincy, 2017-08-08:
1. How did you install ARGoS (binary packages, sources)?
2. On which platform are you working?

Mbt925, 2017-08-08:
Thanks for the reply.

1. binary packages
2. ubuntu 17.04

ilpincy, 2017-08-08:
I'll install Ubuntu 17.04 and check what changed from 16.04.

ilpincy, 2017-08-08:
OK, it seems to be an issue with the Ubuntu 16.04 package, which apparently can't be run on Ubuntu 17.04. If I compile ARGoS on Ubuntu 17.04, then everything works. I'll make packages later in the week - in the meantime, install ARGoS from the sources. Make sure to uninstall the binary package first.

Mbt925, 2017-08-08:
Thank you for checking.

Mbt925, 2017-08-08:
How can I uninstall the binary package?

ilpincy, 2017-08-08:
Type 

    sudo dpkg -r argos3_simulator

closed 2017-08-15 by Mbt925