ilpincy/argos3 #43: Problem compiling from source code

Mbt925, 2017-08-09:
As you suggested, i tried to compile ARGoS from source code, but i am getting this:

```
-:~/Desktop/argos3/build_simulator$ cmake ../src
-- GCC/G++ version >= 4.3
-- Could NOT find GooglePerfTools (missing:  GOOGLEPERFTOOLS_LIBRARY GOOGLEPERFTOOLS_INCLUDE_DIR) 
-- Configuring done
-- Generating done
-- Build files have been written to: /Desktop/argos3/build_simulator
```

allsey87, 2017-08-09:
GooglePerfTools is not required. What happens when you run:
```
cd /Desktop/argos3/build_simulator
make
```

Mbt925, 2017-08-09:
Thanks. it's finished successfully.
argos3 command can not be recognized by the system. How should I fix this?

allsey87, 2017-08-09:
The instructions for running ARGoS from source are located here: https://github.com/ilpincy/argos3#running-the-argos-simulator

Please read this documentation carefully before continuing.

ilpincy, 2017-08-09:
You should run the following commands:

    $ mkdir build
    $ cd build
    $ cmake -DCMAKE_BUILD_TYPE=Release ../src
    $ make
    $ make doc
    $ sudo make install

closed 2017-08-09 by ilpincy

Mbt925, 2017-08-09:
Thank you very much. The one command i missed was:
`$ sudo make install`

I was able to run a an example, however i am getting this warning. Is it ok?
`[WARNING]` Error opening directory "/home/mohsen/Desktop/argos3/build_simulator/plugins/simulator/physics_engines/physx/": No such file or directory

ilpincy, 2017-08-09:
Just close and reopen the console. You probably tried to run ARGoS from the sources, and to do it executed the setup script.

ilpincy, 2017-08-09:
Also, make sure that the commands you wrote for compiling are exactly those that I gave you. For instance, it makes a huge difference in performance if you write

    $ cmake -DCMAKE_BUILD_TYPE=Release ../src

or if you write

    $ cmake ../src

The first one is much faster.