ilpincy/argos3 #45: Registering an arbitrary number of anchors

allsey87, 2017-08-16:
Hi @ilpincy,

I'm currently cleaning up a featherstone-based multi-body physical model and prototype model for the Dynamics3d engine. Regarding the anchors, I would like to know what you think would be cleanest solution to the following:

I have an arbitrary number of links (not predictable at compile time), some of which have associated anchors. To update these anchors, I obviously shouldn't (can't) create an arbitrary number of callback functions which each anchor is registered against. To resolve this, I have been using a single callback function for updating the links. This works by using the anchor's index as a key to a ```std::unordered_map``` inside a given Dynamics3d physics model. The map then provides the callback function with the positions and orientations that I need to send back to the space.

This works, but I feel it could be done better. Any ideas? Perhaps it is possible to relax the requirements of / overload ```RegisterAnchorMethod``` to take a function object? At the moment this is not possible:

```c++
class CFunctor {
public:
   CFunctor(UInt32 un_val) : m_unVal(un_val) {}

   void operator()(SAnchor& s_anchor) {
      std::cout << s_anchor.Id;
   }
private:
   UInt32 m_unVal;

};
```
Compiler output when passing instance of CFunctor to ```RegisterAnchorMethod```:
```
src/argos3/core/simulator/physics_engine/physics_model.h:300:12: note:
template argument deduction/substitution failed:
src/plugins/robots/foot-bot/simulator/dynamics2d_footbot_model.cpp:63:18: note:
cannot convert ‘cFunctor’ (type ‘argos::CDynamics2DFootBotModel::CFunctor’) to type 
‘void (argos::CDynamics2DFootBotModel::*)(argos::SAnchor&)’
          cFunctor);
```

ilpincy, 2017-08-19:
> [...] This works by using the anchor's index as a key to a std::unordered_map inside a given Dynamics3d physics model. The map then provides the callback function with the positions and orientations that I need to send back to the space.

This looks to me like a sensible approach. If the number of elements in the map is not large, you might find that using a custom bucket-based implementation is faster. I'd go down this road only in a second moment, and if really necessary.

> Perhaps it is possible to relax the requirements of / overload RegisterAnchorMethod to take a function object? At the moment this is not possible:
I designed the callback to strictly be a method of the physics model. This makes it more memory efficient and easier to write. Using a functor would be more flexible, but less efficient.

If you are really concerned about speed, you might experiment with having a single, standard callback for every anchor (callback as it is defined now), but then within the callback dispatch on the right functor defined by you. This kinda defeats the purpose of the anchors as they are designed now, but it could work.

allsey87, 2017-08-20:
Ok, I will use the std::unordered_map approach for now. Thanks for the feedback!

closed 2017-08-20 by allsey87