ilpincy/argos3 #47: Segmentation fault when robots with Lua controllers are selected

alanmillard, 2017-11-01:
Shift+clicking on a robot should select the robot and display its internal variable values in the Lua editor, but this results in a segmentation fault. I cannot reproduce this problem on my Mac, but it happens every time under Linux. I'm running the latest version of ARGoS (3.0.0-beta48).

The attached file contains the .argos configuration file, .lua controller, and an example core dump.

[core_dump.zip](https://github.com/ilpincy/argos3/files/1434575/core_dump.zip)

ilpincy, 2017-11-01:
I'll look into it. Is this related to the selection bug that is already open? That is due to the Intel driver managing badly multiple graphics cards.

allsey87, 2017-11-01:
@alanmillard could you check if you have `lua_editor="true"` set inside the OpenGL tag?

allsey87, 2017-11-01:
Never mind, I see you do have this variable set from core_dump/obstacle_avoidance.argos. This indeed seems like it may be a duplicate of #40 then.

alanmillard, 2017-11-01:
This certainly looks like it may be a duplicate of #40.

Let me know if there are any tests that I can run for you on this system to provide further information that may be of use.

ilpincy, 2017-11-01:
Try disabling the Intel video card. Unfortunately there isn't much I can do :(

closed 2017-11-01 by ilpincy

allsey87, 2017-11-02:
@ilpincy is it possible to propose a series of tests that we can use to probe this issue a little bit deeper? It would be good if we can collaborate and get #40 resolved at some point.

From my testing, it seems that the entities are being drawn in the selection buffer such that they are offset from what is being drawn in the display buffer. This corruption becomes apparent when the `makeCurrent()` and `doneCurrent()` lines are removed. So at a best guess on my end, the segfault is a result of calling the `makeCurrent()` and `doneCurrent()` functions when drawing incorrectly inside the selection buffer...

ilpincy, 2017-11-04:
I doubt the bug is due to code in ARGoS. If so, it would happen on other Linux boxes, other drivers, etc. It only happens with a specific driver on machines that have multiple graphics cards.

beltrame, 2017-11-04:
Actually it happens on any machine with Intel cards, whether they have multiple cards or not. It happens on mine and I don't have a multi card system.

ilpincy, 2017-11-04:
Good to know, it doesn't happen on my old laptop which also has a single Intel card. It's still driver specific though.