ilpincy/argos3 #53: Entity selection box not anchored to entity

alanmillard, 2017-12-04:
When selecting an entity, the selection box is not anchored to the entity itself (see image below). If a robot is selected, then the position of the box begins anchored on the robot, and then floats up into the air as the robot moves. Moving the camera also causes the selection box to move around.

I've only been able to reproduce this issue in my Ubuntu 16.04 VM (it doesn't happen under macOS). I'm testing this with `3.0.0-beta48` and Qt 5.5.1.

[image]

alanmillard, 2017-12-08:
I've narrowed this down to [Line 540 of `qtopengl_widget.cpp`](https://github.com/ilpincy/argos3/blob/master/src/plugins/simulator/visualizations/qt-opengl/qtopengl_widget.cpp#L540). Commenting out `glPolygonMode(GL_FRONT_AND_BACK, GL_LINE)` "fixes" the problem, which seems odd, because it shouldn't affect any matrices (as far as I'm aware).

`glxinfo | grep "OpenGL version"` prints `OpenGL version string: 3.0 Mesa 17.0.7`.

closed 2018-02-03 by ilpincy

alanmillard, 2018-02-05:
Just an update for anyone else running ARGoS using VMWare Fusion:

So far I've only been able to reproduce this issue in Ubuntu VMs (16.04 and 17.10) running on a Mac via VMWare Fusion (10.1.1). It doesn't seem to be an issue with Mesa, as the bug still occurs after upgrading to version 17.3.3, and everything works fine on a native installation of Ubuntu 16.04 running Mesa 17.2.4.

This seems to be something wrong with the VMWare graphics driver - disabling "Accelerate 3D Graphics" in the VM options fixes the problem, and everything works normally in VirtualBox.

ilpincy, 2018-02-05:
Thanks for the update, that clarifies quite a lot why I can't reproduce the problem on any machine I tried. Should I keep the bug open and mark it as "invalid", meaning that it's not an ARGoS problem?

alanmillard, 2018-02-06:
Yes, I would mark it as invalid. This doesn't seem to be a bug in ARGoS - the line `glPolygonMode(GL_FRONT_AND_BACK, GL_LINE)` should have no side-effects, and only seems to cause problems with VMWare Fusion.

ilpincy, 2018-02-06:
Also, if I remove that line the entire arena becomes a black box...

alanmillard, 2018-02-06:
Sorry, what I meant by commenting out the line "fixing" the problem is that the bounding box doesn't float around anymore, but it will be rendered as a solid block - the same as using `glPolygonMode(GL_FRONT_AND_BACK, GL_FILL)` instead.

I tried printing out the coordinates of `sBBox`, and they're definitely correct - it's just a rendering issue. I haven't been able to reproduce the problem by creating a minimal working example in a separate OpenGL application though, so I wonder if there is something strange going on inside ARGoS that is tolerated by most OpenGL implementations, but not the one used by VMWare Fusion.

ilpincy, 2018-02-07:
That is really weird. I don't know what could be the cause, and unfortunately I don't have much time right now to look into the bug :( If you could have a look into that it would be awesome. Otherwise, as soon as I have some time, I'll try and reproduce it.

alanmillard, 2018-02-07:
I raised this issue on the VMware Fusion forums, and apparently VMware VMs only support the OpenGL 3.3 Core Profile (see here: https://communities.vmware.com/message/2743723). Given that ARGoS is using the deprecated fixed pipeline rendering method that @allsey87 mentioned in #40, which was phased out as of OpenGL 3.1, I don't think we can expect the visualisation plugin to work properly in VMware VMs (although this may only be a problem with VMware Fusion - I haven't tested it with a Windows/Linux host).

This seems like another reason to look into replacing the current visualisation plugin with a modern alternative, unfortunately.

ilpincy, 2018-02-07:
> This seems like another reason to look into replacing the current visualisation plugin with a modern alternative, unfortunately.

I agree, but before the summer I won't be able to do it. I tried asking students to do it, but they mostly got scared and dropped the ball :/

commit 6d3d3ec7 (2018-02-03): Fixes #53 and compilation issue