ilpincy/argos3 #56: Segfault when accessing gripped entity

randUsr, 2018-01-22:
I get a Segemantation fault, when accessing the `EmbodiedEntity` the gripper holds via the `.GetGrippedEntity()` function.
I forked the argos3-examples repository and implemented a loopfunction, that produces the segault with the gripper example.

Steps to reproduce:
1. clone https://github.com/randusr/argos3-examples
1. build & run gripper example
1. press start in gui
1. as soon as the robot grips the cylinder, the Segfault occurs

ilpincy, 2018-01-23:
Thanks for the bugreport! I'll have a look at it in the weekend.

ilpincy, 2018-01-29:
[fix.txt](https://github.com/ilpincy/argos3/files/1671821/fix.txt)
It's a bug in your code. Here the fix. I suggest you to learn how references work in C++, because the bug you had was due to that.

closed 2018-01-29 by ilpincy

randUsr, 2018-01-30:
Thanks for looking through the code and the fix.
Sorry for bothering you with trivialities!