ilpincy/argos3 #58: Segmentation fault (core dumped)

Mbt925, 2018-02-04:
After the latest pull, I am getting a segmentation fault error when running a previously working project:

```
@ubuntu:~/argos3-examples$ argos3 -c experiments/obstacle_avoidance.argos 
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_qtopengl.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./build/controllers/obstacle_avoidance/libobstacle_avoidance.so"
Segmentation fault (core dumped)
```

ilpincy, 2018-02-04:
Can you tell me more about your project? For instance, use gdb to know where the segfault happens?

Mbt925, 2018-02-05:
I compiled ARGoS in release mode. Can I use gdb in this mode?

Also, I noticed that I am getting a black screen when running all ARGoS built-in examples, like synchronization, foraging, etc. The following screenshot belongs to synchronization examples.
[image]

allsey87, 2018-02-05:
@Mbt925 you will need to build the simulator in debug mode to get meaningful output from GDB. Could you do this and provide a backtrace for the segfault?

Mbt925, 2018-02-05:
```
(gdb) where
#0  0x00005555558661d8 in ?? ()
#1  0x00007ffff3248fb8 in argos::CBatteryEquippedEntity::~CBatteryEquippedEntity() () from /usr/local/lib/argos3/libargos3plugin_simulator_entities.so
#2  0x00007ffff3248fe0 in argos::CBatteryEquippedEntity::~CBatteryEquippedEntity() () from /usr/local/lib/argos3/libargos3plugin_simulator_entities.so
#3  0x00007ffff324e59b in void argos::CSpace::RemoveEntity<argos::CBatteryEquippedEntity>(argos::CBatteryEquippedEntity&) ()
   from /usr/local/lib/argos3/libargos3plugin_simulator_entities.so
#4  0x00007ffff324c383 in argos::CSpaceOperationRemoveCBatteryEquippedEntity::ApplyTo(argos::CSpace&, argos::CBatteryEquippedEntity&) ()
   from /usr/local/lib/argos3/libargos3plugin_simulator_entities.so
#5  0x00007ffff3250135 in void argos::CEntityOperation<argos::CSpaceOperationRemoveEntity, argos::CSpace, void>::Dispatch<argos::CBatteryEquippedEntity, argos::CSpaceOperationRemoveCBatteryEquippedEntity>(argos::CSpace&, argos::CEntity&)
    () from /usr/local/lib/argos3/libargos3plugin_simulator_entities.so
#6  0x00007ffff324e7d7 in void argos::CEntityOperation<argos::CSpaceOperationRemoveEntity, argos::CSpace, void>::Hook<argos::CBatteryEquippedEntity, argos::CSpaceOperationRemoveCBatteryEquippedEntity>(argos::CSpace&, argos::CEntity&) ()
   from /usr/local/lib/argos3/libargos3plugin_simulator_entities.so
#7  0x00007ffff7b2cd93 in void argos::CallEntityOperation<argos::CSpaceOperationRemoveEntity, argos::CSpace, void>(argos::CSpace&, argos::CEntity&) ()
   from /usr/local/lib/argos3/libargos3core_simulator.so
#8  0x00007fffe9bd07e3 in argos::CSpaceOperationRemoveCEPuckEntity::ApplyTo(argo---Type <return> to continue, or q <return> to quit---
s::CSpace&, argos::CEPuckEntity&) ()
   from /usr/local/lib/argos3/libargos3plugin_simulator_epuck.so
#9  0x00007fffe9bd3e6b in void argos::CEntityOperation<argos::CSpaceOperationRemoveEntity, argos::CSpace, void>::Dispatch<argos::CEPuckEntity, argos::CSpaceOperationRemoveCEPuckEntity>(argos::CSpace&, argos::CEntity&) ()
   from /usr/local/lib/argos3/libargos3plugin_simulator_epuck.so
#10 0x00007fffe9bd230b in void argos::CEntityOperation<argos::CSpaceOperationRemoveEntity, argos::CSpace, void>::Hook<argos::CEPuckEntity, argos::CSpaceOperationRemoveCEPuckEntity>(argos::CSpace&, argos::CEntity&) ()
   from /usr/local/lib/argos3/libargos3plugin_simulator_epuck.so
#11 0x00007ffff7b2cd93 in void argos::CallEntityOperation<argos::CSpaceOperationRemoveEntity, argos::CSpace, void>(argos::CSpace&, argos::CEntity&) ()
   from /usr/local/lib/argos3/libargos3core_simulator.so
#12 0x00007ffff7b74b9d in argos::CSpace::Destroy() ()
   from /usr/local/lib/argos3/libargos3core_simulator.so
#13 0x00007ffff7b2fe30 in argos::CSimulator::Destroy() ()
   from /usr/local/lib/argos3/libargos3core_simulator.so
#14 0x0000555555570b85 in main ()
```

Mbt925, 2018-02-05:
Segmentation fault only occurs with epuck related project. So I think the problem is related to the epuck plugin. But black screen occurs with all of the samples.

allsey87, 2018-02-05:
Well looking at the last couple of frames from the backtrace, it can be concluded that the [destructor calling delete on m_pcDischargeModel](https://github.com/ilpincy/argos3/blob/master/src/plugins/simulator/entities/battery_equipped_entity.cpp#L61) is the cause of the segfault. This should be a quick fix, since, at a guess, we are trying to delete something that has already been deleted.

As for the black regions in the simulator, could you provide us with some information about your OS, graphics card, and OpenGL version?

Mbt925, 2018-02-05:
Thank you for investigating the issue. I am using VMware with Ubuntu 17.04.
`OpenGL version string: 3.0 Mesa 17.0.7`

allsey87, 2018-02-05:
@Mbt925 thank you for the information. You mentioned that you pulled in the latest changes, can you give us the commit id from the last time you pull changes from the github repo?

Mbt925, 2018-02-05:
commit 6d3d3ec731ec60c659a51414171fa46abccb9713

Mbt925, 2018-02-05:
I pulled in the last commit and the black region is gone.

closed 2018-05-05 by ilpincy

Mbt925, 2018-05-05:
Is this fixed?

ilpincy, 2018-05-05:
I had about 50 new users since this bug report, and nobody else reported this issue. I can't reproduce it either.

Mbt925, 2018-05-21:
Can you run a simple test with the e-puck robot?

Mbt925, 2018-05-22:
Is there an epuck model in ARGoS?
I am using this plugin:
https://github.com/lgarattoni/argos3-epuck

Where is the code you modified? 😊

ilpincy, 2018-05-22:
Yes, there is a model of the e-puck in ARGoS. Just type `argos3 -q entities`, or (better) get acquainted with the ARGoS file structure by [checking the code](https://github.com/ilpincy/argos3/tree/master/src/plugins/robots/e-puck).

The code you're using is an extension made for the e-puck with the Overo board. You don't need that extension for the type of experiment you're running. The segfault you get is due to some issue with that extension. Contact Lorenzo Garattoni to have it fixed.

The fix I made to your code was to use the generic differential steering actuator and the generic proximity sensor, only to allow for compilation to finish. Nothing fancy.

Mbt925, 2018-05-22:
The code am using is just a sample to illustrate the bug.
My project needs a full epuck model. Can I get that using the ARGoS epuck? I need proximity sensors, ground sensor, omni camera, led actuators and...

ilpincy, 2018-05-22:
All of that stuff is part of the Overo board, so - as I wrote above - you need to contact Lorenzo Garattoni.

ilpincy, 2018-05-22:
I tested it on Mac and I get [the same error you do](https://github.com/demiurge-project/argos3-epuck/issues/11).

Mbt925, 2018-05-22:
I am sorry for this, as it is not directly related to ARGoS. It would be GREAT if you could find a fix.

ilpincy, 2018-05-22:
There is a problem with conflicting names, because they have added a `CBatteryEntity` that clashes with the one provided by ARGoS. Even fixing this and other naming issues, there's a problem I can't identify in `CEpuckRabEquippedEntity`. When I comment out [the line that adds it to a robot](https://github.com/demiurge-project/argos3-epuck/blob/master/src/plugins/robots/e-puck/simulator/epuck_entity.cpp#L291), everything works. I don't have much time left to work on this today. I'll see what I can do in the next days.

Mbt925, 2018-05-23:
Thank you for your time. I am looking forward to it. My project is frozen because of this bug.

ilpincy, 2018-05-23:
I fixed the issue. I made a pull request in the main repo, but until they accept the pull request you can get the fixed code here: https://github.com/ilpincy/argos3-epuck.

ilpincy, 2018-05-23:
They merged the pull request.

Mbt925, 2018-05-23:
Thank you very much.