ilpincy/argos3 #61: freeze when capturing frames

liwei-cn, 2018-02-15:
When I clicked the button 'Capture Frames', the videos got stuck. For version 43, it does not have such a problem. I am not sure if this is because of an update from qt4 to qt5.

Thanks,
Wei
[image]

ilpincy, 2018-02-15:
Can you give a little more detail about the issue?

1. What version of Ubuntu are you running?
2. What video drivers are installed on your system? (Output of `lsmod`)
3. The output of `lspci`

liwei-cn, 2018-02-15:
Hi, I am using Ubuntu 16.04.03. The video driver I am using is: i915. 
lspci:
00:00.0 Host bridge: Intel Corporation Device 5904 (rev 02)
00:02.0 VGA compatible controller: Intel Corporation Device 5916 (rev 02)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI (rev 21)
00:16.3 Serial controller: Intel Corporation Device 9d3d (rev 21)
00:1c.0 PCI bridge: Intel Corporation Device 9d10 (rev f1)
00:1c.2 PCI bridge: Intel Corporation Device 9d12 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Device 9d18 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Device 9d4e (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Device 9d71 (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (4) I219-LM (rev 21)
3a:00.0 Network controller: Intel Corporation Device 24fd (rev 78)
3c:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd Device a804