ilpincy/argos3 #62: ARGoSBuildFlags.cmake clobbers CMAKE_CXX_FLAGS and prevents cross compilation

allsey87, 2018-03-05:
[Yocto](https://www.yoctoproject.org/) is one of the most common build system for embedded systems. When using a CMake-based recipe in this build system, CMAKE_CXX_FLAGS is set up automatically to enable effortless cross compilation.

However, ARGoSBuildFlags.cmake clobbers this variable ([see line 19 of ARGoSBuildFlags.cmake](https://github.com/ilpincy/argos3/blob/master/src/cmake/ARGoSBuildFlags.cmake#L19)). By appending `-Wall` to `CMAKE_CXX_FLAGS` instead of overriding it, we are able to completely build ARGoS for our robot inside the Yocto build system.

Is there a reason why this variable needs to be overridden instead of appended to?

ilpincy, 2018-03-05:
I think it's OK to fix it so Yocto can do its job. I'll work on it today in the evening.

allsey87, 2018-03-05:
Great, from our testing, Yocto seems to only set up `CMAKE_C(XX)_FLAGS`. It is fine to override the other variables such as `CMAKE_C(XX)_FLAGS_DEBUG` and `CMAKE_C(XX)_FLAGS_RELEASE`, these variables just contained `-g` and `-DNDEBUG` respectively (possibly the CMake defaults) and overriding them with `-ggdb3` and `-O2 -DNDEBUG` respectively doesn't seem to be an issue.

ilpincy, 2018-03-05:
I'll try and make a generic solution, so Yocto will have full control of those flags.

ilpincy, 2018-03-05:
BTW, I assume you're using a CMake target file, right? If not, please do.

allsey87, 2018-03-05:
For the moment, we are just building ARGoS using `-DARGOS_BUILD_FOR=myrobot` and haven't changed anything else. We are just checking for now whether we can build the core libraries. I'll have to look into what the target file is, but generally speaking, Yocto + CMake automatically handle the installation of all executables and libraries on to an image automatically.

The following is an example of what the `CMAKE_CXX_FLAGS` look like when CMake is triggered by Yocto. Note that the `-march` flag is set and should probably not be `native` for cross compilation.

> CMAKE_CXX_FLAGS = -march=armv7-a -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9  --sysroot=/home/harry/stigbot/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/argos/1.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/harry/stigbot/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/argos/1.0-r0=/usr/src/debug/argos/1.0-r0 -fdebug-prefix-map=/home/harry/stigbot/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/argos/1.0-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/harry/stigbot/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/argos/1.0-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -march=armv7-a -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9  --sysroot=/home/harry/stigbot/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/argos/1.0-r0/recipe-sysroot

ilpincy, 2018-03-05:
CMake requires you to use toolchain files to cross compile. If you don't use those, you're messing it up. It's possible Yocto does it automatically and you don't need to bother. Anyway, be sure to check CMake toolchain files, just to know how to configure things correctly. Here is a brief reference: https://cmake.org/cmake/help/v3.0/manual/cmake-toolchains.7.html#id6.

allsey87, 2018-03-06:
Right, I remember doing this before working with Yocto. These days it is much easier to cross compile, you just need to define a recipe that inherits from the cmake base recipe, and everything is handled automatically...

closed 2018-05-06 by ilpincy

allsey87, 2018-05-07:
I will run this through Yocto and report back if everything is working.

commit 025b9b5c (2018-05-06): Fixes #62