ilpincy/argos3 #67: Issue with compiling argos3 system-wide (linking error for argos3_prof)

daneshtarapore, 2018-04-19:
Am trying to compile argos on Ubuntu 14.04 x86_64 system.

Have made all the dependency packages have been installed. 
sudo apt-get install libfreeimage-dev libfreeimageplus-dev \
  qt5-default freeglut3-dev libxi-dev libxmu-dev liblua5.2-dev \
  lua5.2 doxygen graphviz graphviz-dev asciidoc

However, when compiling, I get the following linker errors for argos3_prof:
Scanning dependencies of target argos3_prof
[ 90%] Building CXX object testing/CMakeFiles/argos3_prof.dir/__/core/simulator/query_plugins.cpp.o
[ 91%] Building CXX object testing/CMakeFiles/argos3_prof.dir/__/core/simulator/main.cpp.o
[ 91%] Building CXX object testing/CMakeFiles/argos3_prof.dir/argos3_prof_automoc.cpp.o
[ 91%] Linking CXX executable argos3_prof
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFReadEXIFDirectory@LIBTIFF_4.0'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFFieldReadCount@LIBTIFF_4.0'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFSetTagExtender@LIBTIFF_4.0'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFCurrentDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libraw.so.15: undefined reference to `GOMP_parallel@GOMP_4.0'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFReadTile@LIBTIFF_4.0'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFSetDirectory@LIBTIFF_4.0'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
......
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFTileSize@LIBTIFF_4.0'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreeimage.so: undefined reference to `TIFFStripSize@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
testing/CMakeFiles/argos3_prof.dir/build.make:169: recipe for target 'testing/argos3_prof' failed
make[2]: *** [testing/argos3_prof] Error 1
CMakeFiles/Makefile2:1734: recipe for target 'testing/CMakeFiles/argos3_prof.dir/all' failed
make[1]: *** [testing/CMakeFiles/argos3_prof.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

Any suggestions to resolve the issue would be much appreciated.

Thanks,
Danesh.

allsey87, 2018-04-19:
Hi Danesh, support for Ubuntu is 14.04 ended a long time ago. Can you upgrade to either 16.04 or 18.04 LTS?

ilpincy, 2018-04-19:
Hi Danesh, Ubuntu 14.04 is not supported anymore because the version of Qt shipped with it is too old. To be able to support both the current Mac OSX and Ubuntu, we had to make a cutoff at Ubuntu 16.04. Anything older than that won't work anymore.

daneshtarapore, 2018-04-19:
Many apologies for the typo. I am using Ubuntu 16.04.4 LTS. The errors reported were with the latest commit of argos3.

Thanks,
Danesh.

ilpincy, 2018-04-19:
In theory `argos_prof` should never be compiled, it's intended only for me to do profiling. Have you installed google-perftools because you saw that CMake was looking for it? ARGoS will compile fine if you remove google-perftools. In the meantime I'll fix this compilation issue.

daneshtarapore, 2018-04-19:
No, google-perftools is not installed.

ilpincy, 2018-04-19:
It might be called gperftools, but argos-prof is only compiled when it is detected in the system. BTW, why are you trying to compile ARGoS from the sources, rather than using the precompiled binaries?

daneshtarapore, 2018-04-19:
I would like to implement the Thymio robot in argos, hence the compilation from source (thanks for your advise!).

Btw, I just installed google-perftools, and now argos3 compiles.

ilpincy, 2018-04-19:
If you plan on making a new robot, please follow the blueprint of the KheperaIV and make a separate repository. It's much simpler to work that way.

I'm checking on my virtual machine what happens with the compilation...

daneshtarapore, 2018-04-19:
Okay, thanks for the suggestion.

ilpincy, 2018-04-19:
... and it worked. I can't make sense of your installation trying to compile `argos3_prof`. That should not happen. Can you share your CMakeCache.txt when the error occurs?

ilpincy, 2018-04-19:
If you check the CMake script here: 
https://github.com/ilpincy/argos3/blob/master/src/testing/CMakeLists.txt#L73

you see that `argos3_prof` is only compiled if you're compiling the simulator and you have Google Perftools installed. I tried on the VM I use to make packages (no gperftools) and the compilation is successful. I don't understand what's going on on your machine...

daneshtarapore, 2018-04-19:
[CMakeLists.txt](https://github.com/ilpincy/argos3/files/1928834/CMakeLists.txt)

ilpincy, 2018-04-19:
I need the file build/CMakeCache.txt, which is generated when you run `cmake`

daneshtarapore, 2018-04-19:
Apologies, posted the wrong one. Here you go: 
[CMakeCache.txt](https://github.com/ilpincy/argos3/files/1928841/CMakeCache.txt)

ilpincy, 2018-04-19:
Reading your file:

```
//Google PerfTools header location
GOOGLEPERFTOOLS_INCLUDE_DIR:PATH=/usr/include/google

//Google PerfTools library location
GOOGLEPERFTOOLS_LIBRARY:FILEPATH=/usr/lib/libtcmalloc.so
```

Those should be set to `NOT-FOUND`.

daneshtarapore, 2018-04-19:
Even when I apt-get remove google-perftools, it still seems to find these files:

//Google PerfTools header location
GOOGLEPERFTOOLS_INCLUDE_DIR:PATH=/usr/include/google

//Google PerfTools library location
GOOGLEPERFTOOLS_LIBRARY:FILEPATH=/usr/lib/libtcmalloc.so

Not sure if any other library is involved here.

ilpincy, 2018-04-19:
So it does seem that you have google perftools installed somehow. Maybe you installed it manually? Anyway, it's enough to remove it and you won't have compilation issues. Or install the official package, which also seems to have worked. Either way, it seems to be a problem specific to your machine.

daneshtarapore, 2018-04-19:
Okay, many thanks.

closed 2018-05-06 by ilpincy