ilpincy/argos3 #76: Argos beta 50 fails to run due to unresolved linking (the corresponding class is not in current release)

eliseofe, 2018-05-17:
The following error occurs:

ubuntu@computer01:~/Ziya/argos3-examples/experiments$ argos3 -c synchronization.argos
[FATAL] Can't load library "/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so" even after trying to add extensions for shared library (so) and module library (so):
/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so: /usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so: undefined symbol: _ZTIN5argos21CMagnetEquippedEntityE
/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so.so: /usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so.so: undefined symbol: _ZTIN5argos21CMagnetEquippedEntityE
/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so.so: OK

CMagnetEquippedEntity is not in current argos release (beta 50), but it is on github (I did not try the github version).

ilpincy, 2018-05-17:
I tried reproducing your problem on Ubuntu 16.04 and Mac OS Sierra, but everything seems to be working correctly.

If you install beta50 and run `argos3 -q all`, what's the output?

I suspect that you have ARGoS installed twice in the system.

eliseofe, 2018-05-17:
Ziya could you please comment on this? Did you install argos twice ? (what
I did today to fix it is apt-get remove argos_simulator and then install
49).

ilpincy, 2018-05-17:
The output of the commands mentioned by @allsey87 and `argos3 -q all` will tell us what's going on :-) We'll wait for an answer.

eliseofe, 2018-05-17:
Cardo ma se non metti Ziya non la mail non ti risponde

deve reinstallare il 50 mo e riprovare, sta lavorando con la 49

eliseofe, 2018-05-17:
First installation did  by dr elio and he removed argos beta 50.(but i am not sure how he installed and removed) and then i installed same version and got same error. 
Do you want me to remove beta 49 and install beta 50?

Regards,
Ziya

ilpincy, 2018-05-17:
Please install beta50 and execute the commands @allsey87 and myself suggested.

ilpincy, 2018-05-17:
Also, please document the exact list of commands you used to uninstall and install ARGoS.

ARArun, 2018-05-18:
I had installed version 49 earlier and it was working fine. Now After I had Upgraded to version 50 using "sudo dpkg -i" facing following problems
[image]

eliseofe, 2018-05-18:
Yes same thing! Guys may you have this magnet thing somewhere so that's why
for you it works?

ilpincy, 2018-05-18:
@ARArun can you post text rather than a screenshot? I need to be able to copy paste from your output.

ilpincy, 2018-05-18:
@eliseofe What I have is the same as you guys. @allsey87 might have more files, but they work for me. I'm looking into it.

ilpincy, 2018-05-18:
I checked the files and there seems to be nothing off. Please try these commands and show me the output in text form (no screenshots, I need to be able to copy-paste):

1. `ls -l /usr/lib/argos3`
2. `ldd /usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so`

What version of Ubuntu are you using? 32 or 64 bits?

ARArun, 2018-05-18:
The result of running the said two commands.

amar@amar-HP-Compaq-8200-Elite-SFF-PC:~/Downloads$ argos3 -v
3.0.0-beta50
amar@amar-HP-Compaq-8200-Elite-SFF-PC:~/Downloads$ argos3 -q all
[FATAL] Can't load library "/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so" even after trying to add extensions for shared library (so) and module library (so): 
/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so: /usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so: undefined symbol: _ZTIN5argos21CMagnetEquippedEntityE
/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so.so: /usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so.so: cannot open shared object file: No such file or directory
/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so.so: OK
amar@amar-HP-Compaq-8200-Elite-SFF-PC:~/Downloads$ ls -l /usr/lib/argos3
total 6520
-rw-r--r-- 1 root root  847744 May  2 10:37 libargos3core_simulator.so
-rw-r--r-- 1 root root  242560 May  2 10:37 libargos3plugin_simulator_dynamics2d.so
-rw-r--r-- 1 root root 2254408 May  2 10:37 libargos3plugin_simulator_dynamics3d.so
-rw-r--r-- 1 root root  693360 May  2 10:37 libargos3plugin_simulator_entities.so
-rw-r--r-- 1 root root  155352 May  2 10:37 libargos3plugin_simulator_epuck.so
-rw-r--r-- 1 root root  200608 May  2 10:37 libargos3plugin_simulator_eyebot.so
-rw-r--r-- 1 root root  435280 May  2 10:37 libargos3plugin_simulator_footbot.so
-rw-r--r-- 1 root root  483744 May  2 10:37 libargos3plugin_simulator_genericrobot.so
-rw-r--r-- 1 root root  269376 May  2 10:37 libargos3plugin_simulator_media.so
-rw-r--r-- 1 root root   89112 May  2 10:37 libargos3plugin_simulator_pointmass3d.so
-rw-r--r-- 1 root root  365264 May  2 10:37 libargos3plugin_simulator_prototype.so
-rw-r--r-- 1 root root  462888 May  2 10:37 libargos3plugin_simulator_qtopengl.so
-rw-r--r-- 1 root root  151168 May  2 10:37 libargos3plugin_simulator_spiri.so
amar@amar-HP-Compaq-8200-Elite-SFF-PC:~/Downloads$ ldd /usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so
	linux-vdso.so.1 =>  (0x00007ffcd51c5000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ff6ce226000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ff6cdf1d000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ff6cdd07000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff6cd93d000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ff6ce9d2000)
amar@amar-HP-Compaq-8200-Elite-SFF-PC:~/Downloads$

ilpincy, 2018-05-18:
Thanks @ARArun, I think I have a solution. I'll make a new beta50 in a few minutes.

ARArun, 2018-05-18:
That would be great and thanks, professor.

ilpincy, 2018-05-18:
OK I made a new beta50 release. Please download it [here](http://iridia.ulb.ac.be/~cpinciroli/extra/h-414/argos3_simulator-3.0.0-x86_64-beta50.deb) and let me know if it works.

ARArun, 2018-05-18:
Thank you, professor. The new link works well.

ilpincy, 2018-05-18:
A-HA!

closed 2018-05-18 by ilpincy

eliseofe, 2018-05-18:
What was the problem?

ilpincy, 2018-05-18:
https://github.com/ilpincy/argos3/commit/d8c1b5dd035410a416534bafb532a41c1d1a8d65

ilpincy, 2018-05-18:
Different versions of `ld` have different linking behaviors, and the order in which libraries are linked may vary. In my tests, the order was such that this issue did not appear. On some systems, the issue causes a crash. The solution was to fix a CMake script that was added recently to make the library dependencies explicit.

eliseofe, 2018-05-18:
I see! Thanks for the explanation and for the fix :-)