ilpincy/argos3 #79: Distributing lights doesn't work

LPitonakova, 2018-06-06:
I am trying to use the distribute tag in order to randomly placed some lights:

```
<distribute>
      <position method="uniform" min="-2,-2,0" max="2,2,0" />
      <orientation method="uniform" min="0,0,0" max="360,0,0" />
      <entity quantity="5" max_trials="100">
        <light id="wallLight" color="yellow" intensity="1.0" medium="leds" />
      </entity>
</distribute>
```

However, this gives an error

`Error inserting a light entity with id "wallLight0". An entity with that id already exists.`

allsey87, 2018-06-06:
Do you also have a light declared in the arena outside of the distribute node? Could you post your whole .argos file ?

ilpincy, 2018-06-06:
I can reproduce the error, working on it now.

LPitonakova, 2018-06-06:
I do not. I checked that id "wallLight" doesn't appear anywhere else. Sorry I don't have the argos file anymore, I changed it since defining lights in the above way didn't work. But I did try to replace <light> with <box> in the above code and using <box> worked. So my guess is that light is the problem.

ilpincy, 2018-06-06:
I found the problem. The solution is a little tricky, but I'll do my best.

ilpincy, 2018-06-06:
Fixed in 12c12a5dea240dad94f5e47ebbb1d05fa0e4c8dc. I'm making a new release, because this fix and a previous one by @allsey87 are pretty urgent.

closed 2018-06-06 by ilpincy