ilpincy/argos3 #85: Error initialising controllers

manooshsamiei76, 2018-09-01:
whenever I want to run any of argos3 experiments I encounter the error of initializing controllers.
I attached the picture of what I write and what I get in the terminal.

[image]

ilpincy, 2018-09-01:
Have you checked that the files mentioned in the error message exist?

manooshsamiei76, 2018-09-01:
yes, I checked. They do not exist and this is my problem!
I found the footbot_diffusion library and placed it manually in an artificial directory that I made. Now it runs but with lots of warnings and errors. Do you know why I do not have any folder named Build?

manooshsamiei76, 2018-09-01:
oh, thank you very much! problem solved.

closed 2018-09-01 by ilpincy