ilpincy/argos3 #86: Segfault on entity transfer between physics engines

jharwell, 2018-10-30:
I think I've figured out how to use ARGoS with multiple physics engines, and have the following in my `.argos` file: 

```
 <physics_engines>                                                                                                                                                                    
 315 │     <dynamics2d id="dyn2d0">                                                                                                                                                           
 316 │      <boundaries>                                                                                                                                                                      
 317 │         <top height="1" />                                                                                                                                                             
 318 │         <bottom height="0" />                                                                                                                                                          
 319 │         <sides>                                                                                                                                                                        
 320 │           <vertex point="0,0" />                                                                                                                                                       
 321 │           <vertex point="6,0" />                                                                                                                                                       
 322 │           <vertex point="6,6" />                                                                                                                                                       
 323 │           <vertex point="0,6" />                                                                                                                                                       
 324 │         </sides>                                                                                                                                                                       
 325 │      </boundaries>                                                                                                                                                                     
 326 │     </dynamics2d>                                                                                                                                                                      
 327 │     <dynamics2d id="dyn2d1">                                                                                                                                                           
 328 │      <boundaries>                                                                                                                                                                      
 329 │         <top height="1" />                                                                                                                                                             
 330 │         <bottom height="0" />                                                                                                                                                          
 331 │         <sides>                                                                                                                                                                        
 332 │           <vertex point="6,0" />                                                                                                                                                       
 333 │           <vertex point="12,0" />                                                                                                                                                      
 334 │           <vertex point="12,6" />                                                                                                                                                      
 335 │           <vertex point="6,6" />                                                                                                                                                       
 336 │         </sides>                                                                                                                                                                       
 337 │      </boundaries>                                                                                                                                                                     
 338 │      </dynamics2d>                                                                                                                                                                     
 339 │   </physics_engines>       
```  
Which I think is the correct config to split a 12x6 rectangle between two physics engines (The center of the arena is at (6,3)). However, when the first robot tries to cross physics engines I get a segfault, with the following backtrace in gdb:

```
#1  0x00007ffff7b477e2 in ticpp::Node::FirstChild (this=this@entry=0x7fffffffc450, value=0x7fffffffd128 "dynamics2d", throwIfNoChildren=throwIfNoChildren@entry=false) at /opt/data/git/argos3/src/core/utility/configuration/tinyxml/ticpp.cpp:299
#2  0x00007ffff7b47a1b in ticpp::Node::FirstChild (this=this@entry=0x7fffffffc450, value="dynamics2d", throwIfNoChildren=throwIfNoChildren@entry=false) at /opt/data/git/argos3/src/core/utility/configuration/tinyxml/ticpp.cpp:287
#3  0x00007fffebdb671e in ticpp::Node::IterateFirst<ticpp::Element> (first=<synthetic pointer>, value="dynamics2d", this=0x7fffffffc450) at /opt/data/git/argos3/src/argos3/core/utility/configuration/tinyxml/ticpp.h:768
#4  ticpp::Iterator<ticpp::Element>::begin (parent=0x7fffffffc450, this=0x7fffffffd110) at /opt/data/git/argos3/src/argos3/core/utility/configuration/tinyxml/ticpp.h:1085
#5  argos::NodeExists (str_tag="dynamics2d", t_node=...) at /opt/data/git/argos3/src/argos3/core/utility/configuration/argos_configuration.h:47
#6  argos::CDynamics2DVelocityControl::CDynamics2DVelocityControl (this=0x55555590c3c8, c_engine=..., f_max_force=<optimized out>, f_max_torque=<optimized out>, t_node=0x7fffffffc450) at /opt/data/git/argos3/src/plugins/simulator/physics_engines/dynamics2d/dynamics2d_velocity_control.cpp:28
#7  0x00007fffebdaf043 in argos::CDynamics2DDifferentialSteeringControl::CDynamics2DDifferentialSteeringControl (this=0x55555590c3c8, c_engine=..., f_max_force=<optimized out>, f_max_torque=<optimized out>, f_interwheel_distance=0.14000000059604645, t_node=<optimized out>) at /opt/data/git/argos3/src/plugins/simulator/physics_engines/dynamics2d/dynamics2d_differentialsteering_control.cpp:20
#8  0x00007fffebb4bcc9 in argos::CDynamics2DFootBotModel::CDynamics2DFootBotModel (this=0x55555590c310, c_engine=..., c_entity=...) at /opt/data/git/argos3/src/plugins/robots/foot-bot/simulator/dynamics2d_footbot_model.cpp:58
#9  0x00007fffebb4cd1b in argos::CDynamics2DOperationAddCFootBotEntity::ApplyTo (c_entity=..., c_engine=..., this=<optimized out>) at /opt/data/git/argos3/src/plugins/robots/foot-bot/simulator/dynamics2d_footbot_model.cpp:373
#10 argos::CEntityOperation<argos::CDynamics2DOperationAddEntity, argos::CDynamics2DEngine, argos::SOperationOutcome>::Dispatch<argos::CFootBotEntity, argos::CDynamics2DOperationAddCFootBotEntity> (c_entity=..., t_plugin=..., this=<optimized out>) at /opt/data/git/argos3/src/argos3/core/simulator/entity/entity.h:345
#11 argos::CEntityOperation<argos::CDynamics2DOperationAddEntity, argos::CDynamics2DEngine, argos::SOperationOutcome>::Hook<argos::CFootBotEntity, argos::CDynamics2DOperationAddCFootBotEntity> (this=<optimized out>, t_plugin=..., c_entity=...) at /opt/data/git/argos3/src/argos3/core/simulator/entity/entity.h:335
#12 0x00007fffebdb0666 in argos::CDynamics2DEngine::AddEntity (this=<optimized out>, c_entity=...) at /opt/data/git/argos3/src/plugins/simulator/physics_engines/dynamics2d/dynamics2d_engine.cpp:151
#13 0x00007ffff7b9d6ec in argos::CSpace::AddEntityToPhysicsEngine (this=0x5555557b5380, c_entity=...) at /opt/data/git/argos3/src/core/simulator/space/space.cpp:192
#14 0x00007ffff7b9647f in argos::CPhysicsEngine::TransferEntities (this=0x555555805860) at /opt/data/git/argos3/src/core/simulator/physics_engine/physics_engine.cpp:256
[... cut ...]
#16 0x00007ffff7b9c51a in argos::CSpace::Update (this=0x5555557b5380) at /opt/data/git/argos3/src/core/simulator/space/space.cpp:127
#17 0x00007ffff7b9b1f1 in argos::CDefaultVisualization::Execute (this=0x555555b20520) at /opt/data/git/argos3/src/core/simulator/visualization/default_visualization.cpp:48
#18 0x0000555555557e66 in main (n_argc=3, ppch_argv=0x7fffffffde58) at /opt/data/git/argos3/src/core/simulator/main.cpp:38

```

Am I configuring ARGoS correctly? Or is this a bug in the entity handoff logic?

- John

ilpincy, 2018-11-04:
I'm a little confused by why you get an XML parsing error at runtime. Can you share your code with me so I can reproduce the problem more easily?

jharwell, 2018-11-05:
My code is at https://github.com/swarm-robotics/fordyca.git (devel branch). If I insert the above XML in place of the single physics engine definition in the input file `exp/single-source.argos`, then as soon as the first robot crosses the middle of the arena I see the above error.

If providing a tarball/compiled .so/video/etc would be helpful for debugging, let me know and I would be happy to!

ilpincy, 2018-11-11:
I managed to reproduce the problem. Working on it.

commit 2665c35d (2018-11-13): Fixes #86
closed 2018-11-13 by ilpincy