ilpincy/argos3 #89: Headless Qt/OpenGL rendering

jharwell, 2018-12-23:
So I've been looking at OpenGL and Qt, and now have some idea of how they build on top of each other, in order to understand how to enable rendering without a graphical window, so that frames can be captured on headless machines.

 It appears as though it may be possible to run Qt in a headless mode, though the documentation is a bit spotty on this--any thoughts? I tried simply replacing the `QApplication` base class with the `QCoreApplication` base class, as implied by what I had read, but Qt did not want to attach widgets to it, and failed during initialization.

Secondly, I know it is possible to use OpenGL in a headless context, but there I can't find a way to replace the default Xorg context created by Qt when you call `initializeOpenGLFunctions()`.

Any help is appreciated!

ilpincy, 2018-12-27:
I'm sure Qt supports headless applications. As for truly headless OpenGL, I don't know - I've never tried it, but this seems a good place to start: https://stackoverflow.com/questions/12157646/how-to-render-offscreen-on-opengl.

jharwell, 2018-12-28:
After playing around with ARGoS defaults for rendering, I have managed to get headless rendering to work (kind of). Not by using Qt/OpenGL headless rendering, but by using xvfb (a virtual framebuffer X server) that is available in standard software repos. It's not truly headless rendering, because the host machine that ARGoS is running on still needs to have an xvfb+X installed, but it seems like a reasonable requirement.

By running ARGoS via `xvfb-run "-screen 0 1600x1200x24" argos3 -c exp/testing.argos`, and modifying the Qt/OpenGL visualization plugin to: 

- Unconditionally grab frames
- Have a minimum size of 1600x1200 (when run under xvfb the grabbed frames are the default minimum size of 320x240)

I'm able to grab frames and convert to a nice high resolution video. So, I'm thinking that quasi-headless rendering could be added to ARGoS relatively simply, by modifying the parsing of the `<frames>` subtree, and adding additional attributes to:

- Enable headless rendering so ARGoS can be run under xvfb
- Control the width/height of the virtually rendered frames

In addition, the parsed width/height would have to be communicated to the `CQTOpenGLMainWindow` class so that `minimumSize()` (which controls what Qt sets the window size to for rendering) renders images of the correct size). Also, I think a fatal error should be thrown if the `enable_headless` attribute is true, but `autoplay` is false, because in that case ARGoS will just hang, waiting for the user to press the play button on the invisible/inaccessible main window.

Thoughts?

ilpincy, 2018-12-29:
Thanks @jharwell for the solution you proposed! I like it because it does not require major intervention in the ARGoS codebase. It would be awesome if this were some somehow portable to MacOSX, since many ARGoS users are on this platform.

> So, I'm thinking that quasi-headless rendering could be added to ARGoS relatively simply, by modifying the parsing of the <frames> subtree, and adding additional attributes to:
>
> * Enable headless rendering so ARGoS can be run under xvfb
> * Control the width/height of the virtually rendered frames
> * Specify # of frames to skip when rendering (the fast-forward button functionality of the graphical window).

I agree with this proposal, we just need to find a suitable XML format to make this work well.

> In addition, the parsed width/height would have to be communicated to the CQTOpenGLMainWindow class so that minimumSize() (which controls what Qt sets the window size to for rendering) renders images of the correct size.

We could avoid this, if we parse the window size properties passed to `xvfb-run`.

> Also, I think a fatal error should be thrown if the enable_headless attribute is true, but autoplay is false, because in that case ARGoS will just hang when run under xvfb, waiting for the user to press the play button on the invisible/inaccessible main window.

I think we could say that, if headless mode is on, we print a warning saying that those options are ignored and set to the correct values. I'd keep fatal errors for things that can't be solved automatically in an intuitive manner.

ilpincy, 2018-12-29:
@allsey87: I think the reason why @jharwell is going for `xvfb` is that, in headless machines, ARGoS would need to have access to an xorg server anyway, but in many clusters this is not an option. Using `xvfb` would do the trick.

allsey87, 2018-12-31:
Fair enough. However, I would point out that a solution based on xorg server is not ideal since most major Linux distributions have now switched to Wayland. Perhaps there is an equivalent of xvfb for Wayland, however, I think the features that an equivalent of xvfb would be based on are still in the pipeline for Wayland, since remote desktop clients, screen recorders etc for Wayland are quite rare at the moment.

jharwell, 2019-03-27:
LTS ubuntu version 18.04 still uses Xorg, so I think that 'xvfb` is a good way to go. I'm going to start trying to implement something in the next week or two on this.

closed 2019-10-12 by ilpincy