ilpincy/argos3 #93: Margin for detecting gimbal lock in CQuaternion::ToEulerAngles is too large

allsey87, 2019-04-15:
The (https://github.com/ilpincy/argos3/blob/3.0.0-beta53/src/core/utility/math/quaternion.h#L189)[thresholds for detecting gimbal lock] are quite large and causing glitching it the QtOpenGL visualization. Changing these thresholds from 0.499 to 0.4999 almost eliminates this issue. Is there any reason not to change these values from 0.499 to either 0.4999 or 0.49999?

[image]

ilpincy, 2019-04-15:
> Is there any reason not to change these values from 0.499 to either 0.4999 or 0.49999?

No, feel free to set it to the value that is most appropriate for your simulations to be stable.

allsey87, 2019-04-15:
Ok. I'll push a fix tomorrow changing the value to: 0.49999

ilpincy, 2019-04-15:
Fine for me!

commit b51a1342 (2019-04-16): Fixes #93
closed 2019-04-16 by allsey87