ilpincy/argos3 #96: Improve representation of vectors and quaternions in Lua

allsey87, 2019-06-04:
Assuming there are no objections, I would like to modify the Lua wrapper such that we specify a metatable for the vector and quaternion tables in Lua. This will enable multiplication between quaternions and vectors etc inside Lua code which currently need to be done manually in Lua. The metatable that I will link to a C++ function that:

1. Pulls the (w,) x, y, z values from the stack
2. Creates the temporary vectors or quaternions
3. Performs the arithmetic using the existing ARGoS functions, and 
4. Pushes the result onto the stack.

The only downside to all of this (of which I am aware) will be a small break in the Lua API, since for this to work efficiently, I would need to store the raw w, x, y, and z values of the quaternion as the default representation in Lua. This will break any code that references `some_quaternion.angle` or `some_quaternion.axis`. The break isn't too bad since neither of these keys will exist (which will flag an error) and I can add a function to the metatable that enables:

`[angle, axis] = some_quaternion.to_angle_axis()`

@ilpincy let me know if I have the green light to implement the described functionality.

ilpincy, 2019-06-04:
Makes a lot of sense. Feel free to proceed!

allsey87, 2019-06-07:
I had a very quick look at this today. I will probably add something like `CLuaMath::RegisterMetatables(m_ptLuaState)` to `CLuaController::CreateLuaState()`, where `CLuaMath` is a class of static methods similar to `CLuaUtility`.

allsey87, 2019-06-13:
@ilpincy fcf9ad1d13da6327996e46afc2f69c1bbd8b7ef2 is my first implementation of the proposed features. They pass my basic tests and will be tested more thoroughly in the coming weeks.

closed 2019-06-20 by allsey87

allsey87, 2019-08-29:
@ilpincy 94dec636bfb94773bd1afdd6fc5f7e1cb85f7ad4 is the final version. Instead of using tables, we now use the userdata type/mechanism to create and manage instances of CVector3, CVector2, and CQuaternion, which are stored in memory that is managed by Lua.