ilpincy/argos3 #99: qt_5.9 not found debian

ritaalamino, 2019-07-03:
I am trying to run a simple test after compiling:
`argos3 -c src/examples/experiments/kilobot_nonblocked_movement.argos`

and I get this. Could not find an answer in the issues.

`[INFO] Loaded library "/usr/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_dynamics3d.so"
[INFO] Loaded library "/usr/lib/argos3/libargos3plugin_simulator_entities.so"
[FATAL] Can't load library "/usr/lib/argos3/libargos3plugin_simulator_epuck.so" even after trying to add extensions for shared library (so) and module library (so): 
/usr/lib/argos3/libargos3plugin_simulator_epuck.so: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/lib/argos3/libargos3plugin_simulator_epuck.so)
/usr/lib/argos3/libargos3plugin_simulator_epuck.so.so: /usr/lib/argos3/libargos3plugin_simulator_epuck.so.so: cannot open shared object file: No such file or directory
/usr/lib/argos3/libargos3plugin_simulator_epuck.so.so: OK`

ilpincy, 2019-07-03:
You're not giving much to work with to spot the problem (i.e., which system, Linux or Mac?), but my guess is that you're on Linux and have Anaconda installed. Anaconda comes with its own version of Qt and it messes with Qt-based programs.

If this is the case, type this command:

    $ which qmake

and make sure that the output corresponds to the system path (`/usr/bin/qmake` or `/usr/local/bin/qmake`). If this is not the case, then you need to go into your `.bashrc` and comment out all the references to Anaconda (in particular, make sure the `PATH` and `LD_LIBRARY_PATH` environment variables do not contain Anaconda-related stuff).

After this, erase the `build` directory in the kilobot repository and redo the procedure from scratch.

If I am wrong in my guess, please send me more information on your system. More specifically:
1. Linux or Mac? Which version?
2. How did you install ARGoS, from the sources or from the package?
3. What is the output of `echo $LD_LIBRARY_PATH` and `echo $PATH`?

ritaalamino, 2019-07-03:
I have installed from the git package.

Distributor ID: Debian
Description:    Debian GNU/Linux 9.9 (stretch)
Release:        9.9
Codename:       stretch

echo `$LD_LIBRARY_PATH` returns nothing
and echo `$PATH` returns `/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games`

I'll try to recompile with the steps you have passed.

ilpincy, 2019-07-03:
The root of this problem is that, for some reason, you have compiled ARGoS with Qt 5.9, but then at run-time the OS links a version of Qt different from 5.9. This is not an issue with ARGoS, but with the configuration of your machine. Therefore, make sure that, in compiling ARGoS, you check which version of Qt has been found (make sure it's 5.9!).

ilpincy, 2019-07-03:
Also, what's the output of `which qmake` and `qmake -version`?

ritaalamino, 2019-07-03:
```
which qmake
/usr/bin/qmake
```

```
qmake -version 
QMake version 3.0
Using Qt version 5.7.1 in /usr/lib/x86_64-linux-gnu
```

ilpincy, 2019-07-03:
So that confirms that you're compiling ARGoS with a version of Qt that's different from 5.9. I don't know why you get that error, but it's something weird with the configuration of your system.

ritaalamino, 2019-07-03:
Yes, thanks! I'll fix that.

ritaalamino, 2019-07-10:
Hi, I had to reopen, just to ask something.
I installed argos3 on another computer and worked nicely. But, the problem with my computer insists.
I have installed the right Qt version. 
```
whereis qmake
qmake: /usr/bin/qmake /home/ritaalamino/anaconda3/bin/qmake

qmake --version
QMake version 3.1
Using Qt version 5.9.7 in /home/ritaalamino/anaconda3/lib
```
I have tried to compile the experiments, but I have this error:
```
/usr/local/lib/argos3/libargos3plugin_simulator_footbot.so: undefined reference to `qt_version_tag@Qt_5.9'
collect2: error: ld returned 1 exit status
embedding/mpga/CMakeFiles/mpga_phototaxis.dir/build.make:123: recipe for target 'embedding/mpga/mpga_phototaxis' failed
make[2]: *** [embedding/mpga/mpga_phototaxis] Error 1
CMakeFiles/Makefile2:1669: recipe for target 'embedding/mpga/CMakeFiles/mpga_phototaxis.dir/all' failed
make[1]: *** [embedding/mpga/CMakeFiles/mpga_phototaxis.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
Is it possible to change some configuration that changes the qmake adress?

ilpincy, 2019-07-10:
The `whereis qmake` command shows two versions of Qt installed: one is Anaconda. That's the root of your problem. This is not an issue with ARGoS, it's just Anaconda interfering with your system. Just make sure that `PATH` and `LD_LIBRARY_PATH` do not contain Anaconda folders, and then recompile everything from scratch.

closed 2019-07-10 by ilpincy