ilpincy/argos3 #100: Compile options with "argos --version"

jharwell, 2019-07-13:
As I am currently experimenting with building ARGoS with different compile options ("--fprofile-generate, --fprofile-use, etc"), I would find it useful to have ARGoS print the options it was build with in addition to the version, when invoked with `-v`. Something like like (similar to gcc):

```
ARGOS_USE_DOUBLE=ON
ARGOS_WITH_LUA=ON
ARGOS_BUILD_NATIVE=ON
ARGOSBuilt with: -mtune=native -march=native -O2 ...
<version>
```

As is, I have a hard time remembering what exactly the version I'm currently running was built with. Would this be something others in the community would find valuable if I were to implement it and open a pull request?

ilpincy, 2019-07-13:
I think it would be a valuable addition, thanks!

jharwell, 2019-07-14:
As I'm starting to look at minor modifications to the ARGoS CMake framework to get the needed options into a .h file, I noticed that ARGoS does not use `-flto` for release builds. Does this break the dynamic library loading, or cause other problems to result in its omission?

ilpincy, 2019-07-14:
That option is to switch certain optimizations on, not to make things work, so no - I have never had problems. I have never fiddled with it so far, but I might in future releases. The issue is that, if I make a modification, it must be portable to a wide set of systems. I admit I don't know all the consequences of using that option on all compilers and all the ways in which people might want to link their code to ARGoS.

allsey87, 2019-07-15:
Just my two cents. It would be nice if the output of `argos -v` or ``argos --version`` would be easy to parse by CMake. This would be useful in 3rd party plugins that need to check whether to use double or single precision for Real or which version of C++ (11, 14, 17) has been used to build ARGoS.

jharwell, 2019-07-15:
What is an example of a format easily parsable by CMake? I was thinking of just doing as gcc does, with one key:value pair per line, with a `:` or `=` separator.

jharwell, 2019-07-15:
Another question: Right now, `ARGOS_WITH_LUA` is unconditionally defined (and Lua support compiled in) if the required libraries/cmake package is found. I think adding the `ARGOS_WITH_LUA` option to the cmake configuration so that users (like me!) can disable Lua support even if it is found would be valuable. On my system, the only reason I need the Lua libraries installed is so that the autocompletion framework I use for coding works with the ARGoS headers. Does this sound reasonable?

jharwell, 2019-08-18:
@ilpincy,@allsey87 gentlest of bumps on this. Any thoughts on the changes I've made regarding this issue and /or the output format of the version query? Thanks!

closed 2019-10-12 by ilpincy