ilpincy/argos3 #108: Access to ARGoS threads in loop functions?

jharwell, 2019-07-28:
Currently it does not seem possible to access the ARGoS threads from the loops functions, and if I want to iterate through all robots and collect some metrics from them (what task they are current executing, are they in collision avoidance, etc.) in order to better understand why I'm observing the performance that I am, I have to create my own threads to do this. Right now I'm using OpenMP, and incur an overhead cost of doing this, and will eventually switch to using regular threads, so I'm wondering if there would be any issue with adding a function to the `CSpace` class and/or its derived classes to allow loop functions to use the existing ARGoS threads? In both the `PreStep()` and `PostStep()` functions, the threads aren't used, and so should be available, unless I've missed something.

If this is possible, I'm thinking of modifying ARGoS such that the loop functions can pass a lambda as a `std::function` callback that each thread will call that gets passed a `CControllableEntity` pointer. 

@ilpincy @allsey87 Thoughts?

jharwell, 2019-08-18:
@ilpincy,@allsey87 gentlest of bumps on this. Do you think this idea has merit? Thanks!

allsey87, 2019-08-18:
Just to be clear, you are using OpenMP inside a robot controller and want to collect data on the state of that controller from the loop function?

If so, what is your motivation for using threads? I ask since I have only every used threads in a robot controller on real hardware. Never in the simulator...

jharwell, 2019-08-18:
@allsey87 what I'm asking about is thr feasibility of using ARGoS threads in the loop functions to iterate over each robot in the swarm via `GetEntitiesByType() `. In the controllers I'm using they each implement a number of interfaces which I use to collect information about what the swarm as a whole is doing by collecting said information from each robot and averaging it. Right now I'm using OpenMP in the loop fu`GetEntitiesByType()nctions (*not* the controller) to iterate over the map provided by `GetEntitiesByType()` (I put the entity pointers into a vector before iterating with OpenMP to reduce overhead). This works reasonably well, BUT there are still additional threads that have to be spun up /down each time step I collect from the swarm, which adds additional overhead. 

If instead of this I could use the general purpose ARGoS threads for this I would not have to take the performance hit of the overhead of OpenMP/additional thread maintenence. What I'm not sure on is if it such functionality could be added safely. I *think* it could, and it would be as simple as adding a target in the multithreaded CSpace classes and then exposing it to the loop functions via a callback. But I don't yet know ARGoS well enough to be sure. 

Does that make sense?

allsey87, 2019-08-19:
I am not very familiar with how @ilpincy has implemented the threads in ARGoS and can't really comment further on this matter. However, I would think it would be simpler to use a thread pool to avoid spining up/down the threads, no?

allsey87, 2019-08-19:
>  But I don't yet know ARGoS well enough to be sure.

I think for these sorts of things you will need to investigate and come to your own conclusions about (and to implement your own solution if needed). @ilpincy and I have almost no time for considering new features at the moment. It is even difficult to find time to track down and resolve non-critical bugs!

closed 2021-04-05 by jharwell