ilpincy/argos3 #114: Differential Steering biases still apply with 0 velocity

DarioAl, 2019-11-06:
Hello,
I experienced what I believe to be a bug while testing the Kilobots plugin but it appears to be related to argos3 itself. 
The differential steering biases are always applied even with 0 velocities causing the robots to move when at rest.

What I have done to reproduce the bug (tested on with diffusion_1): 

> diffusion_1.argos

 set any bias to the differential steering, in my case I tried with
```
      <actuators>
        <differential_steering implementation="default" bias_avg="0.1" bias_stddev="0.1"/>
      </actuators>
```

then, in 

> footboot_diffusion.cpp

 set initial velocity to 0 (e.g. _line 16_) and avoid velocities updates (e.g. _comment line 58_).

ilpincy, 2019-11-09:
Having a look at this right now, will report soon. In theory the code [does the correct thing](https://github.com/ilpincy/argos3/blob/master/src/plugins/robots/generic/simulator/differential_steering_default_actuator.cpp#L114), but maybe that check is too brittle. If so, it's an easy fix.

ilpincy, 2019-11-09:
I can't seem to reproduce the problem with the settings you gave me.

DarioAl, 2019-11-10:
I am using Argos3 as it comes with the .deb argos3_simulator-3.0.0-i686-beta54.deb on Ubuntu18.04.
I am not the only one having the issue but I do not have details about the configuration, I will update you as soon as possible. 

--

ilpincy, 2019-11-10:
Oh, that explains it - the fix for this problem has not been released yet. I'll make a new release today.

DarioAl, 2019-11-10:
Perfect, thank you.

closed 2019-11-10 by DarioAl