ilpincy/argos3 #126: Qt-OpenGL changing color affects other objects

jtrautmann, 2020-01-15:
I added my own class which inherits from `CQTOpenGLUserFunctions` and has a registered user function `void Draw(CFootBotEntity& c_entity)`. This function calls `DrawRay`, using the color green. In principle this works, but there is one problem: this leads to the ground also being drawn in green. My guess is that the function `DrawRay` changes the color internally and that it doesn't get changed back. My first thought was that I could maybe change it back using the `SetColor` function, but then I realized there is no function which would give me the current color, so this doesn't seem to be possible.

I built argos using the current state of the master.
OS: Ubuntu 18.04

allsey87, 2020-01-15:
I guess you (via DrawRay) are changing the global color with the `glColor` function, in which case you should take the current OpenGL color and restore it once you are done. I think this can be done with `glGet` with argument `GL_CURRENT_COLOR` or `glGet` with argument `GL_RGBA_MODE`.

jtrautmann, 2020-01-15:
Thank you for your answer. I tried this and it works:

```
float currentColor[4];
glGetFloatv(GL_CURRENT_COLOR,currentColor);
DrawRay(ray, CColor::GREEN);
glColor3ub(currentColor[0],currentColor[1],currentColor[2]);
```

But as you said, this is a bit of a hack. Of course I could duplicate the from `DrawRay`, but duplicate code is never a good solution. And when would it ever make sense to use the function `DrawRay` then? Maybe there is a logic behind this, but I don't really see it. In my eyes, it makes far more sense to reset the color at the end of the `DrawRay` method.

ilpincy, 2020-01-15:
This should not happen. I'll check it later today.

jtrautmann, 2020-02-04:
I noticed this also happens when selecting a robot: the ground also gets white like the selection box.

jtrautmann, 2020-07-07:
I don't know exactly when this was fixed, but it seems the problem doesn't occur anymore.

closed 2020-07-07 by jtrautmann