ilpincy/argos3 #128: Segmentation fault when calling CSimulator::LoadExperiment

jtrautmann, 2020-01-21:
When I call the function CSimulator::LoadExperiment, I get the following error message:

```
Thread 6 "ros_argos_simul" received signal SIGSEGV, Segmentation fault.
0x00007ffff7aebcfb in argos::CARGoSLog::operator<< <char const*> (
    this=0x7ffff7dd3f40 <argos::LOG>,
    t_msg=0x7ffff7b84a3c "[INFO] Using ")
    at /home/jeremias/Programmieren/KIT/argos3/src/argos3/core/utility/logging/argos_log.h:162
162                 *(m_vecStreams[m_mapStreamOrder.find(pthread_self())->second]) << m_sLogColor << t_msg << reset;
```

The backtrace:

```
#0  0x00007ffff7aebcfb in argos::CARGoSLog::operator<< <char const*> (
    this=0x7ffff7dd3f40 <argos::LOG>,
    t_msg=0x7ffff7b84a3c "[INFO] Using ")
    at /home/jeremias/Programmieren/KIT/argos3/src/argos3/core/utility/logging/argos_log.h:162
#1  0x00007ffff7b313fd in argos::CSimulator::InitFramework (
    this=0x7fffdc000ec0, t_tree=...)
    at /home/jeremias/Programmieren/KIT/argos3/src/core/simulator/simulator.cpp:317
#2  0x00007ffff7b30027 in argos::CSimulator::Init (this=0x7fffdc000ec0)
    at /home/jeremias/Programmieren/KIT/argos3/src/core/simulator/simulator.cpp:134
#3  0x00007ffff7b2ff80 in argos::CSimulator::LoadExperiment (
    this=0x7fffdc000ec0)
    at /home/jeremias/Programmieren/KIT/argos3/src/core/simulator/simulator.cpp:124
```

My code looks like this:

```
CSimulator& cSimulator = CSimulator::GetInstance();
cSimulator.SetExperimentFileName(filename);
cSimulator.LoadExperiment();
cSimulator.Execute();
```

allsey87, 2020-01-21:
Can you detail your use case? Are you trying to use ARGoS as a library?

jtrautmann, 2020-01-21:
@allsey87 I want to use ARGoS for learning collision avoidance with reinforcement learning and combine it with ROS. I now realized the problem seems to be that I don't run my code from above in the main thread. When running it in the main thread, everything works fine. So I'll have to think about how to run the code in the main thread to make it work.

closed 2020-01-21 by jtrautmann