ilpincy/argos3 #132: Qt-OpenGL draw arrow with wrong colors

jtrautmann, 2020-03-04:
If I call `CQTOpenGLUserFunctions::DrawRay`, the displayed ray is not in the color I choose. For example, if I choose `CColor::BLUE` as parameter, the ray is displayed in red, and if I choose `CColor::RED`, it gets displayed in yellow.

I built argos using the commit from January 13.
OS: Ubuntu 18.04

ilpincy, 2020-07-03:
Do you have a code snippet I can try?

jtrautmann, 2020-07-03:
Okay I have a minimal example here:

```c++
/* test_user_functions.h */
#ifndef TEST_USER_FUNCTIONS_H
#define TEST_USER_FUNCTIONS_H

#include <argos3/plugins/simulator/visualizations/qt-opengl/qtopengl_user_functions.h>
#include <argos3/plugins/robots/foot-bot/simulator/footbot_entity.h>

using namespace argos;

class TestUserFunctions : public CQTOpenGLUserFunctions {

public:
    TestUserFunctions();

    void Draw(CFootBotEntity& c_entity);

};

#endif

/* test_user_functions.cpp */
#include "test_user_functions.h"

TestUserFunctions::TestUserFunctions() {
    RegisterUserFunction<TestUserFunctions,CFootBotEntity>(
                                                &TestUserFunctions::Draw);
}

void TestUserFunctions::Draw(CFootBotEntity& c_entity) {
    CRay3 ray(CVector3(0,0,0.01), CVector3(1,0,0));
    CColor color = CColor::BLUE;
    DrawRay(ray, color);
}

REGISTER_QTOPENGL_USER_FUNCTIONS(TestUserFunctions, "test")
```

ilpincy, 2020-07-03:
OK, will have a look today and report my findings.

ilpincy, 2020-07-07:
I did test the code but I couldn't reproduce the issue. On what platform are you working? My test was on OSX 10.15.4.

jtrautmann, 2020-07-07:
OK, interesting... I'm working on Ubuntu 18.04.