ilpincy/argos3 #138: Support for 3rd party robot plug-ins with OBJ models

alanmillard, 2020-07-03:
I have one 'clean' version of ARGoS installed system-wide under `/usr/local/`, as well as multiple 'experimental' versions installed in local directories by specifying `CMAKE_INSTALL_PREFIX`. This is so that I can `make install` experimental 3rd party plug-ins into a locally installed version of ARGoS without interfering with anything else. To select which installation of ARGoS I'd like to be active, I use shell scripts to set the following environment variables:

```
export PATH=/PATH/TO/argos3_install/bin:$PATH
export PKG_CONFIG_PATH=/PATH/TO/argos3_install/lib/pkgconfig
export LD_LIBRARY_PATH=/PATH/TO/argos3_install//lib/argos3
export DYLD_LIBRARY_PATH=/PATH/TO/argos3_install/lib/argos3
```

This worked fine until I started experimenting with robot plug-ins that use custom OBJ models, as `CQTOpenGLMainWindow::ReadSettingsPreCreation()` checks to see whether `model_dir` already exists in your settings (on macOS, these are stored in `~/Library/Preferences/info.argos-sim.ARGoS.plist`): https://github.com/ilpincy/argos3/blob/master/src/plugins/simulator/visualizations/qt-opengl/qtopengl_main_window.cpp#L211

As my system-wide installation of ARGoS is the first one that I ever ran (before `model_dir` was defined), it created `info.argos-sim.ARGoS.plist` with paths pointing to `/usr/local/` as returned by `CSimulator::GetInstance().GetInstallationDirectory()` (and therefore `ARGOS_INSTALL_PREFIX`, which is set from `CMAKE_INSTALL_PREFIX`) in this else-clause: https://github.com/ilpincy/argos3/blob/master/src/plugins/simulator/visualizations/qt-opengl/qtopengl_main_window.cpp#L218

Any other installation of ARGoS that is run thereafter will not update the `model_dir` path because it has already been defined, which can result in a fatal error if a custom OBJ file is not found: https://github.com/ilpincy/argos3/blob/master/src/plugins/simulator/visualizations/qt-opengl/qtopengl_obj_model.cpp#L27

This is easily fixed by deleting `~/Library/Preferences/info.argos-sim.ARGoS.plist` so that previous paths are ignored, or by explicitly setting them as described in the README:

```
$ defaults write info.argos-sim.ARGoS MainWindow.texture_dir -string "/PATH/TO/argos3/src/plugins/simulator/visualizations/qt-opengl/textures/"
$ defaults write info.argos-sim.ARGoS MainWindow.model_dir -string "/PATH/TO/argos3/src/plugins/simulator/visualizations/qt-opengl/models/"
$ defaults write info.argos-sim.ARGoS MainWindow.icon_dir -string "/PATH/TO/argos3/src/plugins/simulator/visualizations/qt-opengl/icons/"
$ killall -u YOURUSERNAME cfprefsd
```

This is fine if you're willing to `make install` your plug-in into an ARGoS installation, but my understanding was that setting the `ARGOS_PLUGIN_PATH` environment variable should allow you to run ARGoS with a plug-in that resides somewhere else (including any custom OBJ files that are separate from your ARGoS installation).

Is it possible to set `model_dir` to a colon-separated path, so that it can include both OBJ models that come installed with ARGoS and those packaged with separate plug-ins? Alternatively, is there a better place to store these settings (e.g. in another environment variable)? The file `info.argos-sim.ARGoS.plist` seems like a strange place to keep these settings, as the rest of the fields are things like:

```
LuaEditor.position
LuaEditor.size

MainWindow.docks
MainWindow.position
MainWindow.size
```

These settings frequently change between executions of ARGoS as the user alters the GUI. Although the following path fields are related to visualisation, they are unlikely to change in the same way:

```
MainWindow.icon_dir
MainWindow.model_dir 
MainWindow.texture_dir
```

I'm not sure if this is an issue or a design choice, but I just got tripped up by it!

ilpincy, 2020-07-04:
I think an event better solution could be to remove the saved folder altogether. It's a common cause of annoyance when people switch from packaged to manually installed ARGoS or viceversa. If the storage folder is anyway under the installation path, I see no reason to save it because it can be detected. I think this would solve your problem, right?

alanmillard, 2020-07-06:
Yes, good point :) these folders for built-in ARGoS assets are stored under the installation path anyway, so not sure why they need to be recorded separately - looks like this feature was added for icons/textures 8 years ago (https://github.com/ilpincy/argos3/commit/b9c310af15cbbf113dac8522164160dc459e07de), and then copied recently for models (https://github.com/ilpincy/argos3/commit/886037b8745f3038b1d71736dbea3cce0c158c8a).

What would the best solution be for 3rd party plug-ins with OBJ models that the user does not wish to `make install` into their ARGoS installation though? For example, if ARGoS is installed to `/usr/local` and therefore only looks under `/usr/local/include/argos3/plugins/simulator/visualizations/qt-opengl/models/` then it will miss 3rd party models stored elsewhere. Could `ARGOS_PLUGIN_PATH` also be used to refer to external model directories, or at the semantics of this environment variable restricted to dynamic libraries?

ilpincy, 2020-07-06:
We could add a new environment variable called `ARGOS_ASSET_PATH` with the same logic as `ARGOS_PLUGIN_PATH`.

ilpincy, 2020-07-06:
I have deleted the code that saves/restores the folders, check 167cda10b56c63b728bc9036403e9f6f649cc191.

alanmillard, 2020-07-07:
Great thanks, that should fix it! The corresponding part of the README will need to be updated too, but perhaps not until the next tagged release.

An `ARGOS_ASSET_PATH` environment variable sounds like a good solution for 3rd party plug-ins. Would this be implemented in the same place, or in `GetModelDir()`?