ilpincy/argos3 #140: Add a debug actuator to ARGoS

allsey87, 2020-07-15:
So, I have been using a [debug actuator in argos3-srocs](https://github.com/allsey87/argos3-srocs/blob/master/src/plugins/robots/generic/control_interface/ci_debug_actuator.h) for some time now which is very useful for separating debugging information from multiple robots and selectively printing that information to the console (stderr/stdout) or routing it to a file.

The debug actuator supports multiple interfaces that a robot can write to. One interface I typically use is a `draw` interface to which robot controllers can write strings such as `arrow(X1,Y1,Z1)(X2,Y2,Z2)` which I can then pick up in the qt-opengl user functions to plot this information on the GUI.

I would like to know whether such functionality would be of interest in the core ARGoS repository...

jharwell, 2020-07-15:
My two cents: I think this would be a good thing to add to ARGoS, as it would help with initial development of control algorithms, allowing you to check if the actuator outputs make sense for whatever you are doing.

allsey87, 2020-07-17:
I would state though that using an actuator to achieve this sort of debugging is a hack. Proper APIs for per robot logging/debug drawing should be a core feature of ARGoS. Shoehorned these features into the actuator, loop functions, qt-opengl user functions APIs is bad software design.

alanmillard, 2020-07-17:
Some kind of visual controller debugging feature would certainly be appreciated, whether it's implemented as an actuator hack or as core ARGoS functionality. I've used `CQTOpenGLUserFunctions` before to draw vectors specified by Lua controllers, which is particularly useful for students trying to learn ARGoS and debug their controller code, but the implementation always feels a bit awkward.

allsey87, 2020-07-17:
I definitely agree, recently we have also been using `CQTOpenGLUserFunctions` in our simulations to draw arrows representing the topological relationships between robots.