ilpincy/argos3 #146: Creating movable shapes from Box and Cylinder entity

vivek-shankar, 2020-11-09:
Hai, 

I am working on an approach to collaboratively transport objects of random shapes from one point to another. 
In order to create different shapes, I combined the basic shapes (box and cylinder) to form complex shapes, while I create these shapes, I have some regions that overlap. In this way, I am able to create shapes only if the object is set to be non-moveable. When I set the attribute movable to true the object separates and does not stick together. I can understand that this is normal.  
I am looking for a way to create movable shapes from basic objects that stick together as a single object. Can anyone point me in the right direction to realize this? 
I am adding a screenshot of one of the shapes, I was trying to create.  

Thank you. 

-Vivek

[image]

ilpincy, 2020-11-09:
You need to modify the physics engine to allow for joints. The way I would go with it is either:
1. Define a prism entity, which you configure in the XML
2. If you need hollow objects as in the picture, when you can use the prototype entity and the 3D physics engine (ask @allsey87 for more details, he did all the work)

vivek-shankar, 2020-11-09:
Thank you, @ilpincy !
I don't need hollow objects for the moment.
I think, I will just go ahead and start defining a new entity that builds shapes from the parameters specified in the XML file.

allsey87, 2020-11-09:
The prototype entity with fixed joints should do the trick. Check
out dynamics3d_convex_hull_test.argos and dynamics3d_multi_body_test.argos
in src/testing/examples.

karthiks1701, 2021-03-09:
Following up, by using the prototype entity for a convex hull as mentioned in the  dynamics3d_convex_hull_test.argos we defined an irregular object. In the first photo, the convex hull is defined. In the second photo, the physics engines are defined.
[image]
[image]

We are using Khepera's to push the object. The problem I think that is arising is that Khepera's need a dynamics2d physics engine and the prototype entity requires a dynamics3d physics engine, because of which Khepera's are not able to identify the prototype entity and pass through the entity.(Like shown in the figure below)
[image]
[image]
[image]

Any fixes to avoid this?. 
Thank you.

allsey87, 2021-03-09:
The problem probably is that the Khepera robot only comes with a dynamics2d model, while the prototype entity only provides a dynamics3d model. There are two solutions to this problem (i) create a dynamics3d model for the Khepera robot (since it is just a simple differential drive robot, you could pretty much copy and paste the code from the e-puck dynamics3d model) or (ii) implement a dynamics2d model for the prototype entity.

The former is more straightforward but will be slower (the dynamics2d engine is faster), the latter is faster but more complicated since you need to generate the 2D collision shape based on the shadow of a 3D geometry (which would only work for objects without moving joints). In either case, changes to the code would be necessary.

ilpincy, 2021-03-09:
That's an extended version of the e-puck. ARGoS has a simpler version of the e-puck here: https://github.com/ilpincy/argos3/tree/master/src/plugins/robots/e-puck. The models @allsey87 is referring to are https://github.com/ilpincy/argos3/blob/master/src/plugins/robots/e-puck/simulator/dynamics3d_epuck_model.h and https://github.com/ilpincy/argos3/blob/master/src/plugins/robots/e-puck/simulator/dynamics3d_epuck_model.cpp.

karthiks1701, 2021-03-10:
I followed the first suggestion creating a 3d dynamics model for Khepera. It builds successfully, but when I try to run the  testing/testexperiment.argos file, it shows memory error.  Any idea what is going wrong?. ( I can attach/push the khepera files if necessary )
[image]

allsey87, 2021-03-11:
Hey @karthiks1701, when you want to attach output from the terminal, could you copy and paste it into a code environment instead of attaching screenshots?

Now, we need you to do a couple things in order to help you:
1. Uninstall everything that you are not using, e.g., the ARGoS-SRoCS libraries
2. Set CMake to compile in debug mode and attach a backtrace from where the crash occurs
3. Use `git diff` to show us what changes you have made to the repository

karthiks1701, 2021-03-11:
1. Output for point two.

gdb --args argos3 -c testbuzzexperiment.argos
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from argos3...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/local/bin/argos3 -c testbuzzexperiment.argos
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_kheperaiv.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_buzz.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_qtopengl.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_prototype.so"
argos3: malloc.c:2401: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

Program received signal SIGABRT, Aborted.
0x00007ffff6f49438 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bactrace
Undefined command: "bactrace".  Try "help".
(gdb) backtrace
#0  0x00007ffff6f49438 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff6f4b03a in __GI_abort () at abort.c:89
#2  0x00007ffff6f912f8 in __malloc_assert (
    assertion=assertion@entry=0x7ffff70a5250 "(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)", 
    file=file@entry=0x7ffff70a1c85 "malloc.c", line=line@entry=2401, 
    function=function@entry=0x7ffff70a5a98 <__func__.11510> "sysmalloc")
    at malloc.c:301
#3  0x00007ffff6f95436 in sysmalloc (nb=nb@entry=3968, 
    av=av@entry=0x7ffff72d8b20 <main_arena>) at malloc.c:2398
#4  0x00007ffff6f96763 in _int_malloc (
    av=av@entry=0x7ffff72d8b20 <main_arena>, bytes=bytes@entry=3959)
    at malloc.c:3834
#5  0x00007ffff6f981d4 in __GI___libc_malloc (bytes=3959) at malloc.c:2920
#6  0x00007ffff2c982a4 in btAlignedAllocDefault(unsigned long, int) ()
   from /usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d.so
#7  0x00007ffff2c4f324 in btMultiBody::btMultiBody(int, double, btVector3 const&, bool, bool, bool) ()
   from /usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d.so
#8  0x00007ffff2cc77d3 in argos::CDynamics3DMultiBodyObjectModel::CDynamics3DMultiBodyObjectModel(argos::CDynamics3DEngine&, argos::CComposableEntity&, unsigned---Type <return> to continue, or q <return> ---Type <return> to continue, or q <return> to quit---

karthiks1701, 2021-03-11:
3. Output for point 3.

git diff --cached
diff --git a/src/plugins/robots/kheperaiv/simulator/dynamics3d_kheperaiv_model.cpp b/src/plugins/robots/kheperaiv/simulator/dynamics3d_kheperaiv_model.cpp
new file mode 100644
index 0000000..f1ee3e9
--- /dev/null
+++ b/src/plugins/robots/kheperaiv/simulator/dynamics3d_kheperaiv_model.cpp
@@ -0,0 +1,208 @@
+/**
+ * @file <argos3/plugins/robots/e-puck/simulator/dynamics3d_epuck_model.cpp>
+ *
+ * @author Michael Allwright - <allsey87@gmail.com>
+ */
+
+#include "dynamics3d_kheperaiv_model.h"
+
+#include <argos3/plugins/simulator/physics_engines/dynamics3d/dynamics3d_engine.h>
+#include <argos3/plugins/simulator/physics_engines/dynamics3d/dynamics3d_shape_manager.h>
+
+#include <argos3/plugins/simulator/entities/wheeled_entity.h>
+#include <argos3/plugins/robots/kheperaiv/simulator/kheperaiv_entity.h>
+
+namespace argos {
+
+   /****************************************/
+   /****************************************/
+
+   CDynamics3DKheperaIVModel::CDynamics3DKheperaIVModel(CDynamics3DEngine& c_engine,
+                                                CKheperaIVEntity& c_entity) :
+      /* technically, the CDynamics3DMultiBodyObjectModel should be initialized with 7 children
+         links, however, setting it to 7 makes the epuck less stable for reasons. */
+      CDynamics3DMultiBodyObjectModel(c_engine, c_entity, 3, false),
+      m_cWheeledEntity(c_entity.GetWheeledEntity()) {
+      /* get the required collision shapes */
+      std::shared_ptr<btCollisionShape> ptrBodyShape =
+         CDynamics3DShapeManager::RequestCylinder(m_cBodyHalfExtents);
+      std::shared_ptr<btCollisionShape> ptrWheelShape =
+         CDynamics3DShapeManager::RequestCylinder(m_cWheelHalfExtents);
+      /* calculate the inertia of the collision objects */
+      btVector3 cBodyInertia;
+      btVector3 cWheelInertia;
+      ptrBodyShape->calculateLocalInertia(m_fBodyMass, cBodyInertia);
+      ptrWheelShape->calculateLocalInertia(m_fWheelMass, cWheelInertia);
+      /* calculate a btTransform that moves us from the global coordinate system to the
+         local coordinate system */
+      const SAnchor& sOriginAnchor = c_entity.GetEmbodiedEntity().GetOriginAnchor();
+      const CQuaternion& cOrientation = sOriginAnchor.Orientation;
+      const CVector3& cPosition = sOriginAnchor.Position;     
+      const btTransform& cStartTransform = btTransform(
+         btQuaternion(cOrientation.GetX(),
+                      cOrientation.GetZ(),
+                     -cOrientation.GetY(),
+                      cOrientation.GetW()),
+         btVector3(cPosition.GetX(),
+                   cPosition.GetZ(),
+                  -cPosition.GetY()));
+      /* create a CAbstractBody::SData structure for each body */
+      CAbstractBody::SData sBodyData(
+         cStartTransform * m_cBodyOffset,
+         m_cBodyGeometricOffset,
+         cBodyInertia,
+         m_fBodyMass,
+         GetEngine().GetDefaultFriction());
+      CAbstractBody::SData sLeftWheelData(
+         cStartTransform * m_cLeftWheelOffset,
+         m_cWheelGeometricOffset,
+         cWheelInertia,
+         m_fWheelMass,
+         m_fWheelFriction);
[... 293 lines cut ...]
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
(END)

karthiks1701, 2021-03-11:
`git diff --cached
diff --git a/src/plugins/robots/kheperaiv/simulator/dynamics3d_kheperaiv_model.cpp b/src/plugins/robots/kheperaiv/simulator/dynamics3d_kheperaiv_model.cpp
new file mode 100644
index 0000000..f1ee3e9
--- /dev/null
+++ b/src/plugins/robots/kheperaiv/simulator/dynamics3d_kheperaiv_model.cpp
@@ -0,0 +1,208 @@
+/**
+ * @file <argos3/plugins/robots/e-puck/simulator/dynamics3d_epuck_model.cpp>
+ *
+ * @author Michael Allwright - <allsey87@gmail.com>
+ */
+
+#include "dynamics3d_kheperaiv_model.h"
+
+#include <argos3/plugins/simulator/physics_engines/dynamics3d/dynamics3d_engine.h>
+#include <argos3/plugins/simulator/physics_engines/dynamics3d/dynamics3d_shape_manager.h>
+
+#include <argos3/plugins/simulator/entities/wheeled_entity.h>
+#include <argos3/plugins/robots/kheperaiv/simulator/kheperaiv_entity.h>
+
+namespace argos {
+
+   /****************************************/
+   /****************************************/
+
+   CDynamics3DKheperaIVModel::CDynamics3DKheperaIVModel(CDynamics3DEngine& c_engine,
+                                                CKheperaIVEntity& c_entity) :
+      /* technically, the CDynamics3DMultiBodyObjectModel should be initialized with 7 children
+         links, however, setting it to 7 makes the epuck less stable for reasons. */
+      CDynamics3DMultiBodyObjectModel(c_engine, c_entity, 3, false),
+      m_cWheeledEntity(c_entity.GetWheeledEntity()) {
+      /* get the required collision shapes */
+      std::shared_ptr<btCollisionShape> ptrBodyShape =
+         CDynamics3DShapeManager::RequestCylinder(m_cBodyHalfExtents);
+      std::shared_ptr<btCollisionShape> ptrWheelShape =
+         CDynamics3DShapeManager::RequestCylinder(m_cWheelHalfExtents);
+      /* calculate the inertia of the collision objects */
+      btVector3 cBodyInertia;
+      btVector3 cWheelInertia;
+      ptrBodyShape->calculateLocalInertia(m_fBodyMass, cBodyInertia);
+      ptrWheelShape->calculateLocalInertia(m_fWheelMass, cWheelInertia);
+      /* calculate a btTransform that moves us from the global coordinate system to the
+         local coordinate system */
+      const SAnchor& sOriginAnchor = c_entity.GetEmbodiedEntity().GetOriginAnchor();
+      const CQuaternion& cOrientation = sOriginAnchor.Orientation;
+      const CVector3& cPosition = sOriginAnchor.Position;     
+      const btTransform& cStartTransform = btTransform(
+         btQuaternion(cOrientation.GetX(),
+                      cOrientation.GetZ(),
+                     -cOrientation.GetY(),
+                      cOrientation.GetW()),
+         btVector3(cPosition.GetX(),
+                   cPosition.GetZ(),
+                  -cPosition.GetY()));
+      /* create a CAbstractBody::SData structure for each body */
+      CAbstractBody::SData sBodyData(
+         cStartTransform * m_cBodyOffset,
+         m_cBodyGeometricOffset,
+         cBodyInertia,
+         m_fBodyMass,
+         GetEngine().GetDefaultFriction());
+      CAbstractBody::SData sLeftWheelData(
+         cStartTransform * m_cLeftWheelOffset,
+         m_cWheelGeometricOffset,
+         cWheelInertia,
+         m_fWheelMass,
+         m_fWheelFriction);
+      CAbstractBody::SData sRightWheelData(
+         cStartTransform * m_cRightWheelOffset,
[... 292 lines cut ...]
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
(END)
`

karthiks1701, 2021-03-11:
3. output for git diff --staged 

```
diff --git a/src/plugins/robots/kheperaiv/simulator/dynamics3d_kheperaiv_model.cpp b/src/plugins/robots/kheperaiv/simulator/dynamics3d_kheperaiv_model.cpp
new file mode 100644
index 0000000..f1ee3e9
--- /dev/null
+++ b/src/plugins/robots/kheperaiv/simulator/dynamics3d_kheperaiv_model.cpp
@@ -0,0 +1,208 @@
+/**
+ * @file <argos3/plugins/robots/e-puck/simulator/dynamics3d_epuck_model.cpp>
+ *
+ * @author Michael Allwright - <allsey87@gmail.com>
+ */
+
+#include "dynamics3d_kheperaiv_model.h"
+
+#include <argos3/plugins/simulator/physics_engines/dynamics3d/dynamics3d_engine.h>
+#include <argos3/plugins/simulator/physics_engines/dynamics3d/dynamics3d_shape_manager.h>
+
+#include <argos3/plugins/simulator/entities/wheeled_entity.h>
+#include <argos3/plugins/robots/kheperaiv/simulator/kheperaiv_entity.h>
+
+namespace argos {
+
+   /****************************************/
+   /****************************************/
+
+   CDynamics3DKheperaIVModel::CDynamics3DKheperaIVModel(CDynamics3DEngine& c_engine,
+                                                CKheperaIVEntity& c_entity) :
+      /* technically, the CDynamics3DMultiBodyObjectModel should be initialized with 7 children
+         links, however, setting it to 7 makes the epuck less stable for reasons. */
+      CDynamics3DMultiBodyObjectModel(c_engine, c_entity, 3, false),
+      m_cWheeledEntity(c_entity.GetWheeledEntity()) {
+      /* get the required collision shapes */
+      std::shared_ptr<btCollisionShape> ptrBodyShape =
+         CDynamics3DShapeManager::RequestCylinder(m_cBodyHalfExtents);
+      std::shared_ptr<btCollisionShape> ptrWheelShape =
+         CDynamics3DShapeManager::RequestCylinder(m_cWheelHalfExtents);
+      /* calculate the inertia of the collision objects */
+      btVector3 cBodyInertia;
+      btVector3 cWheelInertia;
+      ptrBodyShape->calculateLocalInertia(m_fBodyMass, cBodyInertia);
+      ptrWheelShape->calculateLocalInertia(m_fWheelMass, cWheelInertia);
+      /* calculate a btTransform that moves us from the global coordinate system to the
+         local coordinate system */
+      const SAnchor& sOriginAnchor = c_entity.GetEmbodiedEntity().GetOriginAnchor();
+      const CQuaternion& cOrientation = sOriginAnchor.Orientation;
+      const CVector3& cPosition = sOriginAnchor.Position;     
+      const btTransform& cStartTransform = btTransform(
+         btQuaternion(cOrientation.GetX(),
+                      cOrientation.GetZ(),
+                     -cOrientation.GetY(),
+                      cOrientation.GetW()),
+         btVector3(cPosition.GetX(),
+                   cPosition.GetZ(),
+                  -cPosition.GetY()));
+      /* create a CAbstractBody::SData structure for each body */
+      CAbstractBody::SData sBodyData(
+         cStartTransform * m_cBodyOffset,
+         m_cBodyGeometricOffset,
+         cBodyInertia,
+         m_fBodyMass,
+         GetEngine().GetDefaultFriction());
+      CAbstractBody::SData sLeftWheelData(
+         cStartTransform * m_cLeftWheelOffset,
+         m_cWheelGeometricOffset,
+         cWheelInertia,
+         m_fWheelMass,
+         m_fWheelFriction);
[... 449 lines cut ...]
+      
+      /* lower base data */
+      static const btVector3 m_cBodyHalfExtents;
+      static const btScalar m_fBodyMass;
+      static const btTransform m_cBodyOffset;
+      static const btTransform m_cBodyGeometricOffset;
+      /* wheel data */
+      static const btVector3 m_cWheelHalfExtents;
+      static const btScalar m_fWheelMass;
+      static const btTransform m_cWheelGeometricOffset;
+      static const btTransform m_cLeftWheelOffset;
+      static const btTransform m_cRightWheelOffset;
+      static const btVector3 m_cBodyToRightWheelJointOffset;
+      static const btVector3 m_cRightWheelToBodyJointOffset;
+      static const btQuaternion m_cBodyToRightWheelJointRotation;
+      static const btVector3 m_cBodyToLeftWheelJointOffset;
+      static const btVector3 m_cLeftWheelToBodyJointOffset;
+      static const btQuaternion m_cBodyToLeftWheelJointRotation;
+      static const btScalar m_fWheelMotorMaxImpulse;
+      static const btScalar m_fWheelFriction;
+   };
+}
+
+#endif
```

karthiks1701, 2021-03-11:
3. Output for git diff 

```
git diff
diff --git a/src/plugins/robots/kheperaiv/CMakeLists.txt b/src/plugins/robots/kheperaiv/CMakeLists.txt
index 4c682a0..6c3bcc9 100644
--- a/src/plugins/robots/kheperaiv/CMakeLists.txt
+++ b/src/plugins/robots/kheperaiv/CMakeLists.txt
@@ -19,6 +19,7 @@ endif(BUZZ_FOUND)
 if(ARGOS_BUILD_FOR_SIMULATOR)
   set(ARGOS3_HEADERS_PLUGINS_ROBOTS_KHEPERAIV_SIMULATOR
     simulator/dynamics2d_kheperaiv_model.h
+    simulator/dynamics3d_kheperaiv_model.h
     simulator/kheperaiv_entity.h
     simulator/kheperaiv_light_rotzonly_sensor.h
     simulator/kheperaiv_measures.h
@@ -68,6 +69,7 @@ if(ARGOS_BUILD_FOR_SIMULATOR)
     ${ARGOS3_SOURCES_PLUGINS_ROBOTS_KHEPERAIV}
     ${ARGOS3_HEADERS_PLUGINS_ROBOTS_KHEPERAIV_SIMULATOR}
     simulator/dynamics2d_kheperaiv_model.cpp
+    simulator/dynamics3d_kheperaiv_model.cpp
     simulator/kheperaiv_entity.cpp
     simulator/kheperaiv_light_rotzonly_sensor.cpp
     simulator/kheperaiv_measures.cpp
```

allsey87, 2021-03-11:
Looks like some sort of memory corrupt issue during the btMultiBody construction. These can occur for numerous reasons. The first thing I would always do is be 100% sure that you are working with a completely clean build and that no older versions of libraries are installed/being linked against. After this, I would like to check whether you are able to run ARGoS with the E-puck instead of the KheperaIV.

karthiks1701, 2021-03-11:
will this example work?. Also will the controller of dynamics2d(epuck_obstacleavoidance_controller) E-puck work for dynamics3d E-puck as well?. if you have an example of E-puck dynamics3d kindly share it.  

https://github.com/ilpincy/argos3-examples/tree/master/experiments/epuck_obstacleavoidance.argos

allsey87, 2021-03-11:
If you change the remove the dynamics2d in your previous configuration file
the e-puck should just use that engine.

If multiple engines are available for a given non-static entity, I believe
ARGoS throws an error, right @ilpincy?

>

ilpincy, 2021-03-11:
@allsey87 Yep, see [here](https://github.com/ilpincy/argos3/blob/master/src/core/simulator/entity/embodied_entity.cpp#L253), called from [here](https://github.com/ilpincy/argos3/blob/master/src/core/simulator/space/space.cpp#L186).

ilpincy, 2021-03-11:
@karthiks1701 The physics engine does not affect the rest of the simulator. ARGoS is designed to treat the physics engine just as any other plugin. So, that example, you simply need to change `dynamics2d` into `dynamics3d` and that's it.

karthiks1701, 2021-03-12:
```
soma@soma:~/argos3-examples$ argos3 -c experiments/epuck_obstacleavoidance.argos 
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_kheperaiv.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_buzz.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_qtopengl.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_prototype.so"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./build/controllers/epuck_obstacleavoidance/libepuck_obstacleavoidance.so"
[INFO] The physics engine "dyn2d" will perform 10 iterations per tick (dt = 0.01 sec)
^Z
[1]+  Stopped                 argos3 -c experiments/epuck_obstacleavoidance.argos
soma@soma:~/argos3-examples$ argos3 -c experiments/epuck_obstacleavoidance.argos 
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_kheperaiv.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_buzz.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_qtopengl.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_prototype.so"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./build/controllers/epuck_obstacleavoidance/libepuck_obstacleavoidance.so"
[INFO] The physics engine "dyn3d" will perform 20 iterations per tick (dt = 0.005 sec)
```
I think I have removed all the unnecessary link(as shown in the output). I also rebuilt argos from scratch.
e-puck with dynamics2d engine works fine. 
e-puck with dynamics 3d case, it does not crash, but it doesn't move(it is just freezed).
[image]

allsey87, 2021-03-12:
Chances are that the dynamics engine is running, it is just that the model has not been calibrated properly. That is, I think the robots will probably start moving if you set the speed higher.

allsey87, 2021-03-12:
I guess the issue is that something isn't being linked together correctly. @karthiks1701 if you are willing to commit to calibrating and testing the dynamics3d model for the Khepera IV, (i.e., adjusting the radius/height of the bodies, setting the motor impulse correctly) so that it matches the speed etc of the dynamics2d model, I can open a pull request against [argos3-kheperaiv](https://github.com/ilpincy/argos3-kheperaiv) and ensure that everything is correct from a compilation/linking perspective.

@ilpincy let me know if you agree with the above solution, if so, I will open a PR against [argos3-kheperaiv](https://github.com/ilpincy/argos3-kheperaiv) to include a dynamics3d model.

karthiks1701, 2021-03-12:
Yes, I think the best solution, for now, is to have a dynamics3d engine for kheperaIV. I can work on calibrating the values for kheperaIV. I will also check if e-puck works for higher speeds.

ilpincy, 2021-03-12:
@allsey87 I'm ok with this!
@karthiks1701 Thanks for all the work!

karthiks1701, 2021-03-12:
@allsey87, you were right increasing the velocity worked for e-puck obstacle avoidance and it dodged the prototype entity as well!.

jharwell, 2021-03-12:
Late to the party, but something that might be worth doing before a final merge is to run the simple ARGoS experiment through valgrind to verify there aren't any read/write violations in the code to be committed, i.e. `valgrind argos3 -c test.argos`.

karthiks1701, 2021-03-14:
I am sorry, should I do so before opening a PR in argos3-khepera?. @ilpincy @allsey87

allsey87, 2021-03-14:
@jharwell I am pretty sure these issues are related to something being off with the linking, but for sure, using Valgrind would be a good solution for further debugging if I am wrong :)

@karthiks1701 I will open the PR on Monday, in the meanwhile, could you have a go at tuning the dynamics3d model for the e-puck to match the dynamics2d model? (for example, the speed, acceleration, ability to push a 100 gram box)? This way, you will know exactly what to do once @ilpincy accepts the PR for the kheperaIV model.

allsey87, 2021-03-14:
I was able to reproduce @karthiks1701's issue and have opened a new issue on the KheperaIV repository: https://github.com/ilpincy/argos3-kheperaiv/issues/6

Once this issue is resolved, I can open a PR for the new dynamics model.

karthiks1701, 2021-03-29:
In the meantime, I tried to create a buzz controller for e-puck, by basically copy-pasting the code from the footbot buzz controller. But there seems to be a error for the range and bearing sensor used by the epuck sensor. Any idea how to solve it @ilpincy and @allsey87. I also had a quick look at [(https://github.com/demiurge-project/argos3-epuck/tree/master/src/plugins/robots/e-puck/simulator)], they seem to have a bug fix for this, could that be the problem?. 

```
 argos3 -c epuck.argos 
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_epuck.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_entities.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_media.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_pointmass3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics3d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_kheperaiv.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3core_simulator.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_spiri.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_dynamics2d.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_footbot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_buzz.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_genericrobot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_eyebot.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_qtopengl.so"
[INFO] Loaded library "/usr/local/lib/argos3/libargos3plugin_simulator_prototype.so"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] The physics engine "dyn2d" will perform 10 iterations per tick (dt = 0.01 sec)
[FATAL] Failed to initialize the space.
[FATAL] Failed to initialize entity "ep1".
[FATAL] Failed to initialize controllable entity "controller_0".
[FATAL] Can't set controller for controllable entity "controller_0"
[FATAL] Error initializing the range and bearing medium sensor
[FATAL] Error parsing attribute "medium"
Attribute does not exist <ticpp.h@1791>
```

allsey87, 2021-03-30:
@karthiks1701 the Github issue tracker is only really for code issues with ARGoS. If you need help using ARGoS you should post something to the ARGoS forum.

karthiks1701, 2021-03-30:
Okay, I will take it to Buzz repository. Thanks!

allsey87, 2021-03-30:
The problem has nothing to do with Buzz and you should ask for help at https://www.argos-sim.info/forum/ not on the Buzz issue tracker. Issue trackers are usually only for issues in the code base.

karthiks1701, 2021-03-30:
Okay. understood!.

allsey87, 2021-04-02:
@karthiks1701 did you have ago at calibrating the dynamics3d model for the e-puck?

karthiks1701, 2021-04-02:
Yes, there is a slight mistake in one of the constants

karthiks1701, 2021-04-02:
```
   const btVector3    CDynamics3DEPuckModel::m_cBodyHalfExtents(0.0362, 0.0236, 0.0362);
   const btScalar     CDynamics3DEPuckModel::m_fBodyMass(0.242);
   const btTransform  CDynamics3DEPuckModel::m_cBodyOffset(btQuaternion(0.0, 0.0, 0.0, 1.0), btVector3(0.0,0.00125,0.0));
   const btTransform  CDynamics3DEPuckModel::m_cBodyGeometricOffset(btQuaternion(0.0, 0.0, 0.0, 1.0), btVector3(0.0, -0.0236, 0.0));
   const btVector3    CDynamics3DEPuckModel::m_cWheelHalfExtents(0.02125,0.0015,0.02125);
   const btScalar     CDynamics3DEPuckModel::m_fWheelMass(0.06);
   const btTransform  CDynamics3DEPuckModel::m_cWheelGeometricOffset(btQuaternion(0.0, 0.0, 0.0, 1.0), btVector3(0.0,-0.0015,0.0));
   const btTransform  CDynamics3DEPuckModel::m_cLeftWheelOffset(btQuaternion(btVector3(1,0,0), SIMD_HALF_PI), btVector3(0.0, 0.02125, -0.0255));
   const btTransform  CDynamics3DEPuckModel::m_cRightWheelOffset(btQuaternion(btVector3(1,0,0), SIMD_HALF_PI), btVector3(0.0, 0.02125, 0.0255));
   const btVector3    CDynamics3DEPuckModel::m_cBodyToRightWheelJointOffset(0.0, -0.0036, 0.0255);
   const btVector3    CDynamics3DEPuckModel::m_cRightWheelToBodyJointOffset(0.0, 0.0015, 0.0);
   const btQuaternion CDynamics3DEPuckModel::m_cBodyToRightWheelJointRotation(btVector3(1,0,0), SIMD_HALF_PI);
   const btVector3    CDynamics3DEPuckModel::m_cBodyToLeftWheelJointOffset(0.0, -0.0036, -0.0255);
   const btVector3    CDynamics3DEPuckModel::m_cLeftWheelToBodyJointOffset(0.0, 0.0015, 0.0);
   const btQuaternion CDynamics3DEPuckModel::m_cBodyToLeftWheelJointRotation(btVector3(1,0,0), SIMD_HALF_PI);
   /* TODO calibrate these values */
   const btScalar     CDynamics3DEPuckModel::m_fWheelMotorMaxImpulse(0.5);
   const btScalar     CDynamics3DEPuckModel::m_fWheelFriction(0.1);
```

All the quaternions used in m_cLeftWheelOffset,m_cRightWheelOffset,m_cBodyToLeftWheelJointRotation,m_cBodyToRightWheelJointRotation should have the vector(1,0,0). I don't have an understanding, but I figured it should be the same for all the wheels so I tried various configurations. After this change, it moves with higher velocity or higher impulse or both. But increasing the impulse/velocity is causing oscillations in the final convergence of e-puck to the desired target location. I am yet to find parameters which make this as smooth as the 2-D case, but I think varying these two should work. Let me know if this makes sense.

allsey87, 2021-04-02:
Hey @karthiks1701, I don't think you need to change those offsets/joint rotations. These values are the same as those in the pipuck model from a different repository which works fine.
```c++
const btTransform  CDynamics3DPiPuckModel::m_cLeftWheelOffset(btQuaternion(btVector3(-1,0,0), SIMD_HALF_PI), btVector3(0.0, 0.02125, -0.0255));
const btTransform  CDynamics3DPiPuckModel::m_cRightWheelOffset(btQuaternion(btVector3(1,0,0), SIMD_HALF_PI), btVector3(0.0, 0.02125, 0.0255));
```
I don't want to get into details about all the coordinate systems here, but in a nutshell these transforms are moving the wheel from the center of the model to the sides. Rotating the left wheel 90 degrees around the X axis, and the right wheel -90 degrees around the X axis.

The most important thing here is the following lines of code from [dynamics3d_epuck_model.cpp](https://github.com/ilpincy/argos3/blob/master/src/plugins/robots/e-puck/simulator/dynamics3d_epuck_model.cpp):
```c++
m_ptrLeftMotor->setVelocityTarget(m_cWheeledEntity.GetWheelVelocities()[0]);
m_ptrRightMotor->setVelocityTarget(m_cWheeledEntity.GetWheelVelocities()[1]);
```
We need to match the units between `setVelocityTarget` and `GetWheelVelocities`. I suspect one is in cm/s while the other is in radians/sec (or something like that).

Once the units above are correct, I think the values that you need to play with to approximate the dynamics2d model are the following:
```c++
const btScalar     CDynamics3DEPuckModel::m_fWheelMotorMaxImpulse(0.5);
const btScalar     CDynamics3DEPuckModel::m_fWheelFriction(0.1);
```