ilpincy/argos3 #147: Robot overlap after deployment

edobrb, 2020-11-13:
Hello, 

during the distribution of the robots in the arena it happens that two footbots overlap after the deployment (I'm using version 3.0.0-beta56). The footbots look like this:
[image]

Moreover, all the proximity sensor return 0 in this particular state.

The deploy is described as follow:
```xml
<distribute>
	<position method="uniform" min="-1.5,-1.5,0" max="1.5,1.5,0" /> 
	<orientation method="gaussian" mean="0,0,0" std_dev="360,0,0" /> 
	<entity quantity="10" max_trials="10000">
		<foot-bot id="fb" rab_range="1">
			<controller config="lua" />
		</foot-bot>
	</entity>
</distribute>
```

[Here](https://github.com/ilpincy/argos3/files/5538222/configuration.zip) the specific .argos file can be downloaded.

Best regards

alanmillard, 2020-11-17:
Thanks for pointing this out @edobrb!

On a related note - `CSpace::Init` does not call `IsCollidingWithSomething` for manually placed entities, so users may explicitly place robots such that they intersect - was this intended @ilpincy? Presumably this check is skipped to allow arena wall geometry to intersect without errors?

ilpincy, 2020-11-17:
Correct - it's the intended behavior when you manually add stuff. The bugfix you contributed is correct though!

closed 2020-11-17 by ilpincy