ilpincy/argos3 #149: camera settings in .argos-file is ignore in latest in master

GameMonkey, 2020-12-08:
If I use the latest release of argos3, there are no issues in defining camera placements under the `qt-opengl` XML tag.

However, if I use the latest commit on the master branch, my camera settings are ignored.

allsey87, 2020-12-08:
Hi Martin,

Can you include the XML of camera settings that you tried to set? The problem probably is that configuration of the cameras in master has changed and includes automated placement when no valid configuration is given. You can find update documentation about how to configure the cameras by running `argos3 -q qt-opengl`

GameMonkey, 2020-12-08:
Of course. As far I can see, it matches the format given in the documentation

```
    <qt-opengl autoplay="false">
      <user_functions library="./libMyQTLib"
                      label="libMyQTLibLabel" />

      <camera>
        <placement idx="0" position="0,0,16.0" look_at="0.1,0,0" />
        <placement idx="1" position="4,0,7" look_at="4,0.1,0.1" />        
        <placement idx="2" position="3,3,4" look_at="1,6,0" />
        <placement idx="3" position="2,3,2" look_at="0,1,0" />
      </camera>
    </qt-opengl>
```

allsey87, 2020-12-08:
This is not the correct format for the Qt-OpenGL plugin on the master branch. If you have compiled the code from master and installed it, the output of `argos3 -q qt-opengl` gives an example of how to configure the cameras:

```xml
<visualization>
  <qt-opengl>
    <camera>
      <placements>
        <placement index="0" position="2,2,2" look_at="1,1,1" />
        <placement index="1" position="1,0,7" look_at="1,0,0" />
        <placement index="2" position="3,3,4" look_at="1,6,0" />
        <placement index="3" position="2,3,2" look_at="0,1,0" />
      </placements>
    </camera>
  </qt-opengl>
</visualization>
```

GameMonkey, 2020-12-08:
Ahh, I checked the format on the wrong branch, my bad! 

I'll change the format in my XML.

allsey87, 2020-12-08:
No problem

closed 2020-12-08 by allsey87