ilpincy/argos3 #150: Error when installing with Homebrew MacOS 10.15.7

callmecampos, 2021-01-20:
I get the following error when installing ARGoS with Homebrew on MacOS Catalina 10.15.7, following [https://www.argos-sim.info/core.php](https://www.argos-sim.info/core.php). Any idea why I'm getting namespace errors with cmath?

```
> brew tap ilpincy/argos3
Error:
  homebrew-core is a shallow clone.
To `brew update`, first run:
  git -C /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core fetch --unshallow
This command may take a few minutes to run due to the large size of the repository.
This restriction has been made on GitHub's request because updating shallow
clones is an extremely expensive operation due to the tree layout and traffic of
Homebrew/homebrew-core and Homebrew/homebrew-cask. We don't do this for you
automatically to avoid repeatedly performing an expensive unshallow operation in
CI systems (which should instead be fixed to not use shallow clones). Sorry for
the inconvenience!
==> Tapping ilpincy/argos3
Cloning into '/usr/local/Homebrew/Library/Taps/ilpincy/homebrew-argos3'...
remote: Enumerating objects: 130, done.
remote: Total 130 (delta 0), reused 0 (delta 0), pack-reused 130
Receiving objects: 100% (130/130), 17.64 KiB | 361.00 KiB/s, done.
Resolving deltas: 100% (43/43), done.
Tapped 1 formula (27 files, 46.5KB).
> brew install bash-completion qt argos3
Error:
  homebrew-core is a shallow clone.
To `brew update`, first run:
  git -C /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core fetch --unshallow
This command may take a few minutes to run due to the large size of the repository.
This restriction has been made on GitHub's request because updating shallow
clones is an extremely expensive operation due to the tree layout and traffic of
Homebrew/homebrew-core and Homebrew/homebrew-cask. We don't do this for you
automatically to avoid repeatedly performing an expensive unshallow operation in
CI systems (which should instead be fixed to not use shallow clones). Sorry for
the inconvenience!
Warning: qt 5.15.2 is already installed and up-to-date
To reinstall 5.15.2, run `brew reinstall qt`
==> Downloading https://homebrew.bintray.com/bottles/bash-completion-1.3_3.catalina.bottle.tar.gz
######################################################################## 100.0%
==> Pouring bash-completion-1.3_3.catalina.bottle.tar.gz
==> Caveats
Add the following line to your ~/.bash_profile:
  [[ -r "/usr/local/etc/profile.d/bash_completion.sh" ]] && . "/usr/local/etc/profile.d/bash_completion.sh"
==> Summary
🍺  /usr/local/Cellar/bash-completion/1.3_3: 189 files, 607.9KB
==> Installing argos3 from ilpincy/argos3
==> Downloading https://homebrew.bintray.com/bottles/cmake-3.19.3.catalina.bottle.tar.gz
Already downloaded: /Users/felipecampos/Library/Caches/Homebrew/downloads/97957bfe5f617c8d884429e1bc2f4c2533cafb17c6a4132917062335c96c0e29--cmake-3.19.3.catalina.bottle.tar.gz
==> Downloading https://github.com/ilpincy/argos3/archive/3.0.0-beta56.tar.gz
==> Downloading from https://codeload.github.com/ilpincy/argos3/tar.gz/3.0.0-beta56
######################################################################## 100.0%
==> Installing dependencies for ilpincy/argos3/argos3: cmake
==> Installing ilpincy/argos3/argos3 dependency: cmake
==> Pouring cmake-3.19.3.catalina.bottle.tar.gz
==> Caveats
Emacs Lisp files have been installed to:
  /usr/local/share/emacs/site-lisp/cmake
==> Summary
🍺  /usr/local/Cellar/cmake/3.19.3: 6,376 files, 63.9MB
==> Installing ilpincy/argos3/argos3
==> cmake ../src -DARGOS_BUILD_NATIVE=ON -DCPACK_PACKAGE_VERSION_MAJOR=3 -DCPACK_PACKAGE_VERSION_MINOR=0 -DCPACK_PACKAGE_VERSION_PATCH=0 -DCPACK_PACKAGE_RELEASE=beta56 -DARGOS_BREW_QT_CELLAR=/usr/local/Cellar/qt
==> make
Last 15 lines from /Users/felipecampos/Library/Logs/Homebrew/argos3/02.make:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:327:9: error: no member named 'islessgreater' in the global namespace
using ::islessgreater;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:328:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:329:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^
13 errors generated.
make[2]: *** [core/CMakeFiles/argos3core_simulator.dir/utility/networking/tcp_socket.cpp.o] Error 1
13 errors generated.
make[2]: *** [core/CMakeFiles/argos3core_simulator.dir/utility/rate.cpp.o] Error 1
make[1]: *** [core/CMakeFiles/argos3core_simulator.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/ilpincy/homebrew-argos3/issues
```

ilpincy, 2021-01-21:
It seems to be an error in XCode: https://stackoverflow.com/questions/61236772/clang-doesnt-compile-cmath-in-macos-catalina

If you try and compile a basic program with `cmath`, does it work?

ilpincy, 2021-01-27:
I have found this thread: https://forums.swift.org/t/is-anyone-else-getting-this-error-when-building-the-compiler-from-master-on-macos/36113/4. It seems to be a common problem, that affects many formulas beyond ARGoS.