ilpincy/argos3 #151: Error when running an example.

AshwinRaviraj, 2021-02-02:
When I runt this on my MacOS `argos3-examples % argos3 -c experiments/synchronization.argos`

I get the following error 
` [INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_qtopengl.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_dynamics3d.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3core_simulator.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_pointmass3d.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_media.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_entities.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_footbot.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_dynamics2d.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_genericrobot.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_spiri.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_prototype.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_epuck.dylib"
[INFO] Loaded library "/usr/local/Cellar/argos3/3.0.0-beta56/lib/argos3/libargos3plugin_simulator_eyebot.dylib"
[INFO] Not using threads
[INFO] Using random seed = 124
[INFO] Using simulation clock tick = 0.1
[INFO] Total experiment length in clock ticks = unlimited
[INFO] Loaded library "./build/controllers/footbot_synchronization/libfootbot_synchronization.so"
[INFO] The physics engine "dyn2d" will perform 10 iterations per tick (dt = 0.01 sec)
[FATAL] Failed to initialize the visualization. Parse error in the <visualization> subtree.
[FATAL] Error initializing QTOpenGL camera settings
[FATAL] Error parsing attribute "idx"
Attribute does not exist <ticpp.h@1791>
`

ilpincy, 2021-02-02:
How did you install ARGoS? I see `beta56`, but it should be `beta57`. The examples were updated when I made the new release.

closed 2021-02-02 by ilpincy

AshwinRaviraj, 2021-02-02:
I cloned this repo `git clone https://github.com/ilpincy/argos3-examples.git argos3-examples`

ilpincy, 2021-02-02:
I know, but the issue is that you're running the old ARGoS beta56, while you should be running the new beta57 instead. The examples work with beta57 now.

AshwinRaviraj, 2021-02-02:
So, I should just update Argos from homebrew?

AshwinRaviraj, 2021-02-02:
Even after that I still gives me the error? Could you explain to me how I can update it from 56 to 57

ilpincy, 2021-02-02:
Yes. There might be a compilation issue, on which I am working now, due to a bad formula for `pkg-config` which breaks compilation for ARGoS and lots of other packages. Compiling from the sources works perfectly well.

AshwinRaviraj, 2021-02-02:
Alright I will try to clone the following repo https://github.com/ilpincy/argos3.git and follow the READme

ilpincy, 2021-02-02:
These are the commands to run:
```
$ cd argos3
$ mkdir build
$ cd build
$ cmake -DCMAKE_BUILD_TYPE=Release
$ make
$ make doc
$ make install
```

AshwinRaviraj, 2021-02-02:
Thanks! I will give it a shot!

AshwinRaviraj, 2021-02-02:
Works!!

ilpincy, 2021-02-02:
Yay! 🥇