ilpincy/argos3 #156: Garbled Spiri model

ilpincy, 2021-02-16:
The Spiri model uses `CQTOpenGLObjModel`. Before beta57, it worked perfectly. Since beta57, it looks garbled (see screenshot):

[image]

I have spent some time trying to fix this issue, but I can't seem to find the problem. `GLError()` returns no error anywhere.

allsey87, 2021-02-17:
I'll look into it by the end of the week and get back to you with either an updated model or a bug fix :)

ilpincy, 2021-02-17:
Thank you! Very appreciated! :-)

allsey87, 2021-02-18:
@mk-heinrich would you have time to look into this with me tomorrow? I guess we just need to reexport the Spiri model with the same settings as used by the other robots

allsey87, 2021-02-19:
Commit b98187cf04a1d431daf68fabdccb270e0193f79b is to blame. It changes part of the code to use GLdouble leaving the just unchanged. I remember looking at this PR and thinking that it looks like a very quick find-and-replace-all job with little testing... I hope it hasn't introduced errors into other parts of the code...

commit ac67ed5b (2021-02-19): Fixes #156
closed 2021-02-19 by allsey87