ilpincy/argos3 #157: how to configure Argos camera?

alacalcasis, 2021-02-20:
Hello, this might be quite an easy issue, but I couldn't find how to configure Argos camera. For some reason when I look at camera configuration xml file I see this:

<camera>
  <placements>
    <placement index="0" position="707.124,0,7.5" look_at="0,0,0" up="0,0,1" lens_focal_length="65" />
  </placements>
</camera>

but it should be something like this:

<camera>
   <placement idx="0" position="-2,0,2" look_at="0,0,0" up="0.707107,0,0.707107" lens_focal_length="20" />
</camera>

but I couldn't find how to change it, thank you in advance,

alacalcasis, 2021-02-20:
Sorry, the previous message missed xml code for camera configuration:

This is what Argos shows as camera configuration

Hello, this might be quite an easy issue, but I couldn't find how to configure Argos camera. For some reason when I look at camera configuration xml file I see this:
"
<camera>
  <placements>
    <placement index="0" position="707.124,0,7.5" look_at="0,0,0" up="0,0,1" lens_focal_length="65" />
  </placements>
</camera>
"

but it should be something like this:

"
<camera>
   <placement idx="0" position="-2,0,2" look_at="0,0,0" up="0.707107,0,0.707107" lens_focal_length="20" />
</camera>
"

but I couldn't find how to change it, thank you in advance,

alacalcasis, 2021-02-20:
OK, clearly I don't know how to insert xml code, let me try again:
`Hello, this might be quite an easy issue, but I couldn't find how to configure Argos camera. For some reason when I look at camera configuration xml file I see this:

<camera>
  <placements>
    <placement index="0" position="707.124,0,7.5" look_at="0,0,0" up="0,0,1" lens_focal_length="65" />
  </placements>
</camera>

but it should be something like this:

<camera>
   <placement idx="0" position="-2,0,2" look_at="0,0,0" up="0.707107,0,0.707107" lens_focal_length="20" />
</camera>

but I couldn't find how to change it, thank you in advance,`

ilpincy, 2021-02-20:
Please use the forum for these requests. The issue tracker is for bugs and enhacements. For examples to configure the camera, please refer to the [ARGoS examples](https://github.com/ilpincy/argos3-examples) and make sure you're using the latest version (beta57).

closed 2021-02-20 by ilpincy