ilpincy/argos3 #159: Setting up Travis CI

jharwell, 2021-02-26:
@ilpincy I have a small build matrix working for ARGoS compilation on travis-ci, which should serve as a starting point for eventually adding unit and integration tests. Right now it just tests "does ARGoS compile" on multiple platforms. I have a few questions before I open the pull request:

- What specific ubuntu versions does ARGoS support ? The github page lists requirements by compiler, not ubuntu version. ARGoS currently compiles via the instructions on the github page on xenial, bionic, and focal, so those are the ones I put in the build matrix. 
- Is testing compilation with the default gcc version that ships with xenial, bionic, and focal sufficient, or should I set travis up to test with specific gcc versions on specific ubuntu versions ? 
- What specific OSX versions does ARGoS support ? The github page doesn't say.  In travis, OSX versions are categorized by the XCode version, and ARGoS currently compiles via the instructions on the github page for XCode 11 and XCode 12. XCode 9/10 sort of worked, but timed out because of how long it took to update homebrew to install the necessary packages.
- Does ARGoS only support specific clang versions in addition to specific XCode versions, and if so what are they ? The github page does not say. 
- What branch in the ARGoS repo should travis integration be attached to ? It can be any branch, though it would probably be best if it was not master, so that we had a staging area for testing pull requests, etc., and only merged to master after sufficient vetting with manual and automated unit testing. Perhaps creating a new branch like "devel", "integration", or "staging" would be the way to go, and require all future ARGoS pull requests target that branch instead of master ?

A few additional notes:
- Travis does not support opensuse currently, so I can't use ci to test compilation.
- I cannot enable travis for the main ARGoS repo--I do not have write access, so it will have to be done by @ilpincy or @allsey87  (maybe just @ilpincy -- I don't know if only owners can set this up or not) after the pull request is merged.

ilpincy, 2021-02-26:
Thanks for setting this up! A few answers:

> What specific ubuntu versions does ARGoS support ? The github page lists requirements by compiler, not ubuntu version. ARGoS currently compiles via the instructions on the github page on xenial, bionic, and focal, so those are the ones I put in the build matrix.

ARGoS should work on any recent Linux distribution. I don't give a special attention to Ubuntu, although I have packages for 18.04 onwards. I also test it on OpenSUSE and I have people who test it on other distributions. I would prefer if the tests were not distribution-specific, because the fact that ARGoS runs on Ubuntu is not sufficient.

> Is testing compilation with the default gcc version that ships with xenial, bionic, and focal sufficient, or should I set travis up to test with specific gcc versions on specific ubuntu versions ?

Any gcc version released in the past 3 years should compile ARGoS. Same for LLVM. Both compilers must work - this means that sometimes we have accept warnings that are not checked in the same way across compilers. Again, using Ubuntu to test compilation is not sufficient.

> What specific OSX versions does ARGoS support ? The github page doesn't say. In travis, OSX versions are categorized by the XCode version, and ARGoS currently compiles via the instructions on the [...]

No specific OSX version, again. Anything released since 2018 should work.

> Does ARGoS only support specific clang versions in addition to specific XCode versions, and if so what are they ? The github page does not say.

All the versions released in the past 3 years.

> What branch in the ARGoS repo should travis integration be attached to ? It can be any branch, though it would probably be best if it was not master, so that we had a staging area for testing pull [...]

Having a `devel` branch would be a good idea!

> Travis does not support opensuse currently, so I can't use ci to test compilation.

This is a problem and possibly a deal breaker :( I need ARGoS to work as widely as possible. At the moment, when I make a release, I compile ARGoS on multiple virtual machines by hand. It's tedious, but it guarantees that ARGoS works widely.

> I cannot enable travis for the main ARGoS repo--I do not have write access, so it will have to be done by @ilpincy or @allsey87 (maybe just @ilpincy -- I don't know if only owners can set this up or not) after the pull request is merged.

I can give you write access, that wouldn't be an issue. The important thing though is that we agree on what gets added to ARGoS and how bugs are addressed.

jharwell, 2021-02-27:
Thanks! I will change my matrix to be compiler+version based rather than OS based. Upon further digging, I think I can work around the opensuse issue by using docker and building/testing ARGoS inside an opensuse container which is running in the host OS.

ilpincy, 2021-02-27:
That would be perfect! Can Travis also create packages automatically?

jharwell, 2021-02-27:
You can run whatever shell commands you need to inside the ci environment, so I think so. The only issue would be figuring out how to get the built package to be pushed to the ARGoS website. You would probably need to setup an authentication key for the server which hosts the ARGoS images/packages.

ilpincy, 2021-02-27:
I usually store the Ubuntu packages on Google Drive, and take advantage of the OpenSUSE build system for storage of RPM packages. MacOSX needs no packages (well, they'd be nice).

jharwell, 2021-02-28:
Working on getting opensuse testing setup. The binary package links on the ARGoS website are broken for opensuse :-(. When I follow the setup instructions on the github page, I get:

```
65296db21133:/ # zypper ar -n openSUSE-13.2-Graphics http://download.opensuse.org/repositories/graphics/openSUSE_13.2/ graphics                                                               
Adding repository 'openSUSE-13.2-Graphics' .............................................................................................................................................[done]
Repository 'openSUSE-13.2-Graphics' successfully added                                                                                                                                        
                                                                                                                                                                                              
URI         : http://download.opensuse.org/repositories/graphics/openSUSE_13.2/                                                                                                               
Enabled     : Yes                                                                                                                                                                             
GPG Check   : Yes                                                                                                                                                                             
Autorefresh : No                                                                                                                                                                              
Priority    : 99 (default priority)                                                                                                                                                           
                                                                                                                                                                                              
Repository priorities are without effect. All enabled repositories share the same priority.                                                                                                   
65296db21133:/ #           zypper refresh;                                                                                                                                                    
Retrieving repository 'openSUSE-13.2-Graphics' metadata ...............................................................................................................................[error]
Repository 'openSUSE-13.2-Graphics' is invalid.                                                                                                                                               
[graphics|http://download.opensuse.org/repositories/graphics/openSUSE_13.2/] Valid metadata not found at specified URL                                                                        
History:                                                                                                                                                                                      
 - [graphics|http://download.opensuse.org/repositories/graphics/openSUSE_13.2/] Repository type can't be determined.                                                                          
 ```

http://download.opensuse.org/repositories/graphics/openSUSE_13.2/ says "resource no longer available" if I try to go to the webpage.  I am able to compile ARGoS on OpenSUSE, but without any of the visualization stuff enabled as a result of this missing graphics package/repository.

ilpincy, 2021-02-28:
Thanks, I'll check that. I have a script that makes a couple of fixes there, that might be the problem. The issue is that the earlier I can look into this is next week.

jharwell, 2021-03-01:
Thanks!

In the meantime, I have gotten CI set up for compilers in ~last 3 years:

- gcc 7-10 on ubuntu
- clang 6-12 on ubuntu
- clang 10-12 on OSX (the ARGoS install instructions say you need to have the latest version of XCode anyway, so not going back further than XCode 10 is OK imo)
- gcc 7-10 on openSUSE

ilpincy, 2021-03-01:
That's perfect! At the moment, there is a known issue on MacOSX on HomeBrew. I narrowed it down to the `pkg-config` formula. However, if you compile directly from the source it works perfectly. I haven't had the time to get the solution to the HomeBrew issue yet.

jharwell, 2021-03-23:
@ilpincy Have you had a chance to look at the openSUSE issue ? The pull request is ready to go sans that issue.

ilpincy, 2021-03-30:
I'm working on fixing a few things in ARGoS today - I'll look into this issue as well.

jharwell, 2021-04-12:
@ilpincy I don't seem to be able to setup travis for this repo, because I am not the owner (even though I have write access). I think you will need to setup the travis CI github integration.

jharwell, 2022-07-05:
OBE--github actions is the way to go.

closed 2022-07-05 by jharwell