ilpincy/argos3 #160: Actuator quadrotor_speed methods dont affect drones' movements

YasmineMoumou, 2021-03-02:
The methods setRotationalSpeed() and setLinearVelocity() provided by the quadrotor_speed actuator (by GetActuator<CCI_QuadRotorSpeedActuator>) don't work. When we try giving a velocity or a rotation to our drones, they do not move. We are only able to control their movement when we replace it with a quadrotor_position actuator and we affect its position.

ilpincy, 2021-03-30:
This is because those methods are currently not implemented (simply for lack of time and use). Implementing them is not too hard, it boils down to setting the speeds rather than the positions in the PointMass3D models. If you feel like checking it out, the code to change is here:

- [CPointMass3DQuadRotorModel interface](https://github.com/ilpincy/argos3/blob/master/src/plugins/simulator/physics_engines/pointmass3d/pointmass3d_quadrotor_model.h)
- [CPointMass3DQuadRotorModel implementation](https://github.com/ilpincy/argos3/blob/master/src/plugins/simulator/physics_engines/pointmass3d/pointmass3d_model.cpp)

SGauvin, 2021-04-08:
This would be nice since there is a 3rd year undergrad project at Polytechnique Montréal where we use ARGoS to simulate drones!

ilpincy, 2021-04-08:
This has been fixed a few days ago, I'm making a new release today.

closed 2021-04-08 by ilpincy