ilpincy/argos3 #163: Default e-puck does not move in 3D

jharwell, 2021-04-01:
I tried the e-puck that comes with ARGoS with the dynamics3D model, and the robot does not move even though I set the wheel speeds. I checked via some debug print statements that the left/right bullet "motors" are being set correctly from the wheel velocities. It's also definitely not a rendering issue--I checked the robot's origin anchor, and it doesn't move from where I put it at the start of simulation.

Any idea why a the robot wouldn't move even if I set the wheel speeds ? @allsey87 this might be a question for you, because it involves the 3D physics engine.

ilpincy, 2021-04-02:
Without looking at the code, this looks like a calibration problem (assuming the code is correct, which I assume it is).

allsey87, 2021-04-02:
Indeed, it needs to be calibrated so the dynamics3d model roughly matches the dynamics2d model. I was suggesting in #146 that @karthiks1701 have a go at fixing this in return for resolving ilpincy/argos3-kheperaiv#6 and adding a dynamics3d model for the Khepera IV.

allsey87, 2021-04-02:
Let use issue #165 for resolving this.

closed 2021-04-02 by allsey87