ilpincy/argos3 #164: Cleaning up `argos3 -q` documentation

jharwell, 2021-04-02:
Hi Carlo,

Following up on the battery issue, I've noticed that there is a lot of copy and paste for common documentation between multiple robots (such as how to configure the battery). I'd like to clean this up so that the documentation for each sensor/actuator that a given robot model has is in exactly one place (probably the `*EquippedEntity.cpp` file), and do something like I did for the `CNoiseInjector` class, where there is a static function to create a tailed documentation string for each robot. Would you be open to merging such a change if I implement it ?

Unrelated, I noticed that the `Init()` functions for the perspective/omnidirectional camera have different names for the XML attributes they expect than the ones that are documented in the robot entity cpp file. For example, `COmndirectionalCameraEquippedEntity::Init()` expects "aperture" and "offset", while when the camera is initialized in the foot-bot, "omnidirectional_camea_aperture" is expected ("offset" is not configurable). Is it supposed to be this way, or are these two supposed to be in sync? In `CFootBotEntity::Init()`, the `CBatteryEquippedEntity::Init()` is called if the `<battery>` node exists, but this is not done for either of the cameras. This seems inconsistent as well. Not sure if there is a good reason for the current implementation or not (if not I'm happy to work on bringing them in sync).

ilpincy, 2021-04-02:
This makes sense! I need to make a new release tonight for allow my students to use ARGoS (I spent 12+ hours fixing compilation issues across devices, and I'm still not done). If you do this cleaning up, it'll go in the following release.

The discrepancies are probably just due to name changes that didn't get propagated to the documentation, so they do need fixing.

I don't have a specific reason why the camera `Init()` methods are not consistent. I think this was done by @allsey87 - maybe he can explain the design choice.

allsey87, 2021-04-02:
Nope, I never touched the `COmndirectionalCameraEquippedEntity`, that was someone else :)

ilpincy, 2021-04-02:
That seems to narrow it down... :-) @jharwell please go ahead with normalizing the cameras!

jharwell, 2021-04-05:
See pull request #167.

closed 2021-04-05 by jharwell