ilpincy/argos3 #169: Convex hull: prototype entity not detected by proximity sensors in dynamics3D

karthiks1701, 2021-04-12:
[image]
[image]

As seen in the above picture, proximity sensors are not detecting objects created by the prototype entity for the convex hull. It works for other objects created from the prototype entity like box, cylinder, and spheres. @allsey87 any idea what is going wrong?

allsey87, 2021-04-12:
I will try recreate the issue on my end. Can you add a [MRE](https://stackoverflow.com/help/minimal-reproducible-example) ARGoS configuration file to the issue that reproduces the issue?

karthiks1701, 2021-04-14:
```
<?xml version="1.0" ?>
<argos-configuration>

  <!-- ************************* -->
  <!-- * General configuration * -->
  <!-- ************************* -->
  <framework>
    <system threads="0" />
    <experiment length="0"
                ticks_per_second="10"
                random_seed="124" />
  </framework>

  <!-- *************** -->
  <!-- * Controllers * -->
  <!-- *************** -->
  <controllers>

    <epuck_obstacleavoidance_controller id="fdc"
                                        library="build/controllers/epuck_obstacleavoidance/libepuck_obstacleavoidance.so">
      <actuators>
        <differential_steering implementation="default" />
      </actuators>
      <sensors>
        <proximity implementation="default" show_rays="true" />
      </sensors>
      <params velocity="50" />
    </epuck_obstacleavoidance_controller>

  </controllers>

  <!-- *********************** -->
  <!-- * Arena configuration * -->
  <!-- *********************** -->
  <arena size="5, 5, 1" center="0,0,0.5">

    <box id="wall_north" size="4,0.1,0.1" movable="false">
      <body position="0,2,0" orientation="0,0,0" />
    </box>
    <box id="wall_south" size="4,0.1,0.1" movable="false">
      <body position="0,-2,0" orientation="0,0,0" />
    </box>
    <box id="wall_east" size="0.1,4,0.1" movable="false">
      <body position="2,0,0" orientation="0,0,0" />
    </box>
    <box id="wall_west" size="0.1,4,0.1" movable="false">
      <body position="-2,0,0" orientation="0,0,0" />
    </box>

      <prototype id="push_object" movable="false">
      <body position="0,0,0" orientation="0,0,0" />
      <links ref="body">
        <link id="body" geometry="convex_hull" mass="10" position="-0.3,-0.3,0.0" orientation="0,0,0">
          (0.15, .30, 0.0) (-0.15, .30, 0) (.15, -.30, 0) (-.15, -.30, 0)
          (.30, .15, 0) (-.30, .15, 0) (.30, -.15, 0) (-.30, -.15, 0)
          (.15, .30,  0.20) (-.15, .30,  0.20) (.15, -.30,  0.20) (-.15, -.30,  .20)
          (.30, .15,  .20) (-.30, .15,  .20) (.30, -.15,  .20) (-.30, -.15,  .20)
          (0.000, 0.000, 0.00) (0.000, 0.000, 0.00)
        </link>
      </links>
    </prototype>

    <prototype id="pt0" movable="true">
    <body position="0,0,0" orientation="0,0,0" />
    <links ref="base">
      <link id="base" geometry="sphere" radius="0.3" mass="10" position="1,1,0" orientation="0,0,0" />
    </links>
    </prototype>

    <!--
        You can distribute entities randomly. Here, we distribute
        10 e-pucks in this way:
        - the position is uniformly distributed
        on the ground, in the square whose corners are (-2,-2) and (2,2)
        - the orientations are non-zero only when rotating around Z and chosen
        from a gaussian distribution, whose mean is zero degrees and
        standard deviation is 360 degrees.
    -->
    

    <distribute>
      <position method="uniform" min="-2,-2,0" max="2,2,0" />
      <orientation method="gaussian" mean="0,0,0" std_dev="360,0,0" />
      <entity quantity="50" max_trials="100">
        <e-puck id="fb">
          <controller config="fdc" />
        </e-puck>
      </entity>
    </distribute>

    <!--
        We distribute 5 boxes uniformly in position and rotation around Z.
    -->
 

  </arena>

  <!-- ******************* -->
  <!-- * Physics engines * -->
  <!-- ******************* -->
  <physics_engines>
    <!--<dynamics2d id="dyn2d" />-->
    <dynamics3d id="dyn3d" iterations="20">
    <floor height="0.001"/>
      <gravity g="9.8" />
    </dynamics3d>
  </physics_engines>

  <!-- ********* -->
  <!-- * Media * -->
  <!-- ********* -->
  <media />

  <!-- ****************** -->
  <!-- * Visualization * -->
  <!-- ****************** -->
  <visualization>
    <qt-opengl>
      <camera>
        <placements>
          <placement index="0" position="0,0,13" look_at="0,0,0" up="1,0,0" lens_focal_length="65" />
        </placements>
      </camera>
    </qt-opengl>
  </visualization>

</argos-configuration>
```

Apologies for the late reply. I also tried for multiple e-pucks. It doesn't recognize other e-pucks. The given Argos code uses the epuck-obstacle avoidance controller present in  [argos3-examples](https://github.com/ilpincy/argos3-examples/blob/master/experiments/epuck_obstacleavoidance.argos) repository.

allsey87, 2021-04-14:
Ok great, I will try to have a look at this tomorrow.

allsey87, 2021-04-15:
Ok, I see the same thing on my end. I look into the reason why this is happening soon.

karthiks1701, 2021-04-16:
Okay Thanks.

allsey87, 2021-04-17:
I will probably have a go at investigating this a bit further towards the end of next week. In the meanwhile, have you tried some more simple shapes? Also, I am not sure whether you should have duplicate points when specifying the convex hull.

Another question that comes to mind is whether or not objects are colliding correctly with the convex hull...

karthiks1701, 2021-04-18:
I will check out this concerns.

allsey87, 2021-04-21:
It seems that for the convex hull link type, `CDynamics3DEngine::CheckIntersectionWithRay` is not even getting called. I think @ilpincy uses the bounding box to determine whether or not a ray query is passed to a particular engine. So at a guess, I would say that there is something off in the bounding box calculation for this particular link type. I will keep digging.

allsey87, 2021-04-21:
Ok, I understand this bug now. By default, Bullet Physics uses the flag `kF_UseSubSimplexConvexCastRaytest` to select a specific algorithm when performing a ray test. I changed this flag to `kF_UseGjkConvexCastRaytest`, which, at the time of the decision, was more accurate (the alternative algorithm was missing some hits). I have upgraded Bullet a couple times since I wrote this code, so I will need to review that decision. Worse case scenario, I will have to perform the ray test twice if the first algorithm doesn't find any hits.

Any thoughts @ilpincy? I think it might make sense to do another release after I fix this...

closed 2021-04-22 by allsey87

ilpincy, 2021-04-22:
Thanks for spotting the problem, it's quite a subtle one! I will make another release this weekend, this fix is important. Do you consider this solution "stable", or you expect to have more work soon?

allsey87, 2021-04-22:
I believe this solution is stable, but I will need to check this the next time I upgrade Bullet (which probably won't happen before December).

ilpincy, 2021-04-22:
That's stable enough :-) Thanks again!

commit 7c8a51b6 (2021-04-22): Fixed 3D ray casting in 3D dynamics engine

Switch the ray test algorithm from kF_UseGjkConvexCastRaytest to the default kF_UseSubSimplexConvexCastRaytest. In previous versions of Bullet, kF_UseSubSimplexConvexCastRaytest was not accurate enough, however, it appears to have been improved, while kF_UseGjkConvexCastRaytest now fails to reliably detect intersections with convex hulls. By not specifying this flag, we are reverting to using the default kF_UseSubSimplexConvexCastRaytest. Fixes #169.