ilpincy/argos3 #172: How to overcome "Lua53Config.cmake  /   lua53-config.cmake" directory issue..?

kashif-hub, 2021-05-11:
How to overcome the issue related to the installation of the ARGoS module and cmake .. command results some error related to "Lua53Config.cmake / lua53-config.cmake" which is shoen in snapshot..?
[image]

ilpincy, 2021-05-11:
This is not an issue with the ARGoS core, but with the module code. They have a misconfigured `CMakeLists.txt`. Please submit an issue with their repository.

closed 2021-05-11 by ilpincy

ilpincy, 2021-05-11:
I don't know - please refer to the people who made that code. Again, this is not the space to ask for coding advice. This is a space to report issues with the ARGoS core.