ilpincy/argos3 #174: Thread affinity in ARGoS

jharwell, 2021-05-14:
In my profiling, I've noticed you can get non-neglible performance improvements (10-25%) using ARGoS pthreads if you tell the OS that each thread is only allowed to run on 1 core, rather than allowing the OS to schedule them wherever it wants. Especially for larger swarms, this improves cache performance for larger controller objects (in terms of memory).

The relevant system call is `pthread_setaffinity_p()'.

Would you be open to merging a change adding this capability to ARGoS? I would make it off by default.

ilpincy, 2021-05-14:
Yes, I think it would be a great addition. Why would you keep it off by default? What's the downside? I don't expect it to break anything.

jharwell, 2021-05-14:
My thoughts in keeping it off by default were:

1. The principle of least surprise: ARGoS has never pinned threads by default, so it might be unexpected behavior.
2. If you are running ARGoS on a system you have exclusive control over (say it has 16 cores), then pinning threads to cores can give a performance boost. If you are running ARGoS on a system you have to share with other users (e.g., a supercomputing cluster, or a shared lab server), then if another user pins their 8 threads to cores 0-7, and you also pin your 8 threads to cores 0-7 (you have no way of knowing in general if other users will pin their threads or not, and if so what cores they will pin to), then you will see degraded performance in comparison to letting the OS handle scheduling threads.

ilpincy, 2021-05-14:
Reason #2 is compelling. Agreed!

jharwell, 2021-05-14:
I was thinking of adding a `pin_threads` attribute to the `<system>` tag, unless you object. How can I document this new option ? The `<system>` tag does not appear in any of the `argos3 -q ....` results. Would the generated ARGoS manpage by a good place for it ?

ilpincy, 2021-05-14:
I would suggest `pin_threads_to_cores`, it's longer but more descriptive. As for documentation, this is indeed a deficiency of the current automatic doc generation. I would say leave the doc part out for the time being, I'll just add it to the website manual.

closed 2021-07-23 by jharwell